emsesp/EMS-ESP32 #1491: 2 fields for Bosch Easycontrol CT200 thermostat lost in version 3.6.3 and 3.6.4

nicootje55, 2023-12-09:
### PROBLEM DESCRIPTION
From version 3.6.3 (also in 3.6.4) the folowing fields are lost in the thermostaat Easycontrol CT200 apparaat/type.
-	hc1 Dauwpunttemperatuur
-	hc1 Relatieve luchtvochtigheid
The field hc1 Ruimtetemperatuur van afstandsbediening is stil there. So there is only 1 entrie.
In version 3.5.1 these entries excist (total of 3 entries) and are working perfect.

### REQUESTED INFORMATION
output of http://ems-esp.local/api/system v3.6.4  (only the Devices part):
  "Devices": [
    {
      "type": "boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 3000-7000iAW/SupraEco/Geo 5xx/WLW196i/WSW196i",
      "device id": "0x08",
      "product id": 172,
      "version": "01.20",
      "entities": 143,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A",
      "handlers ignored": "0x049F 0x04A0 0x04A3 0x04AC 0xF7"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "73.04",
      "entities": 57,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031D 0x031E",
      "handlers ignored": "0x02E0 0x02EA 0x0507 0x061E 0x0509 0x050A 0x0508 0x0239"
    },
    {
      "type": "thermostat",
      "name": "EasyControl/CT200",
      "device id": "0x38",
      "product id": 203,
      "version": "04.02",
      "entities": 1,
      "handlers received": "0x042B 0x047B"
    }
  ]

output of http://ems-esp.local/api/system v3.5.1  (only the Devices part):
  "Devices": [
    {
      "type": "boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 3000-7000iAW/SupraEco/Geo 5xx/WLW196i/WSW196i",
      "device id": "0x08",
      "product id": 172,
      "version": "01.20",
      "entities": 151,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0x049F 0x04A0 0x04A3 0x04AC 0xF7 0x0497 0x2E 0x3B"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "73.04",
      "entities": 51,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x02F5 0x023A 0x0240 0xBB 0x023E",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x02D0 0x02D2 0x031B 0x031D 0x031E",
      "handlers ignored": "0x02E0 0x02EA 0x061E 0x0508 0x0509 0x050A 0x0507 0x0291 0xBF 0x0239"
    },
    {
      "type": "thermostat",
      "name": "EasyControl CT200",
      "device id": "0x38",
      "product id": 203,
      "version": "04.02",
      "entities": 3,
      "handlers received": "0x042B 0x047B"
    }
  ]

### TO REPRODUCE
The same happens after each restart:

### EXPECTED BEHAVIOUR
Get following fields back:
-	hc1 Dauwpunttemperatuur
-	hc1 Relatieve luchtvochtigheid

### SCREENSHOTS
None at the moment.

### ADDITIONAL CONTEXT
None at the moment.

MichaelDvP, 2023-12-09:
Both values are from telegram 0x47B, but in both cases the telegram is received. We have changed dewpoint from offset 0 as UINT to offset 2 with higher resolution, but humidiy is unchanged and should be displayed. Please check the log for this telegram or use telnet with `read 38 47B`
Are the names registered in settings/customization?

nicootje55, 2023-12-10:
In the log i see: 2023-12-10 08:59:51.274 I 23: [emsesp] Recognized new thermostat with deviceID 0x38
But i also see the following errors:
2023-12-10 09:06:00.848 W 38: [telegram] Incomplete Rx: 6C 08 08 90 FF 0C 01 01 40
2023-12-10 09:07:00.752 W 39: [telegram] Incomplete Rx: 6C 08 08 90 FF 0C 01 01 40

telnet   read 38 47b   gives mostly no answer but once i got: 
000+00:24:28.283 W 3: [telegram] Incomplete Rx: 6C 08 08 00 FF 00 03 9F 00 00 00 00 00 00 00 00 07 08 00 00 00 00

telnet   read 42 47b   gives   Invalid DeviceID 
telnet   read 38 47    gives     000+00:25:42.605 N 4: [emsesp] thermostat(0x38) -W-> Me(0x0B), ?(0x47), data: <empty> 

Thera are 5 names registered in settings/customization. 
But only   hc1 Ruimtetemperatuur van afstandsbediening (hc1/remotetemp)   has a value.
The following names have no value:
-	hc1 Ruimtetemperatuur van afstandsbediening (hc1/remotetemp)
-	hc1 Relatieve luchtvochtigheid (hc1/airhumidity)
-	hc1 Offset interne temperatuur (hc1/intoffset)
-	hc1 battery (hc1/battery)

MichaelDvP, 2023-12-10:
Please attach a full og of a few minutes.

nicootje55, 2023-12-10:
[log.txt](https://github.com/emsesp/EMS-ESP32/files/13627434/log.txt)

nicootje55, 2023-12-10:
A little longer log.
[log (2).txt](https://github.com/emsesp/EMS-ESP32/files/13627439/log.2.txt)

nicootje55, 2023-12-10:
Here i have a new log with v3.6.4 for sure.
[log (3).txt](https://github.com/emsesp/EMS-ESP32/files/13627542/log.3.txt)

MichaelDvP, 2023-12-10:
> Incomplete Rx: 0B 0B 88 88 FF FF 00 00 19 19 03 03 9D 9D EA

Funny echos to all telegrams sent by ems-esp, but the destination get's it right and replys. I think maybe a very short bus-cable causing a reflection.

> [emsesp] thermostat(0x38) -W-> thermostat(0x10), RemoteHumidity(0x047B), data: 03 FF

The remote thermostat sends humidity as invalid (no value 0xFF) and no highres dewpoint, only the single byte 03. Seems strange to publish dewpoint of 3 degrees with no valid humidity measurement. What was shown as humidity with older firmware, your first log shows the same values for this telegram?

nicootje55, 2023-12-10:
About the funny echos: 
the bus cable between my self made adapter and the boiler is about 50cm and the cable to the thermostat is about 10 meters. What can i do against the echoes?

about data: 03 FF:
In the Bosch easycontrol CT200 i could disable the humidity sensor. I think it makes it more flexible for me to cool in the summer. On the dashboard uit gives the value -1.

Attached the log of v3.6.2 where humidity is visible.
[_EMS ESP32 issue CT200 v6.3.2 log.txt](https://github.com/emsesp/EMS-ESP32/files/13628956/_EMS.ESP32.issue.CT200.v6.3.2.log.txt)

MichaelDvP, 2023-12-11:
Oh, -1 is an invalid value, you can not have -1% humidity, it was a bug to show it (read as signed instead of unsigned).
Cable length is ok, Bosch recommend > 20cm. Seems you have build a nullmodem and receive the own sending on the adapter.
Check the sending resistors and the bus connection on your board. Sending should draw ~20mA but not result in voltage drop. I think your input (coils/protection) have to high resistance.

nicootje55, 2023-12-11:
I will look to my bus for the echoes later.
It sounds logical that -1% humidity is invalid.
I changed the humidity to on on the CT200 thermostat and now hundity is shown on the dashboard in version 3.6.4.
But the dew-point is not there and should be there because it was also there in version 3.6.2.
I attached a log of version 3.6.4.
In the log i see:
[emsesp] thermostat(0x38) -W-> thermostat(0x10), RemoteHumidity(0x047B), data: 05 2C
The humidity is 44% on the thermostat (0x2C). I think 05 is the dew-point.
[log v3.6.4.txt](https://github.com/emsesp/EMS-ESP32/files/13633810/log.v3.6.4.txt)

MichaelDvP, 2023-12-11:
Yes, dewpoint is missing in 3.6.4, try https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

nicootje55, 2023-12-11:
I tested 3.6.5-dev.4 and both humidity and dewpoint were there.
When i switched humidity off in the EasyControl CT200 humidity disappeared in the dashboard but the dew-point was still there.
That is how it is supposed to work.

Conclusion: this issue is solved. Do i have to close it or are you doing that?

MichaelDvP, 2023-12-11:
> When i switched humidity off in the EasyControl CT200 humidity disappeared in the dashboard but the dew-point was still there.

Yes, we normally show all entities that have a valid value, In this case the value of dewpoint is valid, so it is shown. It's strange, because dewpoint calculation needs valid humidity, but maybe humidity is measured, but not shown. 
>  Do i have to close it or are you doing that?

Better you close.
I'l ldo a PR to add t to official dev.