emsesp/EMS-ESP32 #1494: Subscribed MQTT topics not detecting changes by EMS-ESP

proddy, 2023-12-11:
Found some peculiar behaviour with MQTT and I'm not sure what is causing it.

EMS-ESP is correctly subscribing to the topic `ems-esp/thermostat/#` as shown in the Mosquitto verbose logs. But when I publish to `ems-esp/thermostat/hc1/seltemp` it does not get picked up by EMS-ESP.

I suspect something with the base/basename but can't find it. 

@MichaelDvP have you seen this before?

MichaelDvP, 2023-12-12:
No, have not realized it before, my esp32 with dev.2 (2.12.23) is working good. The S3 with actual software do not call onMessage.
Don't know what it is, made some tests. If i register a callback within MqttSettingsService and call mqtt::onMessage from there (like onMqttConnect, onMqttDisconnect), it works and structure is more clear. I'll make a PR.

proddy, 2023-12-12:
I can confirm this fixes the issue, I'll add to my PR and push to dev. Thanks @MichaelDvP

commit 9b1e3997 (2023-12-12): fixes #1494
closed 2023-12-12 by proddy