emsesp/EMS-ESP32 #1495: Wrong reading and setting of Dhw comfort mode

wzmigrod, 2023-12-11:
The reading and setting of setting of **Dhw comfort mode** is reversed.
When **eco** mode is set on the boiler then on the ems-esp gateway dashboard it is visible as **comfort mode** (see attached screenshots)
Firmware 3.6.4, boiler Junkers GC2300
Switching the mode to **eco** via gateway dashboard result with boiler going into **comfort mode** and vice versa 
[image]
[image]

mmierke, 2023-12-13:
I can confirm taht for the Buderus GB192i.2

MichaelDvP, 2023-12-13:
> I can confirm that for the Buderus GB192i.2

You confirmed 1 day after i commited a fix. Is the issue still present after the fix or do you confirm for a old (unspecified) software?

wzmigrod, 2023-12-13:
I have tested 3.6.5-test.4, Now the reading is OK, the Dhw Comfort mode is Eco both on the boiler as well as on then gateway dashboard. However, if when I switch the mode to comfort there does not seem to be any real action (the mode setting stays at "Eco" value - after refreshing the dashboard) I cannot check the boiler display as I am away from home but there does not seem any action on the boiler when switching to comfort mode - the DHW temperature should increase to the setting but it does not change at all

MichaelDvP, 2023-12-14:
Check if it is fixed in https://github.com/MichaelDvP/EMS-ESP32/releases

mmierke, 2023-12-14:
> > I can confirm that for the Buderus GB192i.2
> 
> You confirmed 1 day after i commited a fix. Is the issue still present after the fix or do you confirm for a old (unspecified) software?

Sorry, I use 3.6.4 in the moment.

wzmigrod, 2023-12-15:
tried [3.6.5-test.4](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test). But switching form Eco mode to high comfort mode via gateway dashboard does not seem to trigger any change on the boiler

MichaelDvP, 2023-12-16:
> tried 3.6.5-test.4

I was a bit lazy to count up version for a single line change. Have you only checked version number, or downloaded and flashed the build?
Please make a log and change the mode first on the thermostat to eco and comfort again and then on the ems-esp. Then i could see what's going wrong.

wzmigrod, 2023-12-16:
Tried the build (in yesterdays comment i inserted the link to the build that I flashed). NB: In my installation there's no thermostat, just the boiler and the gateway. The boiler get's "heating required" signal on one of terminals form the floor heating controller (for the DHW settings this is not really relevant). So I can  switch DHW mode between eco and comfort  is only via the boiler HMI or form the gateway's dashboard. NB. I am accessing the boiler & gateway remotely so I can only test changing setting via gateway and see if there's any effect on the boiler (in comfort mode the DHV temperature should increase to the set temperature). And it does not, When I change the setting to comfort via gateway dashboard no action is triggered on the boiler and after refreshing the dashboard /read entities the DHV mode status is still eco. Anyway, will register the log.

wzmigrod, 2023-12-16:
and the log after the change of DHW mode on the gateway
[emsesp log.txt](https://github.com/emsesp/EMS-ESP32/files/13693877/emsesp.log.txt)

MichaelDvP, 2023-12-17:
Please start log a minute before you change and change the mode on the working interface (boilerUI or bosch interface, whatever is working). Stop the log > 1 minute after the change,

wzmigrod, 2023-12-17:
Then we need to park the topić for 2 weeks as only then I will be on site. Now the only access I have is remote via the gateway

wzmigrod, 2023-12-30:
New log attached.
Now I am on site and I can see that the EMS dashboard shows ECO status of DHW mode no matter what the real mode set on the boiler is.
LOg attached (switching bewteen ECO and comfort mode and vice-versa)
[log.txt](https://github.com/emsesp/EMS-ESP32/files/13799256/log.txt)

MichaelDvP, 2023-12-31:
Can you please use telnet for logging, the weblog event system sometimes skips messages if brower/system/connection is too slow. In your log half of the messages (including the relevant) are missing (check message index).

wzmigrod, 2024-01-01:
Then check this log. If still relvant messages missing then please advise which strings/patterns I should search for in messages
[Ems_esp_log.txt](https://github.com/emsesp/EMS-ESP32/files/13804845/Ems_esp_log.txt)

MichaelDvP, 2024-01-02:
That is much better, all telegrams are listed. I've found these changes:
```
e000+11:17:56.888 N 106: [emsesp] boiler(0x08) -B-> All(0x00), UBAParameterWWPlus(0xEA), data: 00 (offset 26)
e000+11:17:57.068 N 107: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 (offset 12)
...
e000+11:19:24.985 N 187: [emsesp] boiler(0x08) -B-> All(0x00), UBAParameterWWPlus(0xEA), data: 01 (offset 26)
e000+11:19:25.165 N 188: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 (offset 12)
```
The Monitor is not writeable and only shows as state. The parameter should be writeable.
Try in settings to add a custom value like this:
[image]
You can also set one for the monitor (not writeable) and see how it changes.

You did not write how you have changes, i can not know if the changes was eco->comfort or the other way and what 0means and what 1 means.

wzmigrod, 2024-01-02:
[Ems log 201.txt](https://github.com/emsesp/EMS-ESP32/files/13809310/Ems.log.201.txt)
New log change from Eco to comfort and then back
Nb: created a custom value as suggested. 
Now the dashboard shows Eco Mode irrespective of the real Mode of the boiler set on the noiler's UI. Changing the Mode on the gateway works opposite ie settin Eco Mode on the dashboard switches the boile to comfort Mode and vice versa

MichaelDvP, 2024-01-02:
Sorry, i don't understand what is working or not.
Please specify **exactly** what you are setting on which device with which value and what's the result and what is expected.

wzmigrod, 2024-01-02:
OK; once again: changing the mode from the gateway dashboard (via web interface) works but it works opposite to how it should work, i.e. switching the mode from eco to comfort causes the boiler to switch to eco mode (that's visible on boiler's UI); switching from comfort to eco mode causes the boiler to switch to comfort mode
Changing the mode via boiler's UI does not cause any change of the DHW mode visible via gateway dashboard (accessed via web interface) - 
[image]

MichaelDvP, 2024-01-03:
I still don't know how to solve this. the information is confusing. The thread started with *The reading and setting of setting of Dhw comfort mode is reversed.* 
Changing the order of options does not work.  The logs show that the setting in EA,offset 13 does not change at all, but changing the mode on the boiler panel gives values on EA, offset 26 with values 0/1 but since i do not knw what direction you have changed, i don't know what 0  means. I suggested to add a custom value, but there is no feedback if this works and what are the right values.

Last try:
Try in terminal SU mode:
eco: `call system send "0B 08 05 46 AA"`
comfort: `call system send "0B 08 05 46 55"`
i think eco: `call system send "0B 08 EA 1A 01"`
i think comfort `call system send "0B 08 EA 1A 00"`
and report for **each command seperate** what have changed.

wzmigrod, 2024-01-03:
Well...
Apparently sending command from the gateway to the boiler works (but works opposte to what it should be - then I guess it should be easy to get fixed)
Changes on boiler's UI do not trigger any change of mode visible on the gateway's dashboard. For firmware 3.6.4 it has always been "high comfort" no matter what was set on the boiler. Now, with firmware 3.6.5-test.4 it alsways shows ECO
The last log I attached was recorded for the following sequence of changes via boiler's UI: from ECO to COMFORT and then from COMFORT to ECO (wrote that clearly in the previous comment)
Regarding custom entity: it's value shows 1 now; the boiler is in ECO mode; gateway dashboard shows ECO mode (but that it shows all the time). It is not quite clear to me what You would like to  test with thecustom entity - could You be more specific?
I have been a couple of days on site but now for some weeks I can access the gateway remotely (only via web interface) will not be able to use telnet until I am again on site

MichaelDvP, 2024-01-04:
> The last log I attached was recorded for the following sequence of changes via boiler's UI: from ECO to COMFORT and then from COMFORT to ECO (wrote that clearly in the previous comment)

Funny, with a log that contains only a single change.
> Regarding custom entity: it's value shows 1 now; the boiler is in ECO mode

Ahh, ok, now we are getting nearer. **1 for eco mode**. Now i need to know if it changes to 0 if the boiler is in comfort mode.
And next use the entitiy write option in ems-esp dashboard:  write 0, check if the boiler goes to comfort mode, write 1 and check if it goes to eco. 
If this works, test https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test

wzmigrod, 2024-01-04:
Well.. I cannot test as You suggested as I am no longer on site so do not have access  to the boiler UI and the dashboard (gateway web interface) shows ECO all the time
Anyway. What I tested:
Setting the ECO mode on the dashboard (which I know switches the boiler to comfort mode): custom entity changes to 0
Setting the COMFORT mode on the dashboard (which I know switches the boiler to ECO mode): custom entity switches to 1

wzmigrod, 2024-01-04:
Also when switched custom entity to 0 internauci boiler temparature went up approx to dhw setpoibt and then started droppibg when custom entity changed to 1
[image]

MichaelDvP, 2024-01-05:
Ok, then try the test7 build from here.

wzmigrod, 2024-01-05:
tested test7 build to the extent I could. Changing the setting via gateway dashboard and via HA. And boiler reactions are OK; settings done via HA are reflected at the gateway dashboard and vice versa.
The only functionality that I could not test is changing the dhw mode via boiler UI.
I would consider this isssue as ready for closing

proddy, 2024-01-06:
fixed in 3.6.5-dev-8

commit bf95205a (2024-01-06): fix #1495 wwComfort mode
commit efa4f80b (2024-01-06): Merge pull request #1542 from MichaelDvP/dev

fix #1495 and #1536
closed 2024-01-06 by proddy