emsesp/EMS-ESP32 #1498: changes of number.thermostat_hc1_designtemp in HomeAssistant are not processed bij EMS ESP

nicootje55, 2023-12-12:
### PROBLEM DESCRIPTION
In version 3.6.5 dev.3 when i change number.thermostat_hc1_designtemp in HomeAssistant It is not processed bij EMS ESP (not visible in the dashboard).  
Both up and down. 
In v3.6.4 it is working fine.

I saw this behaviour also in 3.6.5 dev.4 (issue #1491), that is the reason i tested it with 3.6.5. dev.3. 

### REQUESTED INFORMATION
I attached a log. From 50 made it up to 52 and down to 48.
[log v3.6.5 dev.3 number issue.txt](https://github.com/emsesp/EMS-ESP32/files/13648600/log.v3.6.5.dev.3.number.issue.txt)

### TO REPRODUCE
Continuous behaviour.

### EXPECTED BEHAVIOUR
Changes in HomeAssistant to be processed by EMS ESP and visible in the dashboard.

MichaelDvP, 2023-12-12:
I think it's the same as #1494, try if one of my latest builds fixes it. https://github.com/MichaelDvP/EMS-ESP32/releases

nicootje55, 2023-12-12:
I just tested with EMS-ESP-3_6_5-test_4-ESP32.bin and can confirm that my problem is gone.
Shall i close this issue?

MichaelDvP, 2023-12-12:
> Shall i close this issue?

Yes, there is also #1494 that can be closed when the official dev is fixed.

nicootje55, 2023-12-13:
Issue closed because #1494 stays open.

closed 2023-12-13 by nicootje55