emsesp/EMS-ESP32 #1516: MQTT: Home assistant discovery - analog sensor invalid dev section

arturzx, 2023-12-19:
### PROBLEM DESCRIPTION

Invalid dev section for analog sensor in HA discovery leads to HA displaying all sensors prefixed with analog sensor name.

### TO REPRODUCE

Add analog sensor.

### EXPECTED BEHAVIOUR

Added analog sensor have separated device ids like others and not leads to change other entities names.

### SCREENSHOTS

[image]
[image]

### ADDITIONAL CONTEXT

Discovery topic value for this sensor
`{
  "stat_t": "ems-esp/analogsensor_data",
  "val_tpl": "{{value_json['05'].value if value_json['05'] is defined and value_json['05'].value is defined else wyłączono}}",
  "obj_id": "analogsensor_05",
  "uniq_id": "analogsensor_05",
  "name": "CWU potrzeba grzania",
  "pl_on": "włączono",
  "pl_off": "wyłączono",
  "dev": {
    "name": "CWU potrzeba grzania",
    "ids": [
      "ems-esp"
    ]
  },
  "avty": [
    {
      "t": "ems-esp/status",
      "val_tpl": "{{'online' if value == 'online' else 'offline'}}"
    },
    {
      "t": "ems-esp/analogsensor_data",
      "val_tpl": "{{'online' if value_json['05'] is defined and value_json['05'].value is defined else 'offline'}}"
    }
  ],
  "avty_mode": "all"
}`

arturzx, 2023-12-19:
Screenshot from HA devices list:
[image]

proddy, 2023-12-28:
See #1528

proddy, 2024-01-03:
will fix with 1528

closed 2024-01-03 by proddy