emsesp/EMS-ESP32 #1520: Data reading problem

stefandamann, 2023-12-21:
Hello,

i get an error when i want to open a device

Unexpected Application Error!
Cannot read properties of null (reading 'data')
TypeError: Cannot read properties of null (reading 'data')
    at b.pP [as constructor] (http://192.168.10.202/assets/index-zC5sSXrP.js:196:203979)
    at b.D [as render] (http://192.168.10.202/assets/index-zC5sSXrP.js:1:9540)
    at I (http://192.168.10.202/assets/index-zC5sSXrP.js:1:7049)
    at E (http://192.168.10.202/assets/index-zC5sSXrP.js:1:2181)

When i restart the ESP network connection is lost and I need to use the Hotspot.

bbqkees, 2024-01-01:
If this happens after a firmware update you need to empty the browser cache or press F5.

stefandamann, 2024-01-01:
Thx that helped

stefandamann, 2024-01-01:
Thx that helped

closed 2024-01-01 by stefandamann