emsesp/EMS-ESP32 #1521: ES79 (Sieger Controller) not found

heilo71, 2023-12-21:
Controller (thermostat) Sieger ES 79 not found

### PROBLEM DESCRIPTION
After Upgrade to Version 3.6.4 the Sieger controller ES79 is unknown device on the ESM bus .

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
2023-12-21 13:27:30.000 INFO 1: [emsesp] Starting EMS-ESP version 3.6.4
2023-12-21 13:27:30.000 INFO 3: [emsesp] Loading board profile S32
2023-12-21 13:27:30.000 INFO 5: [emsesp] Starting Custom entity service
2023-12-21 13:27:30.000 INFO 6: [temperaturesensor] Starting Temperature sensor service
2023-12-21 13:27:30.000 INFO 7: [analogsensor] Starting Analog sensor service
2023-12-21 13:27:30.000 INFO 8: [emsesp] Loaded EMS device library (117 records)
2023-12-21 13:27:32.307 INFO 9: [emsesp] Starting NTP service
2023-12-21 13:27:32.307 INFO 10: [emsesp] WiFi connected with IP=xxxxx, hostname=ems-espBK15
2023-12-21 13:27:32.311 INFO 11: [emsesp] mDNS responder service started
2023-12-21 13:27:32.311 INFO 12: [system] NTP connected
2023-12-21 13:27:37.395 INFO 13: [mqtt] MQTT connected
2023-12-21 13:27:45.529 INFO 14: [emsesp] Recognized new boiler with deviceID 0x08
2023-12-21 13:27:46.150 INFO 15: [emsesp] Recognized new controller with deviceID 0x09
2023-12-21 13:27:46.382 NOTICE 16: [emsesp] Unrecognized EMS device (deviceID 0x10, productID 156). Please report on GitHub.
2023-12-21 14:51:23.605 INFO 17: [emsesp] Scanning devices...

**(Please, remember to close the issue when the problem has been addressed)**

MichaelDvP, 2023-12-21:
Support info please. I think it's RC35 compatible, but need to see the received telegrams, a full log of ~3 minutes is also usefull.

heilo71, 2023-12-22:
Hi Michael,

here are detailed informations.

Greetings from Germany,

Heiko

Diese E-Mail wurde von Avast-Antivirussoftware auf Viren geprüft.
www.avast.com

2023-12-22 11:25:09.000 INFO 1: [emsesp] Starting EMS-ESP version 3.6.4
2023-12-22 11:25:09.000 INFO 2: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
2023-12-22 11:25:09.000 INFO 3: [emsesp] Loading board profile S32
2023-12-22 11:25:09.000 INFO 4: [emsesp] Starting Scheduler service
2023-12-22 11:25:09.000 INFO 5: [emsesp] Starting Custom entity service
2023-12-22 11:25:09.000 INFO 6: [temperaturesensor] Starting Temperature sensor service
2023-12-22 11:25:09.000 INFO 7: [analogsensor] Starting Analog sensor service
2023-12-22 11:25:09.000 INFO 8: [emsesp] Loaded EMS device library (117 records)
2023-12-22 11:25:10.626 TRACE 9: [emsesp] 10(0x10) -B-> All(0x00), ?(0x06), data: 15 0C 0B 15 13 36 01 00
2023-12-22 11:25:11.009 TRACE 10: [emsesp] 08(0x08) -B-> All(0x00), ?(0x18), data: 32 01 C4 64 00 01 01 20 60 80 00 02 09 01 BE 00 00 10 30 59 00 CC 00 00 00
2023-12-22 11:25:11.274 TRACE 11: [emsesp] 08(0x08) -B-> All(0x00), ?(0x34), data: 34 02 09 02 09 21 00 01 03 00 00 12 7F 00 00 D9 00
2023-12-22 11:25:11.289 INFO 12: [emsesp] Starting NTP service
2023-12-22 11:25:11.289 INFO 14: [emsesp] mDNS responder service started
2023-12-22 11:25:11.345 INFO 15: [mqtt] MQTT connected
2023-12-22 11:25:13.520 TRACE 16: [emsesp] 10(0x10) -B-> All(0x00), ?(0x3E), data: 80 02 2B 7D 00 00 00 00 00 00 00 00 64 11 32 00
2023-12-22 11:25:20.970 TRACE 17: [emsesp] 08(0x08) -B-> All(0x00), ?(0x18), data: 32 01 C3 64 00 01 01 20 60 80 00 02 09 01 BB 00 00 10 30 59 00 CC 00 00 00
2023-12-22 11:25:21.210 TRACE 18: [emsesp] 08(0x08) -B-> All(0x00), ?(0x34), data: 34 02 09 02 09 21 00 01 03 00 00 12 7F 00 00 D9 00
2023-12-22 11:25:24.313 TRACE 19: [emsesp] Me(0x0B) -R-> 08(0x08), UBADevices(0x07), length: 0x1B
2023-12-22 11:25:24.341 TRACE 20: [emsesp] 08(0x08) -W-> Me(0x0B), UBADevices(0x07), data: 03 01 00 00 00 00 00 00 00 00 00 00 00
2023-12-22 11:25:24.348 TRACE 21: [emsesp] Me(0x0B) -R-> 08(0x08), Version(0x02), length: 0x1B
2023-12-22 11:25:24.388 TRACE 22: [emsesp] 08(0x08) -W-> Me(0x0B), Version(0x02), data: 40 03 06 00 00 00 00 00 00 00
2023-12-22 11:25:24.388 INFO 23: [emsesp] Recognized new boiler with deviceID 0x08
2023-12-22 11:25:24.388 TRACE 24: [emsesp] boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00
2023-12-22 11:25:24.763 TRACE 25: [emsesp] Me(0x0B) -R-> 09(0x09), Version(0x02), length: 0x1B
2023-12-22 11:25:24.785 INFO 27: [emsesp] Recognized new controller with deviceID 0x09
2023-12-22 11:25:25.113 TRACE 29: [emsesp] 10(0x10) -W-> Me(0x0B), Version(0x02), data: 9C 33 01
2023-12-22 11:25:25.113 NOTICE 30: [emsesp] Unrecognized EMS device (deviceID 0x10, productID 156). Please report on GitHub.
2023-12-22 11:25:25.309 TRACE 31: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAErrorMessage1(0x10), length: 0x1B
2023-12-22 11:25:25.338 TRACE 32: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAErrorMessage1(0x10), data: 36 41 00 E3 95 0A 11 03 2D 00 00 00
2023-12-22 11:25:25.583 TRACE 33: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAErrorMessage2(0x11), length: 0x1B
2023-12-22 11:25:25.737 TRACE 35: [emsesp] unknown(0x10) -W-> boiler(0x08), UBASetPoints(0x1A), data: 32 64 64 00
2023-12-22 11:25:25.957 TRACE 36: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAMaintenanceData(0x15), length: 0x1B
2023-12-22 11:25:25.980 TRACE 37: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMaintenanceData(0x15), data: 00 3C 01 01 04 00 00 00
2023-12-22 11:25:26.281 TRACE 38: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAMaintenanceStatus(0x1C), length: 0x1B
2023-12-22 11:25:26.307 TRACE 39: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMaintenanceStatus(0x1C), data: 94 01 0A 1A 16 00 00 00 00 00 00
2023-12-22 11:25:26.529 TRACE 40: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAErrorMessage3(0xC2), length: 0x1B
2023-12-22 11:25:26.544 TRACE 41: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAErrorMessage3(0xC2), data: <empty>
2023-12-22 11:25:26.754 TRACE 42: [emsesp] Me(0x0B) -R-> boiler(0x08), UBATotalUptime(0x14), length: 0x1B
2023-12-22 11:25:26.771 TRACE 43: [emsesp] boiler(0x08) -W-> Me(0x0B), UBATotalUptime(0x14), data: 12 18 8C
2023-12-22 11:25:27.013 TRACE 44: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameters(0x16), length: 0x1B
2023-12-22 11:25:27.056 TRACE 45: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameters(0x16), data: FF 5A 64 00 06 FA 0A 01 05 64 37 02
2023-12-22 11:25:27.252 TRACE 46: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameterWW(0x33), length: 0x1B
2023-12-22 11:25:27.278 TRACE 47: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 34 FB 00 0F FF 02 46 00 FF
2023-12-22 11:25:27.526 TRACE 48: [emsesp] Me(0x0B) -R-> boiler(0x08), UBASettingsWW(0x26), length: 0x1B
2023-12-22 11:25:27.541 TRACE 49: [emsesp] boiler(0x08) -W-> Me(0x0B), UBASettingsWW(0x26), data: <empty>
2023-12-22 11:25:27.725 TRACE 50: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParametersPlus(0xE6), length: 0x1B
2023-12-22 11:25:27.739 TRACE 51: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParametersPlus(0xE6), data: <empty>
2023-12-22 11:25:28.049 TRACE 52: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameterWWPlus(0xEA), length: 0x1B
2023-12-22 11:25:28.063 TRACE 53: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameterWWPlus(0xEA), data: <empty>
2023-12-22 11:25:28.297 TRACE 54: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAFactory(0x04), length: 0x1B
[... cut ...]
2023-12-22 11:29:11.227 TRACE 209: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 32 01 EA 64 20 09 01 25 60 80 00 02 08 01 E1 00 38 12 2D 48 00 C8 00 02 00
2023-12-22 11:29:11.559 TRACE 210: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameterWWPlus(0xEA), length: 0x1B
2023-12-22 11:29:11.574 TRACE 211: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameterWWPlus(0xEA), data: <empty>
2023-12-22 11:29:12.135 TRACE 212: [emsesp] unknown(0x10) -B-> All(0x00), ?(0x06), data: 15 0C 0B 15 17 37 01 00
2023-12-22 11:29:15.028 TRACE 213: [emsesp] unknown(0x10) -B-> All(0x00), ?(0x3E), data: 80 02 2B 7D 00 00 00 00 00 00 00 00 64 11 32 00

MichaelDvP, 2023-12-22:
> here are detailed informations.
Greetings from Germany,

Thanks for the nice greetings, but i hoped also to see the `support info`.

heilo71, 2023-12-22:
[image]
[image]

heilo71, 2023-12-22:
[log (3).txt](https://github.com/emsesp/EMS-ESP32/files/13751761/log.3.txt)

MichaelDvP, 2023-12-22:
With support info we mean the file you can download from ems-esp help page or via http://ems-esp.local/api/system/info. But in the log i see RC35monitor telegram and pictures show a rc35 similar design. Please try the dev from
https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

commit 8d33a89e (2023-12-22): Merge pull request #1524 from MichaelDvP/dev

add ES79 and fix german translation, fixes #1521, #1522
closed 2024-03-23 by proddy