emsesp/EMS-ESP32 #1527: Stuck in bootloop

henklietaert, 2023-12-25:
### PROBLEM DESCRIPTION

My ESP32-S3 was acting up, so I flashed the latest firmware to it. Now it will no longer boot and gets into a bootloop.
This disconnects the device from the pc every 2 seconds, so I had to be fast to get the output of the log.
Device is the BBQKees S3 version, connected to a MacBook Pro using a USB-C cable.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ x] Provide the output of http://ems-esp.local/api/system : NA as the system doesn't boot...

Output of the flasher tool:
```lua

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c130020,len:0xb11cc
load:0x3fc960c0,len:0x6574
load:0x40374000,len:0x88a8
load:0x42000020,len:0x12c5f0
Invalid image block, can't boot.
ets_main.c 329 

```

### TO REPRODUCE

Just flash the BBQKees board with the latest stable firmware using esphome flasher tool in Chrome.

### EXPECTED BEHAVIOUR

Just boot right up and be ready for configuration.

### ADDITIONAL CONTEXT

Tried the default firmware that is available in the tool, and the ESP boots, makes the AP etc. So the board isn't dead.

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2023-12-25:
which EMS-ESP version did you have before and which firmware .bin file did you use?

I don't know what the "esphome flasher tool in Chrome". EMS-ESP uses different partition settings than esphome and comes with its own flash tool.

Does it go into a boot loop when only connected to the Macbook via USB ?

henklietaert, 2023-12-25:
If I recall I had the 3.6.4, and I had the wifi disconnection issues. So I wanted to set the BSSID, but it would no longer connect to the wifi, so that's why I decided to flash it.
Tool that I used: https://web.esphome.io
Runs in the browser, very easy.

If connected to the bus, nothing happens, no led or anything.

So problem is probably connected to the flash tool that I'm using, even though I provide the correct .bin file?

henklietaert, 2023-12-25:
will try the EMS-ESP flasher tool now, and I will update the issue when done

henklietaert, 2023-12-25:
Works again, learned something new today. Didn't know that the flash tool is also important, thought the bin file was everything.

So when using the ems-esp flasher the device booted right back up, time to set the BSSID this time to prevent the wifi meshing issues.

Thanks for the quick support!

closed 2023-12-25 by henklietaert

proddy, 2023-12-25:
Merry Christmas!🎄