emsesp/EMS-ESP32 #1528: HA "name" consistent

proddy, 2023-12-28:
Maybe a good time for checking the other ems-esp internal devices and make all fit in one scheme. See:
https://github.com/emsesp/EMS-ESP32/blob/08c5fd8f64c19ff7b76b672a2914825d51758381/src/temperaturesensor.cpp#L547
https://github.com/emsesp/EMS-ESP32/blob/08c5fd8f64c19ff7b76b672a2914825d51758381/src/web/WebSchedulerService.cpp#L291
https://github.com/emsesp/EMS-ESP32/blob/08c5fd8f64c19ff7b76b672a2914825d51758381/src/web/WebCustomEntityService.cpp#L430
https://github.com/emsesp/EMS-ESP32/blob/08c5fd8f64c19ff7b76b672a2914825d51758381/src/shower.cpp#L214
https://github.com/emsesp/EMS-ESP32/blob/08c5fd8f64c19ff7b76b672a2914825d51758381/src/shower.cpp#L244

_Originally posted by @MichaelDvP in https://github.com/emsesp/EMS-ESP32/issues/1517#issuecomment-1862404354_

arturzx, 2023-12-30:
One more thing - in shower device name is set to "EMS-ESP Shower", in other devices name is concatenated from Mqtt::basename() and name and this leads to "ems-esp ..." name and shower device is not follow this rule. Maybe all devices should have "EMS-ESP ..."  names?

MichaelDvP, 2023-12-30:
@proddy  Maybe make one subroutine and use it for all devices, something like:
```
void Mqtt::add_ha_dev(JsonObject & dev, const char * name) {
    auto cap_name = strdup(name);
    Helpers::CharToUpperUTF8(cap_name); // capitalize first letter
    dev["name"] = Mqtt::basename() + " " + cap_name;
    free(cap_name);
    JsonArray  ids = dev.createNestedArray("ids");
    ids.add(Mqtt::basename() + "-" + name);
}
```

proddy, 2023-12-30:
yes, or pass the `config` and also create the dev nestedObject

proddy, 2024-01-04:
fixed in 3.6.5.dev-8

closed 2024-01-04 by proddy

proddy, 2024-01-06:
looks like:

[image]