emsesp/EMS-ESP32 #1534: true/false and "true"/"false" are being sent to ioBroker as 1/0

Grmlrckr, 2024-01-03:
### PROBLEM DESCRIPTION

No matter if I select "true"/"false" or true/false from the web interface bool format for MQTT/API, ioBroker receives always 0/1. I would have expected a native true/false.
Still my scripts and automations are working, but as we have 1/0 and "1"/"0" available in the dropdown menu in the EMS-ESP32 as well it is either redundant or actually buggy and it really should send true/false.

on/off ("an/aus" in German) actually work as expected and are sent that way to ioBroker.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the output of http://ems-esp.local/api/system :

```lua
  {
  "System Info": {
    "version": "3.6.2",
    "platform": "ESP32",
    "uptime": "020+19:05:52.422",
    "uptime (seconds)": 1796752,
    "free mem": 136,
    "max alloc": 71,
    "free app": 6581,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "heizung",
    "RSSI": -64,
    "IPv4 address": "192.168.178.31/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "192.168.178.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 569691,
    "MQTT queued": 0,
    "MQTT publish fails": 7,
    "MQTT connects": 4,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 89896,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 1978508,
    "bus reads (tx)": 508965,
    "bus writes (tx)": 37,
    "bus incomplete telegrams": 277,
    "bus reads failed": 94,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "de",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": true,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 4,
    "bool dashboard": 3,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 50
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si",
      "device id": "0x08",
      "product id": 123,
      "version": "04.08",
      "entities": 74,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "11.05",
      "entities": 47,
      "handlers received": "0x06 0xA2 0x02BA 0x02BB 0x02BC 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x04",
      "handlers ignored": "0x35 0x1A 0xBF"
    },
    {
      "type": "controller",
      "name": "BC25",
      "device id": "0x09",
      "product id": 125,
      "version": "01.05",
      "entities": 0
    }
  ]
}

```

### TO REPRODUCE

Tried with all settings in EMS-ESP32 for 0/1, "0"/"1", true/false and "true"/"false", always getting 0 or 1 in ioBroker.

### EXPECTED BEHAVIOUR

Expected to receive variant true/false when selected in EMS-ESP32 and 0/1 when selected that variant. As mentioned before, "on/off" behaves as expected.

### SCREENSHOTS

[image]

ioBroker:
[image]

MichaelDvP, 2024-01-03:
If you are using the ioBroker adapter from @tp1de, you should set the option to 0/1, see https://github.com/tp1de/ioBroker.ems-esp/blob/main/doc/ems-esp-es.pdf
Maybe thomas now converts other inputs to numbers. 
If you are using mqtt it should work with true/false. Also the api gives the right values. To test in brower: http://ems-esp.local/api/boiler/wwactive/value

Grmlrckr, 2024-01-03:
Ok, thanks. Then this is an expected behavior and not a bug.

closed 2024-01-03 by Grmlrckr

tp1de, 2024-01-03:
Just to explain:

The iobroker adapter can read km200 and ems-esp gateway.
Km200 boolean entities are numbers 0/1 with states off/on.
To make ems-esp compatible, the values are transformed to 0/1 as well.