emsesp/EMS-ESP32 #1536: Unknown Device ID (Bosch C1200)

Raybom, 2024-01-04:
### PROBLEM DESCRIPTION

Using custom build, and my device is Bosch C1200. It was working okay for one week and after that it did not work. The RX and TX readings are okay but, device ID is unknown.

### REQUESTED INFORMATION

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
[emsesp_settings (2).json](https://github.com/emsesp/EMS-ESP32/files/13827733/emsesp_settings.2.json)

``
{
  "type": "settings",
  "System": {
    "version": "3.6.4"
  },
  "Network": {
    "ssid": "",
    "bssid": "",
    "password": "",
    "hostname": "ems-esp",
    "static_ip_config": false,
    "enableIPv6": false,
    "bandwidth20": false,
    "tx_power": 20,
    "nosleep": false,
    "enableMDNS": true,
    "enableCORS": false,
    "CORSOrigin": "*"
  },
  "AP": {
    "provision_mode": 1,
    "ssid": "ems-esp",
    "password": "ems-esp-neo",
    "channel": 1,
    "ssid_hidden": false,
    "max_clients": 4,
    "local_ip": "192.168.4.1",
    "gateway_ip": "192.168.4.1",
    "subnet_mask": "255.255.255.0"
  },
  "MQTT": {
    "enabled": false,
    "host": "",
    "port": 1883,
    "base": "ems-esp",
    "username": "",
    "password": "",
    "client_id": "ems-esp",
    "keep_alive": 60,
    "clean_session": false,
    "entity_format": 1,
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "publish_time_heartbeat": 60,
    "mqtt_qos": 0,
    "mqtt_retain": false,
    "ha_enabled": false,
    "nested_format": 1,
    "discovery_prefix": "homeassistant",
    "discovery_type": 0,
    "publish_single": false,
    "publish_single2cmd": false,
    "send_response": false
  },
  "NTP": {
    "enabled": false,
    "server": "time.google.com",
    "tz_label": "Europe/Amsterdam",
    "tz_format": "CET-1CEST,M3.5.0,M10.5.0/3"
  },
  "OTA": {
    "enabled": false,
    "port": 8266,
    "password": "ems-esp-neo"
  },
  "Security": {
    "jwt_secret": "ems-esp-neo",
    "users": [
      {
        "username": "admin",
        "password": "admin",
        "admin": true
      },
      {
        "username": "guest",
        "password": "guest",
        "admin": false
      }
    ]
  },
  "Settings": {
    "version": "3.6.4",
    "locale": "en",
    "tx_mode": 2,
    "ems_bus_id": 11,
    "syslog_enabled": false,
    "syslog_level": 3,
    "trace_raw": false,
    "syslog_mark_interval": 0,
    "syslog_host": "",
    "syslog_port": 514,
    "boiler_heatingoff": false,
    "shower_timer": false,
    "shower_alert": false,
    "shower_alert_coldshot": 10,
    "shower_alert_trigger": 7,
    "rx_gpio": 23,
    "tx_gpio": 5,
    "dallas_gpio": 18,
    "dallas_parasite": false,
    "led_gpio": 2,
    "hide_led": false,
    "low_clock": false,
    "telnet_enabled": true,
    "notoken_api": false,
    "readonly_mode": false,
    "analog_enabled": true,
    "pbutton_gpio": 0,
    "solar_maxflow": 30,
    "board_profile": "CUSTOM",
    "fahrenheit": false,
    "bool_format": 1,
    "bool_dashboard": 1,
    "enum_format": 1,
    "weblog_level": 9,
    "weblog_buffer": 100,
    "weblog_compact": true,
    "phy_type": 0,
    "eth_power": 0,
    "eth_phy_addr": 0,
    "eth_clock_mode": 0,
    "platform": "ESP32"
  }
}

```

### TO REPRODUCE

Connect the device and it does not recognize device ID.

### EXPECTED BEHAVIOUR

is to recognize device ID.

### SCREENSHOTS

_screenshot will be added_

### ADDITIONAL CONTEXT

LOG data from the error: 
[log (3).txt](https://github.com/emsesp/EMS-ESP32/files/13828036/log.3.txt)

I am using custom build, the schematic is here:
_screenshot will be added_

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2024-01-04:
what makes you think it's a software issue? Can you post the System support info (https://emsesp.github.io/docs/Support/)

Looks like like a Tx issue - cabling probably.

See https://emsesp.github.io/docs/Troubleshooting/#not-all-ems-devices-are-recognized

Raybom, 2024-01-04:
> what makes you think it's a software issue? Can you post the System support info (https://emsesp.github.io/docs/Support/)
> 
> 
> 
> Looks like like a Tx issue - cabling probably.
> 
> 
> 
> See https://emsesp.github.io/docs/Troubleshooting/#not-all-ems-devices-are-recognized
> 
> 
> 
> 

I did not change anything and it was Generic Boiler to become unknown. Thats why I thought it is because unrecognized device ID. 

I will check TX settings.

MichaelDvP, 2024-01-04:
You should not send the settings, we need the system info with the device infos. But in this case the log shows the product-id:
`000+00:00:23.815 NOTICE 11: [emsesp] Unrecognized EMS device (deviceID 0x08, productID 12). Please report on GitHub.`
This is actually not in the database, we need to add.
I'll make a testversion and inform when it is online.

proddy, 2024-01-04:
strange that it worked before!

@MichaelDvP note I'm also pushing a PR today with some Domoticz fixes

Raybom, 2024-01-04:
@MichaelDvP Thanks.

@proddy At some point I think if it keeps failing to recognize device ID it becomes unknown. Also I have tried to update firmware, It did not change the situation. I had this issue with 3.6.2, and 3.6.4.

proddy, 2024-01-04:
ok, then Michael's addition should hopefully fix it

MichaelDvP, 2024-01-04:
> strange that it worked before!

I think this was first the missing tx and the fallback to generic boilder with no product-id. With workingg tx the product-id can be read, but is unknown in this case. 

But i can not find infos to "Bosch C1200", is it a heatpump, or a gas boiler, or hybrid? The log only shows a few ems+ telegrams 0xE3-0xE9 used by a lot of different systems.

Raybom, 2024-01-04:
It is gas boiler. I will get system info I soon as I can. I was using EMS+ . This devices has EMS2.0.
The actual name is C1200W, i forgot to add. And you can use CW200 and CW400 room thermostats.

MichaelDvP, 2024-01-04:
Check with test.7 in https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test

proddy, 2024-01-06:
fixed in 3.6.5-dev-8

closed 2024-01-06 by proddy