emsesp/EMS-ESP32 #1540: Wrong parameter descriptions

Marcwa19197, 2024-01-05:
### PROBLEM DESCRIPTION

Described parameter value range incorrect.

### REQUESTED INFORMATION

```lua
  System information output here:
{
  "System Info": {
    "version": "3.6.4",
    "platform": "ESP32",
    "uptime": "001+14:04:01.789",
    "uptime (seconds)": 137041,
    "free mem": 164,
    "max alloc": 103,
    "free app": 6563,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -58,
    "IPv4 address": "192.168.10.103/255.255.255.0",
    "IPv4 gateway": "192.168.10.1",
    "IPv4 nameserver": "192.168.10.104",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "never",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "192.168.10.80",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 71285,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 2,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 5,
    "temperature sensor reads": 137020,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 179022,
    "bus reads (tx)": 25190,
    "bus writes (tx)": 38,
    "bus incomplete telegrams": 5,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "de",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": true,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 6,
    "bool dashboard": 5,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 10,
    "web log buffer": 10
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GB125/GB135/MC10",
      "device id": "0x08",
      "product id": 72,
      "version": "02.07",
      "entities": 75,
      "handlers received": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x34 0x04",
      "handlers fetched": "0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35",
      "handlers ignored": "0x01 0x03 0xFD 0xFE 0x1F 0x20"
    },
    {
      "type": "thermostat",
      "name": "RC35",
      "device id": "0x10",
      "product id": 86,
      "version": "01.08",
      "entities": 57,
      "handlers received": "0xA3 0x06 0xA2 0x12 0x13 0x42 0x48 0x47 0x49 0x4C 0x52 0x51 0x53 0x56 0x5C 0x5B 0x5D 0x60 0x38 0x39",
      "handlers fetched": "0x3E 0x3D 0x3F 0xA5 0x37",
      "handlers ignored": "0x01 0x1A 0xA4 0x35"
    },
    {
      "type": "controller",
      "name": "BC10/RFM20",
      "device id": "0x09",
      "product id": 68,
      "version": "02.03",
      "entities": 0,
      "handlers ignored": "0x29"
    }
  ]
}

```

Im using a GB125-30 with RC35 Thermostat.
The following parameters seems to have a wrong description (in german):

[image]
The highest accepted input in my case is 40 °C. The lowest is 0 °C.

[image]
The highest accepted input here is 0 °C. The lowest is -1 °C. Can you please explain what this parameter is doing? I did some tests with 0 and -1 but i cannot see a real difference... :-/

Thanks!

MichaelDvP, 2024-01-05:
It's just a hint for min max range. The min/max is not always known, than the the range of storage type is shown. You can edit the min/max settings on the settings/customization page.

Marcwa19197, 2024-01-05:
Thanks. I saw it. And also i got an explanation from buderus for the "ww Ausschalttemperaturdifferenz" parameter. Its the limit in addition to the Warmwater-Temperature when the water-heating is stopped.

closed 2024-01-05 by Marcwa19197