emsesp/EMS-ESP32 #1541: restore "hpactivity-states" => missing fields I use and don't know how to replace

bluegaspode, 2024-01-05:
### PROBLEM DESCRIPTION
I installed 3.6.4 but had to revert to 3.6.2 due to removed field

The following commit: 
https://github.com/emsesp/EMS-ESP32/pull/1400/commits/5c0c0675a28f977b9778d540a579c6bc45fa0238

removed the boolean fields: 
  "hpheatingon": [   "hp heating"  ],
  "hpcoolingon": [   "hp cooling"  ],
  "ww.hpwwon": [    "hp dhw"  ],
  "hppoolon": [    "hp pool"  ],

they were replaced with 
"hp4way": "heating & dhw"

Which unfortunately doesn't allow to differentiate anymore, if the heatpump is running for dhw or heating 

As a usecase: I'm using this fields (especially the differentiation of 'hp heating' and 'dwh hp dhw' to visualize the times/frequency the heatpump is working for heating). 
[image]

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{
  "System Info": {
    "version": "3.6.2",
    "platform": "ESP32",
    "uptime": "000+03:14:57.247",
    "uptime (seconds)": 11697,
    "free mem": 90,
    "max alloc": 47,
    "free app": 6581,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "buderus-heatpump-ems-esp",
    "RSSI": -49,
    "IPv4 address": "192.168.1.110/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "253.0.0.0",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": true,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 247,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 37668,
    "bus reads (tx)": 9611,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 3,
    "bus reads failed": 17,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 50
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 3000-7000iAW/SupraEco/Geo 5xx/WLW196i/WSW196i",
      "device id": "0x08",
      "product id": 172,
      "version": "03.03",
      "entities": 150,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0x02D6 0x0489 0x04AC 0x04C0 0xF9 0xF7 0x0497 0x2E 0x3B"
    },
    {
      "type": "thermostat",
      "name": "TR120RF/CR20RF",
      "device id": "0x38",
      "product id": 249,
      "version": "16.01",
      "entities": 3,
      "handlers received": "0x042B 0x047B",
      "handlers pending": "0x0273",
      "handlers ignored": "0x0A57 0xBF"
    },
    {
      "type": "thermostat",
      "name": "Rego 3000/UI800",
      "device id": "0x10",
      "product id": 253,
      "version": "47.06",
      "entities": 57,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E",
      "handlers ignored": "0xF9 0x04C0 0x02E0 0x02EA 0xE7 0x0507 0x0508 0x0509 0x050A 0x061E 0x059D 0xBF"
    },
    {
      "type": "solar",
      "name": "SM100/MS100",
      "device id": "0x30",
      "product id": 163,
      "version": "27.09",
      "entities": 36,
      "handlers received": "0x0362 0x0363 0x0366 0x0364 0x036A",
      "handlers fetched": "0x0358 0x035A 0x0380 0x038E 0x0391",
      "handlers pending": "0x035D 0x035F 0x035C 0x0361",
      "handlers ignored": "0xE8 0x0368 0x03A9 0x03AA 0xBF"
    },
    {
      "type": "gateway",
      "name": "WiFi module",
      "device id": "0x48",
      "product id": 252,
      "version": "07.02",
      "entities": 0,
      "handlers ignored": "0x2040"
    },
    {
      "type": "connect",
      "name": "Wireless sensor base",
      "device id": "0x50",
      "product id": 252,
      "version": "07.02",
      "entities": 0,
      "handlers ignored": "0x1244 0x1245 0x1246 0x1247 0x1248 0x1249 0x124A 0x124B 0x124C 0x124D 0x124E 0x124F 0x1250 0x1251 0x1252 0x1253 0x2040 0xBF"
    }
  ]
}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_
install 3.6.4 

### EXPECTED BEHAVIOUR

the fields   "hpheatingon",  "hpcoolingon",  "ww.hpwwon",  "hppoolon" are restored

proddy, 2024-01-05:
@MichaelDvP can you remember why we removed the unused burner entities from the heatpump/HIU class?

MichaelDvP, 2024-01-06:
Yes i remember, the states were complete redundant. there is hpActivity as enum and each state was also set as bool, see here:
https://github.com/emsesp/EMS-ESP32/blob/e00eb8e64f1926a6eda8075c34afd8cca38a37d6/src/devices/boiler.cpp#L1519-L1524
To save memory we have removed, also because we know more states in activity (`MAKE_ENUM(enum_hpactivity, FL_(none), FL_(heating), FL_(cooling), FL_(hot_water), FL_(pool), FL_(unknown), FL_(defrost))`)
We have not enough memory to add each enum state as extra bool state.
I can not remember why there were these bool states, maybe we first interprete as bitfield with bit0-heating, bit1-cooling and learned about the other enums later and dont want to break.

bluegaspode, 2024-01-06:
OK - that other field should do, if I add some logic on my own to translate back to simple booleans.

Thanks for point out the other ticket, I didn't find it via search (at least didn't deduct from the title that this was the impulse to remove the other fields.

closed 2024-01-06 by bluegaspode

proddy, 2024-01-06:
no worries, it was good mention. we make lots of changes and not always consistent with recording them