emsesp/EMS-ESP32 #1549: Question: change in JSON format (no space between name and value, no CRLF) - is this intentional?

harvey637, 2024-01-08:
Hi,
with 3.6.5-dev8 I recognized a change in the api output:
old output (part of):
```
...
  "uptimecompheating": 296560,^M
  "uptimecompcooling": 1606,^M
  "totalcompstarts": 1013,^M
  "heatingstarts": 995,^M
  "coolingstarts": 18,^M
...
```
new output (also part of):
`...ating":518897,"uptimecompcooling":3539,"totalcompstarts":2047,"heatingstarts":2016,"coolingstarts":31,"nrgcon...`

As I do a very simple regex-expresssion to fetch the value from API into homematic variables my script was broken.

So my simple questions are:
- are you aware of the removal of the _space_ between the name of the "value" and the "value"
(//`"...control": 12345,<crlf>`// into //`"...control":12345,<NO crlf>`//)

- is this intentionell and will stay so (I saw commits around this topic ....)?
- and what about the lineend? It was CR before, now no more lineend, just continuing ...

This is not a real problem, as long as it stays consistent in the future, just a simple change in my script, only a question ....

thank you and
cu
Harvey

proddy, 2024-01-08:
I recently upgraded the Json library so may have inherently introduced this. Sorry. I changed the API output to show the unformatted Json, i.e. as a single stream of text - before it was '[prettified](https://arduinojson.org/v7/api/json/serializejsonpretty/)' into a nice format, which took up much more data. If it's a big problem, I can change it back.

harvey637, 2024-01-08:
Hi proddy,
no problem at all (by now), it just eliminated some values from yesterday until I found it  :-)

My script is already changed to support _no_space_, so FOR ME no need to go back.
Keeping the code/data smaller always sounds good.

Thank you for your QUICK response!!!
cu
Harvey

proddy, 2024-01-09:
I'll leave this open to remind me to update the CHANGELOG

closed 2024-02-14 by proddy