emsesp/EMS-ESP32 #1554: Mixer parameters - telegram RC300Set2 - 0x02CC/0x02CE

proddy, 2024-01-13:
### Discussed in https://github.com/emsesp/EMS-ESP32/discussions/1553

<div type='discussions-op-text'>

<sup>Originally posted by **martinSezman** January 13, 2024</sup>
Hello,

Could you to consider to add extra thermostat RC310 parameters referred to mixer? Details below.

telegram RC300Set2
hc1: 0x02CC
hc2: 0x02CE

offset 0
Mixer on/off
On=0xFF,Off=0x00
Parameter description in Deutsche manual:
Mischer
Ja: Ausgewählter Heizkreis gemischt.
Nein: Ausgewählter Heizkreis ungemischt.

offset 1
Mixer elapsed time [s]
Range 1-60 (1-3C) multiplied by 10
Parameter description in Deutsche manual:
Mischerlaufzeit
10 ...120 ... 600 s: Laufzeit des Mischers im ausgewählten Heizkreis.

offset 2
Mixer increase [K]
Range 0-20 
Parameter description in Deutsche manual:
Mischeranhebung
0 ... 5 ... 20 K: Anhebung der Wärmeerzeugung für Mischer

My installation details:
{
  "System Info": {
    "version": "3.6.4",
    "platform": "ESP32-C3",
    "uptime": "000+01:37:03.233",
    "uptime (seconds)": 5823,
    "free mem": 105,
    "max alloc": 81,
    "free app": 442,
    "reset reason": "Software reset / Software reset"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -39,
    "IPv4 address": "192.168.2.10/255.255.255.0",
    "IPv4 gateway": "192.168.2.1",
    "IPv4 nameserver": "192.168.2.1",
    "BSSID": "set",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "never",
    "AP security": "wpa2",
    "AP ssid": "ESP32-C3-AP-mode"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 2602,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 12062,
    "bus reads (tx)": 3429,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "C3MINI",
    "locale": "en",
    "tx mode": 4,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 100,
    "web log buffer": 100
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si",
      "device id": "0x08",
      "product id": 123,
      "version": "06.08",
      "entities": 68,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE9"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "18.05",
      "entities": 83,
      "handlers received": "0x06 0xA2 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x0291 0x02CE 0x0292 0x0293 0x0294 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x13 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02D0 0x0469 0x02D2 0x046A 0x031E",
      "handlers ignored": "0x35 0x23 0x1A 0x02E2 0x02EC 0xBF"
    },
    {
      "type": "thermostat",
      "name": "RC200/CW100",
      "device id": "0x39",
      "product id": 157,
      "version": "32.02",
      "entities": 1,
      "handlers received": "0x042C",
      "handlers pending": "0x047C",
      "handlers ignored": "0xBF"
    },
    {
      "type": "mixer",
      "name": "MM100",
      "device id": "0x21",
      "product id": 160,
      "version": "24.05",
      "entities": 4,
      "handlers received": "0x02D8",
      "handlers ignored": "0x1E 0x1A 0x0255 0xF9 0xBF 0x02CE 0xF7"
    },
    {
      "type": "controller",
      "name": "BC25",
      "device id": "0x09",
      "product id": 125,
      "version": "03.03",
      "entities": 0
    },
    {
      "type": "controller",
      "name": "M200/RFM200",
      "device id": "0x50",
      "product id": 218,
      "version": "34.04",
      "entities": 0,
      "handlers ignored": "0x06A1 0x06A2 0x06A3 0x06A4 0x0880 0x0881 0x0882 0x0883 0x0884 0x0885 0x0886 0x0887 0x0888 0xBF"
    }
  ]
}

Regards, 
Martin</div>

MichaelDvP, 2024-01-13:
I've added to my test build https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test
The `mixer increase` is added as `flowtempoffset`. The RC35/MM10 have the same function already included, but as thermostat parameter.

MichaelDvP, 2024-01-14:
Corrected the testbuild dev2, the system info shows 0x2CE on device 0x21 (hc2), so 've changed to type-id 0x2CD for hc1, 0x2CE for hc2, etc.

MichaelDvP, 2024-04-27:
Closing this, it's in 3.7-dev, tested and fixed in #1679 to telegrams 0x2CD, 0x2CE, ff

closed 2024-04-27 by MichaelDvP