emsesp/EMS-ESP32 #1557: Extra DHW activated / deactivated

HansRemmerswaal, 2024-01-14:
### PROBLEM DESCRIPTION

On the thermostat on the boiler there is a button to manually trigger "Extra DWH". When you press the button it shows "Extra DWH activated" when you press again it shows "Extra DHW deactivated. When pressing that button I see the boiler.one_time_charging going on/off, so there seems to be a link.

But... when I change in the EMS-ESP the value of this one_time_charging then I so not see anything changing on the menu of the thermostat on the boiler. Some symbols do not disappear.

I captured the following from the EMS-ESP when pressing the button on the thermostat on the boiler.

┌───────────────────────────────────────┐
│ EMS-ESP version 3.6.5-dev.6           │
│ https://github.com/emsesp/EMS-ESP32   │
│                                       │
│ type help to show available commands  │
│ use su to access Admin commands       │
└───────────────────────────────────────┘

ems-esp:$ watch 10 31d
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0x31D

000+01:36:36.007 N 80: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 08 07

**EXTRA DHW ACTIVATED**
000+01:37:05.259 N 81: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 01 00 02
000+01:37:36.220 N 82: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 01 00 02 07

**EXTRA DWH DEACTIVATED**
000+01:38:23.386 N 83: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 08
000+01:38:36.625 N 84: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 08 07

**EXTRA DHW ACTIVATED**
000+01:43:28.176 N 355: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 01 00 02
000+01:43:28.358 N 356: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode(0x02F5), data: FF (offset 11)
000+01:43:28.755 N 357: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x0497), data: 01 90 01 CC 01 CC 01 FE 02 1C
000+01:43:32.206 N 358: [emsesp] boiler(0x08) -R-> thermostat(0x10), RC300Settings(0x0240), length: 0x01 (offset 12)
000+01:43:32.228 N 359: [emsesp] thermostat(0x10) -W-> boiler(0x08), RC300Settings(0x0240), data: FF (offset 12)
000+01:43:36.745 N 360: [emsesp] thermostat(0x10) -W-> boiler(0x08), ?(0xE7), data: 00 00 01
000+01:43:36.794 N 361: [emsesp] thermostat(0x10) -W-> boiler(0x08), ?(0x02E0), data: 01 1D 64 00 01
000+01:43:36.823 N 362: [emsesp] thermostat(0x10) -W-> boiler(0x08), ?(0x02EA), data: 01
000+01:43:36.894 N 363: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 07 B3 00 00 04 00 00 00 00 09 64 37 00 00 00 2E (offset 22)
000+01:43:37.079 N 364: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Floordry(0x0267), data: 00 00
000+01:43:37.282 N 365: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 1D 01 29 FF 64 02 00 02 B2 00 00 01 0C 00 00                                                                                 00 00 01 28 01 28
000+01:43:37.470 N 366: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 02 08 00 00 00 00 00 01 28 64 64 55
000+01:43:37.665 N 367: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 2F 02 40 01 D8 00 00 00 00 41 3C 00 04 00 00 00 00 00 00 00 00 00                                                                                 00 00
000+01:43:37.932 N 368: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x049F), data: 00 00 0A 8C 08 34 00 00 07 08 00 00 07 08
000+01:43:38.106 N 369: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x04A0), data: 0E 10
000+01:43:38.304 N 370: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x04A3), data: 00 01 00 00 00 00 00 00 00 00 00 0B 00 00 00 00 00 00 00 00 08 00 00 00
000+01:43:38.646 N 371: [emsesp] thermostat(0x10) -W-> boiler(0x08), ?(0x061E), data: 01
000+01:43:38.691 N 372: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 01 00 02 07
000+01:43:38.923 N 373: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 00 C5 22 28 1D 00 28 28 05 A0 02 03 03 05 A0 05 A0 00 00 11 01                                                                                 02
000+01:43:39.337 N 374: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0x0508), data: 00 00 00 00 00 00 00 00
000+01:43:39.536 N 375: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0x0509), data: 00 00 00 00 00 00 00 00
000+01:43:39.998 N 376: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0x050A), data: 00 00 00 00 00 00 00 00
000+01:43:40.258 N 377: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0x0507), data: 00 00 00 00 00 00 00 00
000+01:43:41.531 N 378: [emsesp] thermostat(0x38) -W-> thermostat(0x10), RemoteTemp(0x042B), data: 00 C5 07 B3 01
000+01:43:41.569 N 379: [emsesp] thermostat(0x38) -W-> thermostat(0x10), RemoteHumidity(0x047B), data: 09 2E 00 5E
000+01:43:42.155 N 380: [emsesp] boiler(0x08) -R-> thermostat(0x10), RC300Settings(0x0240), length: 0x01 (offset 12)
000+01:43:42.178 N 381: [emsesp] thermostat(0x10) -W-> boiler(0x08), RC300Settings(0x0240), data: FF (offset 12)
000+01:43:42.794 N 382: [emsesp] boiler(0x08) -B-> All(0x00), HpPower(0x048D), data: 01 00 10 08 01 00 00 01 00 00 00 1F 00
000+01:43:43.060 N 383: [emsesp] thermostat(0x10) -W-> boiler(0x08), ?(0xE7), data: 00 00 00
000+01:43:43.098 N 384: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode(0x02F5), data: 00 (offset 11)
000+01:43:43.816 N 385: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x0497), data: 00 32 00 64 00 64 00 96 00 E6

**EXTRA DWH DEACTIVATED**
000+01:43:45.572 N 386: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 08
000+01:43:46.844 N 387: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 1D 01 28 FF 64 02 00 02 B2 00 00 01 0C 00 00                                                                                 00 00 01 28 01 29
000+01:43:47.032 N 388: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 02 08 00 00 00 00 00 01 29 64 64 55
000+01:43:47.228 N 389: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 00 02 40 01 D8 00 00 00 00 41 3C 00 00 00 00 00 00 00 00 00 00 00                                                                                 00 00
000+01:43:47.495 N 390: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x049F), data: 00 00 0A 8C 08 34 00 00 07 08 00 00 07 08
000+01:43:47.669 N 391: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x04A0), data: 0E 10
000+01:43:47.867 N 392: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x04A3), data: 00 01 00 00 00 00 00 00 00 00 00 0B 00 00 00 00 00 00 00 00 08 00 00 00
000+01:43:52.219 N 393: [emsesp] boiler(0x08) -R-> thermostat(0x10), RC300Settings(0x0240), length: 0x01 (offset 12)
000+01:43:52.241 N 394: [emsesp] thermostat(0x10) -W-> boiler(0x08), RC300Settings(0x0240), data: FF (offset 12)
ems-esp:$ watch off
Watching telegrams is off

MichaelDvP, 2024-01-14:
Have you tried to write to wwExtra1?
Just create a custom entitiy for device 10, type 0x031D, offset 0, UINT and write in dashboard 0 or 1.

HansRemmerswaal, 2024-01-14:
Cool, was not aware that this was possible.
But for some reason I can't update the value via the dashboard. There is no edit option and when I click on the value I can't change it.

[image]

[image]

HansRemmerswaal, 2024-01-14:
Cool, was not aware that this was possible.

[image]

ems-esp:$
000+03:28:31.728 N 11: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300WWmode2(0x031D), length: 0x01
000+03:28:31.751 N 12: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300WWmode2(0x031D), data: 00
**000+03:28:35.115 I 13: [command] Calling command 'custom/wwwExtra1' (set custom value on ems) with value 1**
000+03:28:35.178 N 14: [emsesp] Me(0x0B) -W-> thermostat(0x10), RC300WWmode2(0x031D), data: 01
000+03:29:01.634 N 15: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: **00 00 08 07**

This looks different then triggered from the menu on the boiler. Also nothing is happening on the menu.

000+03:29:31.429 N 16: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300WWmode2(0x031D), length: 0x01
000+03:29:31.456 N 17: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300WWmode2(0x031D), data: 00
000+03:30:01.715 N 18: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 08 07
000+03:30:31.005 N 19: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300WWmode2(0x031D), length: 0x01
000+03:30:31.028 N 20: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300WWmode2(0x031D), data: 00
000+03:31:01.750 N 21: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 08 07
**000+03:31:14.340 N 22: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 01 00 02
000+03:31:23.076 N 23: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 08**
000+03:31:31.780 N 24: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300WWmode2(0x031D), length: 0x01
000+03:31:31.804 N 25: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300WWmode2(0x031D), data: 00
000+03:32:01.666 N 26: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 08 07

HansRemmerswaal, 2024-01-14:
Do I need to read this as Me(0x0B) telling the thermostat the value should be 01 and that the thermostat(0x10) is telling me that I'm wrong and that the value is 00?

000+04:05:09.298 I 126: [command] Calling command 'custom/wwwExtra1' (set custom value on ems) with value 1
000+04:05:09.454 N 127: [emsesp] Me(0x0B) -W-> thermostat(0x10), RC300WWmode2(0x031D), **data: 01**
000+04:05:31.074 N 128: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300WWmode2(0x031D), length: 0x01
000+04:05:31.114 N 129: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300WWmode2(0x031D), **data: 00**

MichaelDvP, 2024-01-14:
There was an earlier thread about this, see #140. You have to use thermostat  `wwcharge` to trigger the onetime preparation (0x2F5, offset 11).

HansRemmerswaal, 2024-01-14:
Ah yes, that one is doing the trick. We can close the issue.

For my interest, what then was I capturing in the log?

MichaelDvP, 2024-01-14:
The `wwextra` seems to report that an extra charge was activated. We see often redundant states in different telegrams (or have not find the difference).

HansRemmerswaal, 2024-01-14:
Ok thanks for your help, will close this issue.

closed 2024-01-14 by HansRemmerswaal