emsesp/EMS-ESP32 #1563: Pumpensolldruckwert (Bosch CS5800i AW) missing

timostark, 2024-01-19:
### PROBLEM DESCRIPTION

The setting for the "Pumpensolldruckwert" (differencial pressure of the heat pump - sorry do not know the english word) is not available by default but can be added via custom entity.

Additionally the UOM of this setting (mbar) is not available in custom entities.

### EXPECTED BEHAVIOUR

Telegram is available (see screenshot of custom entity below). I tested changing the value on my heat pump and I noted the following telegram beeing responsible for changing the value:

000+03:28:27.248 N 12: [emsesp] boiler(0x08) -B-> All(0x00), RC300Set2(0x02CC), data: 04 (offset 9)

(entity: 0x8, type: 0x02CC, Offset: 9, UINT).

The factor is 50, the UoM is mbar. When working on the heatpump the minimum value is 150mbar, the maximum 700mbar (so without the factor: 3 to 14).

### SCREENSHOTS

[image]

[image]

After changing on the heatpump:
[image]

Change is reflected in ESP:
[image]

proddy, 2024-01-19:
nice! I'm really happy we added the Custom Entity feature

timostark, 2024-01-19:
ya that is an absolutly great feature.. thanks for that! Small question: When writing back to custom entities is the factor taken into account? I hope that the heatpump does a verification, but I don't want to put e.g. 200*50 mbar as a goal of the pressure on the heatpump :)

MichaelDvP, 2024-01-19:
Yes, on writing the factor becomes a divider. 
You can also use factor `0.05` and UOM `bar`, the values are rendered with 2 decimals, e.g. 0.15 bar to 0.75 bar.

MichaelDvP, 2024-01-19:
The telegram 0x2CC is confusing.
In #508 it contains hc dependent the dhw priority on thermostat, 0x2CC  for hc1, 0x2CE for hc2, not in a row.
In #1554 it contains hc dependent some mixer settings, also 0x2CC for hc1, 0x2CE for hc2
Now the boiler/heatpump has also a telegram 0x2CC,
I think we should check 0x2CC on other devices.
Very uncommon to use the same telegram number for different functions.

timostark, 2024-01-19:
Addition: When changing the value manuell I actually see the following two requests:

000+03:28:49.904 N 30: [emsesp] thermostat(0x10) -W-> boiler(0x08), RC300Set2(0x02CC), data: 05 (offset 9)
000+03:28:49.962 N 31: [emsesp] boiler(0x08) -B-> All(0x00), RC300Set2(0x02CC), data: 05 (offset 9)

so basically once from the thermostat to the boiler (same telegram 02CC) and once from the boiler to all..

Would you normally write on the thermostat or the boiler in that case?

MichaelDvP, 2024-01-19:
> 00+03:28:49.904 N 30: [emsesp] thermostat(0x10) -W-> boiler(0x08), RC300Set2(0x02CC), data: 05 (offset 9)

Yes, that's a problem. We actually only check if the telegram is registered on the device and set the values. 
Luckily for now the 0x2CC is for mixer offset 0, 1, 2, for thermostat offset 3 and for boiler offset 9, the other offsets are not processed. 
But we need to add logic how to process these telegrams and to avoid conflicts.

@proddy I think of these rules to process:
- src broadcast
- src write to ems-esp (reply to request)
- dest, only if src is not ems-esp

proddy, 2024-01-20:
> @proddy I think of these rules to process:
> 
> * src broadcast
> * src write to ems-esp (reply to request)
> * dest, only if src is not ems-esp

That feels like the right priority order. It's hard to think what could be affected and I'm sure there may be some edge cases to work through later. Let's try this out in dev and see what the feedback is.

MichaelDvP, 2024-01-20:
Ok, i've changed the telegram mapping and added the differential pressure with UOM mbar. 
So @timostark can check the mapping in log.

eaxus90, 2024-01-20:
Many thanks for that! Unfortunately, the custom entity is not working for me. Although it is displayed in the EMS-ESP, it does not work (HA says "unknown").

proddy, 2024-01-20:
> Many thanks for that! Unfortunately, the custom entity is not working for me. Although it is displayed in the EMS-ESP, it does not work (HA says "unknown").

you'll need to provide more details. What does HA's log say? Do you see it in the Settings->Devices&Services->MQTT?

eaxus90, 2024-01-20:
I'm using the latest version of Home Assistant. 
Here is the log: 2024-01-20 15:36:46.814 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.custom_pumpdiffpress: 350.0 (range 0.0 - 100.0)

it seems to be an invalid value.

MichaelDvP, 2024-01-20:
The custom entities do not have a min/max setting. The 0-100 seems to be a HA default? When scaling to bar it should work. 
For the new `boiler/hpsetdiffpress` the min/max can be set in customizations, but i have also set default to 150-750mbar in the new PR.

timostark, 2024-02-01:
As implemented i will close that one.

closed 2024-02-01 by timostark