emsesp/EMS-ESP32 #1565: API get single values (api_data) returns NULL

MichaelDvP, 2024-01-19:
With the ArduinoJson v7 the following sequence outputs null to api_data:
```
                    JsonVariant data = json[attribute_s];
                    output.clear();
                    output["api_data"] = data;
```
because the JsonVariant is cleared too.
Changing to 
```
                    String data = json[attribute_s].as<String>();
                    output.clear();
                    output["api_data"] = data;
```
works, because the string is stored separate. I think we always send the `api_data` as string to web and console.  Do i miss something? I can make a PR.

proddy, 2024-01-19:
no you're correct, that's the only piece I didn't change

proddy, 2024-01-19:
There's also a bug fix in ArduinoJson 7.0.2 today for document.clear() but not sure if its related. 

https://github.com/bblanchon/ArduinoJson/blob/v7.0.2/CHANGELOG.md

MichaelDvP, 2024-01-20:
Tested with 7.0.2, but same issue. Seems the JsonVariant is now only a reference to the json document and is deleted on doc.clear(). 
I've changed to String and make a PR.

closed 2024-01-20 by MichaelDvP

proddy, 2024-01-20:
@MichaelDvP yes, JsonVariant is also a reference. I should have spotted this as I had asked before - https://github.com/bblanchon/ArduinoJson/discussions/2023#discussioncomment-8046413

commit 4d1a428a (2024-01-20): fix #1565, use String for output a single api_data
commit 549a0302 (2024-01-20): Merge pull request #1568 from MichaelDvP/dev

fix #1565 and workaround for #1564