emsesp/EMS-ESP32 #1576: Missing virtual RC100H in 3.6.5-dev.10

Nxtway, 2024-01-22:
### PROBLEM DESCRIPTION

I upgrade from 3.6.5-dev.9 to 3.6.5-dev.10, After upgrade in the ems-esp control panel the virtual RC100H is show, but without any entry. I miss the temperature, humidity and the thrid entry. After downgrade to 3.6.5-dev.9 to entries are shown and again upgrade to 3.6.5-dev.10 they are missing (meening the issue is reproducible).
[image]

Attached are the logfiles with the working functionality in 3.6.5-dev.9 and same log with the issue in 3.6.5-dev.10
[log_3.6.5-dev.9.txt](https://github.com/emsesp/EMS-ESP32/files/14008931/log_3.6.5-dev.9.txt)
[log_3.6.5-dev.10.txt](https://github.com/emsesp/EMS-ESP32/files/14008933/log_3.6.5-dev.10.txt)

### REQUESTED INFORMATION

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

harvey637, 2024-01-23:
hi,
same reason I guess;
also humidity and dewpoint is missing in latest builds (....-10)
cu
Harvey

MichaelDvP, 2024-01-23:
Please try with https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test

Nxtway, 2024-01-23:
Also in the 3.6.5-test.10 the RC100H node has no entries. 
But yes the entries temperature + humidity are moved into the RC310 node.

[image]

MichaelDvP, 2024-01-24:
Sorry, here was a typo,now it should show up again.

MichaelDvP, 2024-01-29:
The test and dev builds are different in some changes, It takes some time to add verified changes from test to dev.

Nxtway, 2024-02-04:
Thanks, the issue is now also fixed in dev build 3.6.5-dev.12

closed 2024-02-04 by proddy