emsesp/EMS-ESP32 #1580: absolut keine Kommunikation vom EMS-ESP zu meiner Buderus GB112-19 Heizung

LMTom, 2024-01-27:
Hallo,
ich habe wLan, MQTT und NTP wie in der Anleitung eingereicht. 
Version: 3.6.4 (ESP32-S3)
Das EMS-ESP kann ich auch ohne Probleme in meinem Netzwerk bedienen, aber es verbindet sich nicht mit meiner GB112 Heizung,
"EMS-Bus getrennt. Wenn diese Warnung nach einigen Sekunden immer noch besteht, überprüfen Sie bitte die Einstellungen und das Board-Profil"
Sys-Log:
2024-01-27 08:10:51.000    INFO 1:      [emsesp]     Starting EMS-ESP version 3.6.4
2024-01-27 08:10:51.000    INFO 2:      [emsesp]     Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
2024-01-27 08:10:51.000    INFO 3:      [emsesp]     Loading board profile S32
2024-01-27 08:10:51.000    INFO 4:      [emsesp]     Starting Scheduler service
2024-01-27 08:10:51.000    INFO 5:      [emsesp]     Starting Custom entity service
2024-01-27 08:10:51.000 WARNING 6:      [emsesp]     Invalid UART Rx/Tx GPIOs. Check config.
2024-01-27 08:10:51.000    INFO 7:      [temperaturesensor] Starting Temperature sensor service
2024-01-27 08:10:51.000    INFO 8:      [analogsensor] Starting Analog sensor service
2024-01-27 08:10:51.000    INFO 9:      [emsesp]     Loaded EMS device library (117 records)
2024-01-27 08:10:53.110    INFO 10:     [emsesp]     Starting NTP service
2024-01-27 08:10:53.114    INFO 11:     [emsesp]     WiFi connected with IP=192.168.178.70, hostname=ems-esp
2024-01-27 08:10:53.141    INFO 12:     [emsesp]     mDNS responder service started
2024-01-27 08:10:53.150    INFO 13:     [mqtt]       MQTT connected

EMS-Telegramme empfangen (Rx) | 0 | 0 |  
EMS-Telegramme gelesen (Tx) | 0 | 0 |  
EMS-Telegramme geschrieben (Tx) | 0 | 0 |  
Temperatursensoren gelesen | 0 | 0 |  
Analogsensoren gelesen | 0 | 0 |  
MQTT-Nachrichten gesendet | 15 | 0 | 100%
API-Aufrufe | 0 | 0

Elektrisch ist das EMS-ESP Verbunden, ich kann mit meinem Multimeter Spannung an den grünen Klemmen messen, mit dem mitgelieferten Kabel funktioniert auch nichts.

Was kann ich tun?
Besten Dank
Grüsse
Tom

proddy, 2024-01-27:
`2024-01-27 08:10:51.000 WARNING 6: [emsesp] Invalid UART Rx/Tx GPIOs. Check config.`

start there, and send your system info (see https://emsesp.github.io/docs/Support/)

bbqkees, 2024-01-29:
I already replied on the same question you send me via email.

Your boiler is the GB112, which is not supported as it has no EMS bus, but the older iRT bus.
You cannot use an EMS Gateway with these iRT boilers.

closed 2024-02-03 by proddy