emsesp/EMS-ESP32 #1581: Missing Parameter "Mischeranhebung"

kriger1965, 2024-01-27:
Please take a few minutes to complete the requested information below.

--> Probably some entidy missing. 

### As displayed I can not find the Parameter for Hc1 Mixer ...temp ... offset or named in RC 310 "Mischeranhebung" ....from 0K ...20K
[image]

[image]

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system

MichaelDvP, 2024-01-27:
Should be in the dev build, added for #1554, commit https://github.com/emsesp/EMS-ESP32/commit/9260db330eb54f1ac505f9657f909a7d43b6220d

kriger1965, 2024-01-27:
Shoud I install the 3.6.5 - dev 10?

proddy, 2024-01-27:
> 9260db3

The dev2 (test) build - https://github.com/emsesp/EMS-ESP32/releases/tag/test

MichaelDvP, 2024-01-28:
It's commited to dev and dev2, but i think not working. We need to fetch the telegram to update the values.
Try this test.10: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test
and give feedback.

kriger1965, 2024-01-28:
I have just installed the new firmware:  EMS-ESP-3_6_5-test_10-ESP32.bin  ...but nothing changed. I cannot see the mixer Parameters , ...

MichaelDvP, 2024-01-28:
It's a bit confusing with test.10 versions, sorry (i havn't count up with the latest changes), download again from my link and try.

kriger1965, 2024-01-29:
...tested yesterday  - no changes detected, missing mixer parameters from GB162/RC310

MichaelDvP, 2024-01-29:
Hmm, maybe in #1554 is something wrong or differs from your setup.
Please log all, then go to the heating and change the settings of Mischanhebung, Mischerlaufzeit, Frostschutz, one by one and back to old values. Wait >10sec before changing each value to make sure we get the telegram on the bus. 
Post the log here and mentions the values you have set. I'll try to identify the setting-telegrams.

proddy, 2024-03-24:
closing due to inactivity. Please re-open if there is something we can help with.

closed 2024-03-24 by proddy