emsesp/EMS-ESP32 #1582: No Ethernet link/connection on Gateway V2

SSSlee, 2024-01-28:
Scenario: 
-first of all, I start with the configuation of a connection to my WIFI access point.

-Later on I took the gatway to the basement. I connected it to the ethernert switch because the access point isn't available on the 
 basement. 

-There was no link or traffic and the gateway establishes the own access point mode.

-A reset while the ethernet port stayed connect doesn't solve the problem.

A factory reset was needed to get the Ethernet interface up. I believe it's because the WiFi connfiguration was removed with the facotory reset.

proddy, 2024-01-28:
version 3.6.4 I presume? Which board profile were you using? I'll try and reproduce.

SSSlee, 2024-01-28:
ah, sorry I missed to attach the required information
[emsesp_info.json](https://github.com/emsesp/EMS-ESP32/files/14076115/emsesp_info.json)

MichaelDvP, 2024-01-28:
Ethernet is disabled if a SSID is set, two ip stacks needs too much ram. 
Just clear the SSID in network settings (Helptext: `leave blank to disable WiFi and enable ETH`)

closed 2024-01-28 by SSSlee

proddy, 2024-02-03:
I've mentioned this in the doc.