emsesp/EMS-ESP32 #1584: custom entity doesn't work

SSSlee, 2024-01-28:
I've tried to configure a custom entity but it doesn't show a value at the end: 

[image]

the terminal log shows the request to the controller 0x50 and the corresponding answer: 

[image]

..but there is no update of the entity: 
[image]

proddy, 2024-01-28:
type ID in the custom entity is 0x1 and the log is reading 0xD1, odd?

SSSlee, 2024-01-28:
Type Id should be 0xD1 , as requested in the log:

[image]

I'm wondering because with verson 3.6.4 it was identified as UBAOutdoorTemp, which is correct: 

[image]

with the update to 3.6.5-dev.10 the type now  seems to be unknown: 

[image]

(both are telnet logs)

MichaelDvP, 2024-01-28:
It works correctly, but i think you like to read a 16 bit value (type SHORT). 

The name of the telegram was taken from any device where we can find it, but now we have same telegram type for different devices with different names, so we check if the telegram is registered in the sending device. We have no registered telegram name on device 0x50.

SSSlee, 2024-01-28:
You are right. I forgot that the ESP32 has 32 bit, sorry for that. With datatype short it works now. 

I'm not shure if I understand your statement right but it brings me to my next question: why the telegram from device 0x50 isn't registered. A custom entity seems to be a workarround? 

My system-info registered device 0x50, but the telegram 0xD1 is set to ignore
[emsesp_info.json](https://github.com/emsesp/EMS-ESP32/files/14076223/emsesp_info.json)

Is there a way to support you to add this device?

MichaelDvP, 2024-01-28:
Can you explain what this device do. I thnk RFM200 is a wireless base for <unknown>
We have telegrams `0x0880 0xD1 0x0889 0xF7 0xBF 0x087F` from it, 0xF7 is menu config, 0xBF error  message, the others are unknown (0xD1 is outdoortemp on boilers).
Can you read these telegrams ( `read 50 880`, etc). to see how much and what info are in these telegrams. Can you identify values?
Then we can add (and name) telegrams and entities, and maybe add a new device category.

BTW: what is the device-id 0x16? Could this be a wirless outdoor sensor T1RF?

SSSlee, 2024-01-28:
Yes, I get the outdoor temperature by a T1RF via the RFM200. In my configuration there is no further thermostat connected to the RFM200

I understand from your previous post that the unknown telegrams are (0x0880), (0x0889) and (0x087F).  (0x087F) seems to be empty. It remains: 

(0x0880), data: 01 04
(0x0889), data: 00 80 80 01

Are there any hints on how I can interpret this data?

Below the entire result of the readout: 

>  **ems-esp:$ read 0x50 0x880**
> 000+01:48:25.342 N 580: [emsesp] controller(0x50) -W-> Me(0x0B), ?**(0x0880), data: 01 04**
> **ems-esp:$ read 0x50 0xD1**
> 000+01:48:36.203 N 581: [emsesp] controller(0x50) -W-> Me(0x0B), ?(0xD1), data: 00 5F
> **ems-esp:$ read 0x50 0x0889**
> 000+01:48:49.095 N 582: [emsesp] controller(0x50) -W-> Me(0x0B), ?**(0x0889), data: 00 80 80 01**
> **ems-esp:$ read 0x50 0xF7**
> 000+01:49:07.402 N 583: [emsesp] controller(0x50) -W-> Me(0x0B), ?(0xF7), data: <empty>
> **ems-esp:$ read 0x50 0xF7**
> 000+01:49:11.053 N 584: [emsesp] controller(0x50) -W-> Me(0x0B), ?(0xF7), data: <empty>
> **ems-esp:$ read 0x50 0xF7**
> 000+01:49:16.634 N 585: [emsesp] controller(0x50) -W-> Me(0x0B), ?(0xF7), data: 00 00 00 00
> **ems-esp:$ read 0x50 0xBF**
> 000+01:49:30.047 N 586: [emsesp] controller(0x50) -W-> Me(0x0B), ?(0xBF), data: 50 DA 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> **ems-esp:$ read 0x50 0x087F**
> 000+01:49:52.664 N 587: [emsesp] controller(0x50) -W-> Me(0x0B), ?(0x087F), **data: 00 00**
> **ems-esp:$ read 0x50 0x087F**
> 000+01:49:57.084 N 588: [emsesp] controller(0x50) -W-> Me(0x0B), ?(0x087F), data: 00 00
> **ems-esp:$ read 0x50 0x087F**
> 000+01:49:59.438 N 589: [emsesp] controller(0x50) -W-> Me(0x0B), ?(0x087F), data: 00 00
>

MichaelDvP, 2024-01-29:
I would expect something like RF strength (connection quality) and battery status. Do you have any other values from ths devcie in the RC300 menu that can match? If check after some time, is any value changing? The 80 could be a `no sensor` or a bitfield.
(0x0880), data: 01 04
(0x0889), data: 00 80 80 01
(0x087F)  data: 00 00

The 0xBF is error status for device 0x50, product 0xDA, no error, 
0xF7 is a config telegram to enable/disable menu entires in the RC300 UI, can not be read from device is this way.

Could you check the complete version info for the devices RFM200 `read 50 2` and T1RF `read 16 2` Could be helpfull if we want to emulate this device combination. Also a log to see if the outdoortemp is broadcasted or only read by your custom entity. 

I think we should reorder the devices: list RFM200 as connect module with outdoortemperature as entity and rename the 0x16 product-id 220 to `T1RF` as extension module without entities. Is this ok?

SSSlee, 2024-01-29:
>ems-esp:$ read 16 2
000+03:48:52.961 N 3: [emsesp] controller(0x16) -W-> Me(0x0B), Version(0x02), data: DC 01 05

( DC matches with the product id & 01 05 seems to be the Version ) 
### from the info file:
{
      "type": "controller",
      "name": "BC30",
      "device id": "0x16",
      "product id": 220,
      "version": "01.05",
      "entities": 0
    },

>ems-esp:$ read 50 2
000+03:48:32.574 N 1: [emsesp] controller(0x50) -W-> Me(0x0B), Version(0x02), data: DA 22 04 00 FF 00 00 00 00

( DA matches with the product id, while 22 04 seems to be the version "34.04" )
### from the info file:
 {
      "type": "controller",
      "name": "M200/RFM200",
      "device id": "0x50",
      "product id": 218,
      "version": "34.04",
      "entities": 0,
      "handlers ignored": "0x0880 0xD1 0x0889 0xF7 0xBF 0x087F"
    }
### 

The outdoortemp is broadcasted by the RFM200 itself, every minute or if the value has changed:
[image]

content of the registers hasn't changed until yet: 
(0x0880), data: 01 04
(0x0889), data: 00 80 80 01
(0x087F) data: 00 00 

> I think we should reorder the devices: list RFM200 as connect module with outdoortemperature as entity and rename the 0x16 product-id 220 to T1RF as extension module without entities. Is this ok?

I'm not sure how to do but I noticed that the RFM200 isn't registred in the dashboard: 

[image]

SSSlee, 2024-01-29:
With Bt you refer to bluetooth? Maybe it's because this modul is based on Zigbee. 

I guess this RF basestation would have an enlarged communication set if a HM200Rf thermostat would be connected instead of this simple outdoor sensor

MichaelDvP, 2024-01-30:
Th Bt is a missing "u" in "But", sorry for the confusion.
I've changed devices in my testbuild, please check: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test

SSSlee, 2024-01-30:
> Th Bt is a missing "u" in "But", sorry for the confusion. I've changed devices in my testbuild, please check: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test

ok I've installed the test version and the device now is shown as described earlier by you. To be honest I don't know what to do now. Can you please give me a hint what to do?
The only thing I've observed is that the Log shows the device as connect(0x50). I would assume that it's named M200/RFM200 or similar.
[image]

MichaelDvP, 2024-01-31:
> To be honest I don't know what to do now. Can you please give me a hint what to do?

Check that dashboard and system/info show the devices correct with the outdoortemp as entity. Is this mapping ok for you, or is there a better mapping?
If all is ok, close the issue.

Log shows only device category, some device-names are very long, not good to read, e.g.
`thermostat(0x10) -w-> boiler(0x08)`  becomes:
`RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410(0x10) -w-> Enviline/Compress 6000AW/Hybrid 3000-7000iAW/SupraEco/Geo 5xx/WLW196i/WSW196i(0x08)`

proddy, 2024-02-22:
closing. please re-open if its still an issue.

closed 2024-02-22 by proddy