emsesp/EMS-ESP32 #1586: not able to set thermostat/datetime

SSSlee, 2024-01-28:
I tried several types of formats to change date and time of my thermostat but all of them result an error messesage because of their length: 

[image]

results: 

[image]

I'm not sure if that's a bug or it's my fault at the end

MichaelDvP, 2024-01-28:
As mentioned in the helptext of the input box: 
`< NTP | dd.mm.yyyy-hh:mm:ss-day(0-6)-dst(0/1) >`
The thermostat needs a setting of day-of-week and daylight-saving. Bosch day-of-week is Mo-0, Su-6, not like unix-time
If you have enabled NTP you can just enter "NTP" and the ntp time is set to the thermostat. 
With ntp enabled the thermostat clock is also automaticly set by ems-esp if it differs from ntp.

SSSlee, 2024-01-28:
Great, works as described by you:

[image]

closed 2024-01-28 by SSSlee

proddy, 2024-02-03:
I've added some help text to the documentation