emsesp/EMS-ESP32 #1587: MQTT discovery of custom entity  doesn't consider type of data

SSSlee, 2024-01-28:
The datatype of a customized entity isn't mentioned in the belonging MQTT Discovery:

[image]

### Corresponding MQTT discovery: 

{
  "stat_t": "ems-esp/custom_data",
  "val_tpl": "{{value_json['RFM200'] if value_json['RFM200'] is defined}}",
  "obj_id": "custom_RFM200",
  "uniq_id": "custom_RFM200",
  "name": "RFM200",
  "dev": {
    "name": "ems-esp Custom",
    "mf": "EMS-ESP",
    "mdl": "Custom",
    "via_device": "ems-esp",
    "ids": [
      "ems-esp-custom"
    ]
  }
  "avty": [
    {
      "t": "ems-esp/status",
      "val_tpl": "{{'online' if value == 'online' else 'offline'}}"
    },
    {
      "t": "ems-esp/custom_data",
      "val_tpl": "{{'online' if value_json['RFM200'] is defined else 'offline'}}"
    }
  ],
  "avty_mode": "all"
}`

### Discovery for a temperature datatype (in comparsion): 

{
  "uniq_id": "thermostat_dampedoutdoortemp",
  "obj_id": "thermostat_dampedoutdoortemp",
  "stat_t": "ems-esp/thermostat_data",
  "name": "Damped outdoor temperature",
  **"unit_of_meas": "°C",**
  "val_tpl": "{{value_json.dampedoutdoortemp if value_json.dampedoutdoortemp is defined else 0}}",
  **"stat_cla": "measurement",
  "dev_cla": "temperature",**
  "dev": {
    "ids": [
      "ems-esp-thermostat"
    ],
    "name": "ems-esp Thermostat"
  },
  "avty": [
    {
      "t": "ems-esp/status",
      "val_tpl": "{{'online' if value == 'online' else 'offline'}}"
    },
    {
      "t": "ems-esp/thermostat_data",
      "val_tpl": "{{'online' if value_json.dampedoutdoortemp is defined else 'offline'}}"
    }
  ],
  "avty_mode": "all"
}

proddy, 2024-01-28:
looks like we need to add both "unit_of_meas" and the "stat_cla"

proddy, 2024-02-03:
@SSSlee if you add those two manually to the discovery topic (using MQTT Explorer) does it fix your issue? Worth trying before I change the EMS-ESP code and unintentionally break something else.

proddy, 2024-02-10:
fixed in dev13

closed 2024-02-11 by proddy

mattblonski, 2024-04-24:
Hi,
Noticed in v. 3.6.5 MQTT discovery of external temperature sensor for HA is missing "stat_cla" attribute. So this entity is not recored, nor cannot be added to any statistics card.
[image]
I added the attribute manually via MQTT Explorer and HA starts recording it.
[image]

proddy, 2024-04-24:
thanks, created new bug for this