emsesp/EMS-ESP32 #1588: how to send raw message to the ems bus via console?

SSSlee, 2024-01-28:
Can anybody help me with the syntax to send a raw messeage to the ems bus? 

similiar to this line: 
[emsesp] gateway(0x48) -W-> thermostat(0x10), ?(0x02C3), data: 01 18 FF FF FF FF FF FF FF FF FF FF (offset 12)

I assume that these are the time configuration (or a part) to the automatic programs, stored in the RC310.

SSSlee, 2024-01-29:
ok, that works, thank you. But why the high byte of the datatype needs an offset of -1? 0x02C3 becomes 0x01C3 

I'm also struggeling by sending a raw message with another source ID: 

[image]

Only one message was passed through while the content was still the same

MichaelDvP, 2024-01-29:
> But why the high byte of the datatype needs an offset of -1? 0x02C3 becomes 0x01C3

This is a common notation (also used by [norberts1](https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf)) to differ between ems and ems+ telegram-ids. ems has 8 bit id with header 
<src><dst><id><offset><data>
ems+ uses ids as with FF and 16 bit after offset, so we add 0x0100 to the ems+id to differ. 
<src><dst>FF<offset><id(h)><id(l)><data>
e.g. mixer ipm telegram 0x010C is ems+ for <src><dst>FF<offset>00 0C<data>
whereas telegram 0x0C is ems with <src>dst>0C<offset><data>

Telegram 1D for outdoortemp is a ems telegram, so you need to write 
""0B 00 1D 00<data>"
Telegram 2C3 is ems+, so it becomes
"0B 00 FF 00 01 C3 <data>"

You can not write telegrams to the bus with a existing source id, this will always give bus collision. 
The master polls each device and it always replys with <id><break> for no data or a telegram to send. If you try to send with existing id both replys are sendet (as current loop)  and the master gets/repeats garbage.

SSSlee, 2024-01-29:
ok understood, but what's the criteria for a sucessfull transmission? It's send four times and ends with a failed message:
 
[image]

MichaelDvP, 2024-01-30:
Oh, this is a bug. Ems-esp checks reads for the reply and writes for `<ack>/<nack>` from the destination. The broadcast is wrongly handled as write and there is no ack.

closed 2024-01-30 by SSSlee