emsesp/EMS-ESP32 #1597: Missing Parameter "Temp.diff. für Beladung"

timostark, 2024-02-01:
### PROBLEM DESCRIPTION

Parameters "Eco/Eco+/Komfort Temp.diff. für Beladung" (Water) is missing for CS5800i / CS6800i AW.

### REQUESTED INFORMATION

I checked the telegrams.. Attached are the entities I found:

Device: 0x08 (Boiler)
Typ: 0x499
Offset:
- 12 (Komfort)
- 13 (Eco)
- 14 (Eco-Plus)
Value-Type: UINT
Factor: 1
UoM: Probably Kelvin..

At least in the UI800 there are some limitations around it (imo not less than 6, not more than 12 - but I would have to recheck those limitations if they are important for you).

proddy, 2024-02-03:
thanks for reporting this. We'll include in one of the upcoming minor releases.

closed 2024-02-22 by proddy

benni336, 2024-06-16:
@timostark or @proddy : Do you know wether the Bosch Compress 5800i AW is supported? On the [supported device page](https://emsesp.github.io/docs/All-Devices/#all-devices-and-entities) I can only find the 6800i. That suggest that the 5800i AW is supported too. But unfortunately I can find absolute nothing about connecting to an CS5800i AW locally. That is really frustrating and I cannot just "buy" one to figure out later it does not work :(.

Any hint is very welcome! And sorry for OT.

proddy, 2024-06-16:
I guess it does work as @timostark was using it

xarbit, 2024-09-27:
> @timostark or @proddy : Do you know wether the Bosch Compress 5800i AW is supported? On the [supported device page](https://emsesp.github.io/docs/All-Devices/#all-devices-and-entities) I can only [...]

I use it as well with a CS5800i, and in short, it works great! While I’m still getting familiar with heat pumps in general, I’ve been comparing all the entities with the settings in the UI800 and the EMS Gateway. So far, I haven’t found anything missing. I’m still figuring out how to adjust the heat curve using the EMS Gateway and HomeAssistant, but overall, everything is functioning as expected. So yes, you can definitely purchase the gateway for the CS5800i.