emsesp/EMS-ESP32 #1598: Entity: hpea0 / Heating condensate pan (EA0) does not provide the current status

Oktonaut, 2024-02-02:
### PROBLEM DESCRIPTION

Last year I replaced our Buderus GB142-15 condensing boiler with a split heat pump. This is the Logatherm WLW 166i-10 E (indoor unit) in conjunction with WLW-8 SP AR (outdoor unit). EMS-ESP recognized the new device immediately. I just adjusted the inadequate naming of some entities.

So that the condensation drain on the outdoor unit no longer freezes, I retrofitted the heating cable from Buderus. Unfortunately, the associated entity: hpea0 / condensate pan heater (EA0) remains continuously OFF. In order to be able to check the actual status of the heating cable, I connected a Shelly and was able to determine that it was switched on correctly.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ X ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ X ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ X ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ X ] Provide the output of http://ems-esp.local/api/system :

```lua
{
  "System Info": {
    "version": "3.6.4",
    "platform": "ESP32",
    "uptime": "005+05:24:59.720",
    "uptime (seconds)": 451499,
    "free mem": 128,
    "max alloc": 71,
    "free app": 419,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -58,
    "IPv4 address": "192.168.100.100/255.255.255.0",
    "IPv4 gateway": "192.168.100.1",
    "IPv4 nameserver": "253.0.0.0",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": true,
    "server": "192.168.100.1",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 188374,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": true
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 1948790,
    "bus reads (tx)": 429045,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 68,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "locale": "de",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": false,
    "max web log buffer": 50,
    "web log buffer": 50
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 3000-7000iAW/SupraEco/Geo 5xx/WLW196i/WSW196i",
      "device id": "0x08",
      "product id": 172,
      "version": "02.02",
      "entities": 143,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A",
      "handlers ignored": "0xF7 0x049F 0x04A0 0x04A3 0x04AA 0x2E 0x3B 0xF6 0x04B4 0xC6 0xC7 0x0497 0x17 0x36 0x04AB"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "73.03",
      "entities": 58,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E",
      "handlers ignored": "0x0239 0xE7 0x02E0 0x02EA 0x061E 0x0508 0x0509 0x050A 0x0507 0xBF 0xF7"
    },
    {
      "type": "thermostat",
      "name": "RC100H",
      "device id": "0x38",
      "product id": 200,
      "version": "14.01",
      "entities": 4,
      "handlers received": "0x042B 0x047B",
      "handlers fetched": "0x0273",
      "handlers ignored": "0xBF"
    },
    {
      "type": "solar",
      "name": "SM100/MS100",
      "device id": "0x30",
      "product id": 163,
      "version": "27.09",
      "entities": 36,
      "handlers received": "0x0362 0x0363 0x0366 0x0364 0x036A",
      "handlers fetched": "0x0358 0x035A 0x0380 0x038E 0x0391",
      "handlers ignored": "0xE8 0x0368 0x03A9 0x03AA 0xBF 0x038A"
    }
  ]
}
```

### EXPECTED BEHAVIOUR

The entity should always reflect the actual status of the output.

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

For your information, here are the entity names I made
```lua
{
  "type": "customizations",
  "Customizations": {
    "ts": [],
    "as": [],
    "masked_entities": [
      {
        "product_id": 172,
        "device_id": 8,
        "entity_ids": [
          "00uptimecontrol|Betriebszeit Heizen Gesamt",
          "00uptimecompheating|Betriebszeit Kompressor Heizen",
          "00uptimecompcooling|Betriebszeit Kompressor Kühlen",
          "00uptimecompww|Betriebszeit Kompressor Warmwasser",
          "00nrgconstotal|Energieverbrauch Gesamt",
          "00nrgconscomptotal|Energieverbrauch Kompressor Gesamt",
          "00nrgconscompheating|Energieverbrauch Kompressor Heizen",
          "00nrgconscompww|Energieverbrauch Kompressor Warmwasser",
          "00nrgconscompcooling|Energieverbrauch Kompressor Kühlen",
          "00auxelecheatnrgconstotal|Energieverbrauch el. Zusatzheizung Gesamt",
          "00auxelecheatnrgconsww|Energieverbrauch el. Zusatzheizung Warmwasser",
          "00nrgsupptotal|Energieabgabe Gesamt",
          "00nrgsuppheating|Energieabgabe Heizen",
          "00nrgsuppww|Energieabgabe Warmwasser",
          "00nrgsuppcooling|Energieabgabe Kühlen",
          "00nrgsupppool|Energieabgabe Pool"
        ]
      }
    ]
  }
}
```

MichaelDvP, 2024-02-02:
This switch was reported in #1415 as telegram 0x48D, offset 3, bit 6.
Could you check with custom entity the value of 0x48D, offset 3 and if there is a change.

Oktonaut, 2024-02-02:
What exactly should I enter in the form fields for an individual entity?

MichaelDvP, 2024-02-02:
I would say this:
[image]

but Value-Type UINT will also do. You'll get a EA0 entiy in dashboard and mqtt as custom_data. Check if it changes when the heater switches.

Oktonaut, 2024-02-03:
Here it looks a little different:
[image]

When switched off, this returns the value: 08
[image]

Due to the current outside temperature, it is not turning on at the moment.

MichaelDvP, 2024-02-03:
So wait for colder tempertures and check. Or is there a testmenu where you can manual set the output?

Oktonaut, 2024-02-07:
This cannot be switched manually. But today it was cold enough that the heat pump switched to defrost mode. However, the entity set unfortunately has nothing to do with the EA0 switching output. Even though the value changed twice at the end of the defrosting process, it unfortunately does not correspond to EA0.
[image]

Oktonaut, 2024-03-21:
Are there any other possible values that I should log?
There will be more and more spring and therefore not many cold days to be able to do this...

proddy, 2024-03-21:
I can't think of any. Maybe there's a Test Function to simulate the switching on/off of the outdoor unit. Those Buderus systems usually come with test like these which the service guys use