emsesp/EMS-ESP32 #1601: Timer processing in RC310

proddy, 2024-02-03:
### Discussed in https://github.com/emsesp/EMS-ESP32/discussions/1593

<div type='discussions-op-text'>

<sup>Originally posted by **Nxtway** January 30, 2024</sup>
ems-esp supports timer processing of the RC35, [#979](https://github.com/emsesp/EMS-ESP32/discussions/979)
Could these functionality also implemented for the RC310, please ? 

The type-id's in RC310 are:
circulation timer: thermostat(0x10) type(0x0309) | ON - FF | OFF - 00
warm water timer: thermostat(0x10) type(0x02FF) | COMFORT - 02 | ECO - 01 | OFF - ??, maybe 00
heating timer 1: thermostat(0x10) type(0x0683)
heating timer 2 thermostat(0x10)  type(0x068D)

Log files attached:
[WW_timer.txt](https://github.com/emsesp/EMS-ESP32/files/14103769/WW_timer.txt)
[circulation_timer.txt](https://github.com/emsesp/EMS-ESP32/files/14103770/circulation_timer.txt)
[heating_timer1.txt](https://github.com/emsesp/EMS-ESP32/files/14103771/heating_timer1.txt)
[heating_timer2.txt](https://github.com/emsesp/EMS-ESP32/files/14103772/heating_timer2.txt)

An additional improvement might be the holiday timer, but I didn't find the type-id up to now.</div>

MichaelDvP, 2024-02-03:
It's a doublette of #1594

proddy, 2024-02-03:
I'm going mad!

closed 2024-02-03 by proddy