emsesp/EMS-ESP32 #1604: Boost mode not error-free

SeibeGitHub, 2024-02-04:
Buderus GB182i.2-20

BC400  V3.07.43

Boost works correctly from 1-4 hours, from 5 hours it is only active for 45 minutes. It seems as if calculations are being carried out in minutes with an 8-bit data type (decimal max 255) in the BC400. E.g. 4 hours = 240 minutes 5 hours = 300 minutes overflow! 300-255=45 can anyone confirm this? The error lies in the BC400 control

proddy, 2024-03-24:
Perhaps @Bingo2023 can verify?

Bingo2023, 2024-03-24:
EMS-ESP Version
3.6.5-test.16

Set via Home Assistant and checked with Buderus App.

[image]

Seems like 8h is max, 1-8h are working in my system.

Ing-Dom, 2024-05-03:
@SeibeGitHub 
they really talk to you?

I never got anyone from this company talking to me... which way did you contact them?

SeibeGitHub, 2024-05-04:
Sehr geehrter Herr xxxxx,

nach Rückinfo unserer Entwicklungsabteilung ist in der Software der von Ihnen vermutete Fehler einprogrammiert. Mit einer neuen Softwareversion ist jedoch erst ab Mitte 2024 zu rechnen, Erst dann wird auch dieser Fehler dann bereinigt sein.
Wir werden uns mit Ihnen in Verbindung setzen, wenn die neue Software zur Verfügung steht. Bis dahin bitten wir noch um etwas Geduld. Vielen Dank dafür.

Mit freundlichen Grüßen / Best regards

xxxx xxxx

Customer Care Center Esslingen (TTDB/SGS-SCU)
Bosch Thermotechnik GmbH - Buderus Deutschland | Wolf-Hirth-Str. 8 | 73730 Esslingen | GERMANY | [www.buderus.de](http://www.buderus.de/)
Tel. +49 711 9314-732 | [SCU_SGS@buderus.de](mailto:SCU_SGS@buderus.de)

I made a complaint through the installer and after Buderus customer service was at the house I had a contact person.

closed 2024-11-03 by proddy