emsesp/EMS-ESP32 #1609: Missing heatpump energy meter for hotwater

Nxtway, 2024-02-04:
### PROBLEM DESCRIPTION

I miss the energy meter for hot water of my heatpump.

Please can you add the following code in the file heatpump.cpp:
[image]

The result might looks like:
[image]

Thanks a lot and best regards Martin

Nxtway, 2024-02-09:
I tried the new test build 3.6.5-test.12a, but the meter value for hotwater is still missing:
[image]

MichaelDvP, 2024-02-09:
Oh, only added to 0x08 devices, now it should work.

Nxtway, 2024-02-09:
works well in 3.6.5-test.12b

closed 2024-06-21 by proddy