emsesp/EMS-ESP32 #1619: Length of mqtt Broker adress

dgwptech, 2024-02-14:
### PROBLEM DESCRIPTION
I have ordered a Gateway S3-LR (Standard WiFi Edition V3 Long Range) from you.  However, the setup with an MQTT broker does not work because only 63 characters are available for an MQTT broker address in the EMS-ESP.

Please extend this length. Normally a URL has a maximum character length of 2,083 characters, I don't need it to be that long!

Thank you

proddy, 2024-02-14:
It's an IP address, not a URL. With IPv6 that maz length is going to 40 chars

dgwptech, 2024-02-14:
the use of a URL in particular when a cloud service is to be used is not possible?

 

 

	
		
			Von
			Proddy ***@***.***
		
		
			An
			***@***.***
		
		
			CC
			***@***.***; ***@***.***
		
		
			Gespeichert
			14.02.2024 11:50:56
		
		
			Erstellt
			14.02.2024 11:46:00
		
		
			Betreff
			Re: [emsesp/EMS-ESP32] Length of mqtt Broker adress (Issue #1619)
		
	

 

 

It's an IP address, not a URL. With IPv6 that maz length is going to 40 chars

—

In dieser E-Mail wurde potentiell gefährlicher Quellcode entdeckt und entfernt.

proddy, 2024-02-14:
I haven't tried, but it should work. I'll need to test. Are you using something like AWS's IoT? https://aws.amazon.com/blogs/iot/use-aws-iot-core-mqtt-broker-with-standard-mqtt-libraries/

dgwptech, 2024-02-14:
Yes, I use https://docs.hooc.me/de/hooc-gateways/embedded/ a virtual Gateway. They provide an URL, but with more than 63 characters.

 

 

	
		
			Von
			Proddy ***@***.***
		
		
			An
			***@***.***
		
		
			CC
			***@***.***; ***@***.***
		
		
			Gespeichert
			14.02.2024 13:09:49
		
		
			Erstellt
			14.02.2024 13:09:18
		
		
			Betreff
			Re: [emsesp/EMS-ESP32] Length of mqtt Broker adress (Issue #1619)
		
	

 

 

I haven't tried, but it should work. I'll need to test. Are you using something like AWS's IoT? https://aws.amazon.com/blogs/iot/use-aws-iot-core-mqtt-broker-with-standard-mqtt-libraries/

—

In dieser E-Mail wurde potentiell gefährlicher Quellcode entdeckt und entfernt.

closed 2024-02-14 by proddy

dgwptech, 2024-02-15:
perfect! Thank you

 

 

 

	
		
			Von
			Proddy ***@***.***
		
		
			An
			***@***.***
		
		
			CC
			***@***.***; ***@***.***
		
		
			Gespeichert
			14.02.2024 15:11:19
		
		
			Erstellt
			14.02.2024 15:04:19
		
		
			Betreff
			Re: [emsesp/EMS-ESP32] Length of mqtt Broker adress (Issue #1619)
		
	

 

 

Fixed in v3.6.5-dev.14

—

In dieser E-Mail wurde potentiell gefährlicher Quellcode entdeckt und entfernt.