emsesp/EMS-ESP32 #1620: Firmware 3.6.4 is downgrading after restart or power loss

bluediamond1984, 2024-02-14:
### PROBLEM DESCRIPTION

Successful upgraded to firmware 3.6.4. But when I restart the device through webUI or disconnect the power it is downgrading to firmware 3.6.0.

### REQUESTED INFORMATION

<details>
<summary>System information output</summary>

```lua
{
  "System Info": {
    "version": "3.6.0",
    "platform": "ESP32",
    "uptime": "000+00:09:32.543",
    "uptime (seconds)": 572,
    "free mem": 147,
    "max alloc": 95,
    "free app": 8,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -43,
    "IPv4 address": "192.168.20.69/255.255.255.0",
    "IPv4 gateway": "192.168.20.1",
    "IPv4 nameserver": "192.168.20.1",
    "static ip config": true,
    "enable IPv6": false,
    "low bandwidth": true,
    "disable sleep": false,
    "enable MDNS": false,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 211,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "nefit",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 292,
    "bus reads (tx)": 44,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 8
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Topline/GB162",
      "device id": "0x08",
      "product id": 115,
      "version": "03.06",
      "entities": 67,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "controller",
      "name": "BC10",
      "device id": "0x09",
      "product id": 114,
      "version": "01.03",
      "entities": 0
    }
  ]
}

```

</details>

### TO REPRODUCE

Upload latest firmware 3.6.4 through webUI
Restart with button on system status, or disconnect the power
After restart EMS-ESP Version is back to 3.6.0

### EXPECTED BEHAVIOUR

A restart should stay with firmware 3.6.4

### ADDITIONAL CONTEXT

Log after upgrade 3.6.4
```
2024-02-14 12:14:27.000 INFO 1: [emsesp] Starting EMS-ESP version 3.6.4
2024-02-14 12:14:27.000 INFO 2: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
2024-02-14 12:14:27.000 INFO 3: [emsesp] Loading board profile S32
2024-02-14 12:14:27.000 NOTICE 4: [system] Upgrading to version 3.6.4-
2024-02-14 12:14:27.000 INFO 5: [emsesp] Starting Scheduler service
2024-02-14 12:14:27.000 INFO 6: [emsesp] Starting Custom entity service
2024-02-14 12:14:27.000 INFO 7: [temperaturesensor] Starting Temperature sensor service
2024-02-14 12:14:27.000 INFO 8: [analogsensor] Starting Analog sensor service
2024-02-14 12:14:27.000 INFO 9: [emsesp] Loaded EMS device library (117 records)
2024-02-14 12:14:28.917 INFO 10: [emsesp] Starting NTP service
2024-02-14 12:14:28.977 INFO 12: [mqtt] MQTT connected
2024-02-14 12:14:30.155 INFO 13: [system] NTP connected
2024-02-14 12:14:32.216 INFO 14: [emsesp] Recognized new boiler with deviceID 0x08
2024-02-14 12:14:32.585 INFO 15: [emsesp] Recognized new controller with deviceID 0x09
```

Log after reboot
```
2024-02-14 12:15:52.000 INFO 1: [emsesp] Starting EMS-ESP version 3.6.0
2024-02-14 12:15:52.000 INFO 2: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
2024-02-14 12:15:52.000 INFO 3: [emsesp] Loading board profile S32
2024-02-14 12:15:52.000 NOTICE 4: [system] Downgrading to version 3.6.0-
2024-02-14 12:15:52.000 INFO 5: [emsesp] Starting Scheduler service
2024-02-14 12:15:52.000 INFO 6: [emsesp] Starting Custom entity service
2024-02-14 12:15:52.000 INFO 7: [temperaturesensor] Starting Temperature sensor service
2024-02-14 12:15:52.000 INFO 8: [analogsensor] Starting Analog sensor service
2024-02-14 12:15:52.000 INFO 9: [emsesp] Loaded EMS device library (116 records)
2024-02-14 12:15:52.000 INFO 10: [emsesp] Starting Access Point with captive portal on 192.168.4.1
2024-02-14 12:15:54.776 INFO 11: [emsesp] Starting NTP service
2024-02-14 12:15:54.776 INFO 12: [emsesp] WiFi connected with IP=192.168.20.69, hostname=ems-esp
2024-02-14 12:15:54.843 INFO 13: [mqtt] MQTT connected
2024-02-14 12:15:58.904 INFO 14: [system] NTP connected
2024-02-14 12:16:02.001 INFO 15: [emsesp] Stopping Access Point
2024-02-14 12:16:06.198 INFO 16: [emsesp] Recognized new boiler with deviceID 0x08
2024-02-14 12:16:06.814 INFO 17: [emsesp] Recognized new controller with deviceID 0x09

```

proddy, 2024-02-14:
This is a known problem with previous versions and the way the firmware was flashed to the BBQKees devices. You will need to make a backup of your settings and re-flash using the Flash tool. See https://emsesp.github.io/docs/#breaking-changes-in-v36x

closed 2024-02-14 by proddy

bluediamond1984, 2024-02-14:
Thanks for the quick response. That's what I had done when I upgraded to 3.5.1.
The warning in the docs description is from 3.4 to 3.6. Apparently I needed to flash it also from 3.5.1 to 3.6.x.