emsesp/EMS-ESP32 #1624: Missing Parameter: Start / Stop Temperature

timostark, 2024-02-19:
### PROBLEM DESCRIPTION

For Bosch CS5800 / CS6800 (or Buderus WLW186i) the parameters Start && Stop temperature are missing for Eco, Eco+, Komfort and Extra (for extra only the stop temperature).

### REQUESTED INFORMATION

Via EMS telegrams I identified the following parameters..Note that at least on the UI-800 the heatpump is expecting at least a difference of 6K between Start and Stop.

Start-Temperatures:

Komfort: Device-Id: 0x8, Type-Id: 0xEA, Offset: 6
Eco: Device Id: 0x8, Type-Id: 0xEA, Offset: 18
Eco+: 0x, 0xEA, Offset 27

Stopp-Temperatures:
Komfort: 0x8, 0x499, Offset 8
Eco: 0x8, 0x499, Offset: 9
Eco+: 0x8, 0x499, Offset 10
Extra: 0x8, 0xEA, Offset 16

MichaelDvP, 2024-02-21:
For 0xEA it's just another Bosch renaming,see here:
https://github.com/emsesp/EMS-ESP32/blob/8ebc552cac04039c4eaf2d6d1934f4ef8e5ff98e/src/devices/boiler.cpp#L1488-L1521
For 0x499, could you suggest short/long names for the entities, i don't know what a stop temperature is and what heating do with it.

timostark, 2024-02-26:
@MichaelDvP Thanks for your answer - i appereantly have missed the existance of the WWPlus attributes here. Just as a hint: maybe it would be helpful while adding a custom entity there is a warning/info text that there is a standard entity for the added custom entity.

Regarding stop temperature: In the WArm-Water Temperature settings there is a start and a stop temperature for every single mode (Eco+, Eco, Comfort). As per my understanding a start temperature of 38 Degree and a Stop Temperature of 45 degree means, that the water should be heated in case the temperature sensor goes below 38 degree and heat until 45 degree. 

The german name in the info menu is: "Eco Starttemperatur" and "Eco Stopptemperatur".

Goal is basically to make some adjustments to the comfort level in case of good solar weather forecast (example: normally the range is 37 --> 45. In case of very good solar weather forecast during the day this is adjusted to 42 --> 48).

MichaelDvP, 2024-02-26:
In this 499 telegram we have now for each dhw mode a stop,temp a diff temp and an off temp. Seems to me very redundant, what are the differences? 
My old boiler have only setpoint an hysteresis for on/off (hysteresis applies to all modes), that looks logical to me. 

But i have added the stop values to my test-build https://github.com/MichaelDvP/EMS-ESP32/releases

timostark, 2024-02-26:
Hmm.. in my boiler (CS5800 i AW with UI800 - see https://junkers-de-de-b.boschtt-documents.com/download/pdf/file/6721833375.pdf?token=u0lmged8q6pkjs8vjo9dp92c40 ) i can find "Ladedelta (TC1-TW1) für Komfortbetrieb." (Basically the flow-temperature increase above the warm-water temperatur - so we define how fast the warm-water is loading) and the start-/stop temperature for each mode.

So for each mode i can uniquly define how fast the warm-water loading will be performed (6 to 12 Kelvin) and from where to where the water temperature will be increased. That is a little bit more fine granular than just ahving a hystersis for all modes in your boiler.

Is there an easy way to test your test-build? (some documentation for that anywhere?).

closed 2024-05-08 by proddy