emsesp/EMS-ESP32 #1634: 2 warm water

richii93, 2024-02-23:
### PROBLEM DESCRIPTION

I do have a Buderus GB162 + RC310. There are 3 heating circuits and 2 warm water circuits connected.
The ems-esp gateway can see all the heating circuits (HC1, HC2, HC3) but only one warm water circuit shows up, expect for "dhw circuit 1 extra" and "dhw circuit 2 extra".
The RC310 does show both warm water circuits and the settings for each one. 

### REQUESTED INFORMATION

```lua
{"System Info":{"version":"3.6.5-dev.15","uptime":"000+00:02:12.597","uptime (seconds)":132,"platform":"Tasmota Arduino v2.0.14","sdk":"4.4.6.240105","free mem":139,"max alloc":67,"used app":1586,"free app":398,"partition":"app1","reset reason":"Software reset CPU / Software reset CPU"},"Network Info":{"network":"WiFi","hostname":"ems-esp","RSSI":-78,"BSSID":"set","TxPower setting":78,"static ip config":false,"enable IPv6":false,"low bandwidth":false,"disable sleep":false,"enable MDNS":true,"enable CORS":false,"AP provision mode":"disconnected","AP security":"wpa2","AP ssid":"ems-esp"},"NTP Info":{"NTP status":"connected","enabled":true,"server":"time.google.com","tz label":"Europe/Amsterdam"},"OTA Info":{"enabled":false,"port":8266},"MQTT Info":{"MQTT status":"disconnected","enabled":false,"client id":"ems-esp","keep alive":60,"clean session":false,"entity format":1,"base":"ems-esp","discovery prefix":"homeassistant","discovery type":0,"nested format":1,"ha enabled":false,"mqtt qos":0,"mqtt retain":false,"publish time heartbeat":60,"publish time boiler":10,"publish time thermostat":10,"publish time solar":10,"publish time mixer":10,"publish time other":10,"publish time sensor":10,"publish single":false,"publish2command":false,"send response":false},"Syslog Info":{"enabled":false},"Sensor Info":{"temperature sensors":0,"temperature sensor reads":0,"temperature sensor fails":0,"analog sensors":0,"analog sensor reads":0,"analog sensor fails":0},"API Info":{"API calls":0,"API fails":0},"Bus Info":{"bus status":"connected","bus protocol":"Buderus","bus telegrams received (rx)":427,"bus reads (tx)":125,"bus writes (tx)":0,"bus incomplete telegrams":0,"bus reads failed":0,"bus writes failed":0,"bus rx line quality":100,"bus tx line quality":100},"Settings":{"board profile":"CUSTOM","locale":"de","tx mode":2,"ems bus id":11,"shower timer":false,"shower alert":false,"phy type":0,"rx gpio":23,"tx gpio":5,"dallas gpio":19,"pbutton gpio":0,"led gpio":2,"hide led":true,"notoken api":false,"readonly mode":false,"fahrenheit":false,"dallas parasite":false,"bool format":1,"bool dashboard":1,"enum format":1,"analog enabled":true,"telnet enabled":true,"max web log buffer":50,"web log buffer":19},"Devices":[{"type":"boiler","name":"Topline/GB162","device id":"0x08","product id":115,"version":"05.09","entities":78,"handlers received":"0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04","handlers fetched":"0x14 0x16 0x33","handlers pending":"0xBF 0xC2 0x1A 0x35 0x2A 0xD1 0xE3 0xE4 0xE5 0xE9","handlers ignored":"0x1E"},{"type":"thermostat","name":"RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410","device id":"0x10","product id":158,"version":"74.04","entities":120,"handlers received":"0x06 0xA2 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x031E 0x0267","handlers fetched":"0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x02A7 0x02BB 0x02B1 0x029D 0x0291 0x02CE 0x0292 0x02D0 0x0293 0x0294 0x02F5 0x031B 0x023A 0x0240","handlers pending":"0xA3 0x12 0x13 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02D2 0x046A"},{"type":"mixer","name":"MM100","device id":"0x20","product id":160,"version":"24.05","entities":4,"handlers received":"0x02D7","handlers ignored":"0x16 0x33 0x0255 0x0291 0x02E1 0x02EB"},{"type":"mixer","name":"MM100","device id":"0x21","product id":160,"version":"24.05","entities":7,"handlers received":"0x02D8","handlers fetched":"0x02CE","handlers ignored":"0x0255 0x02E2 0x02EC 0x16 0x33 0x0292"},{"type":"mixer","name":"MM100","device id":"0x22","product id":160,"version":"24.05","entities":4,"handlers received":"0x02D9","handlers ignored":"0x16 0x33 0x0255 0x0293 0x02E3 0x02ED"},{"type":"mixer","name":"MM100","device id":"0x29","product id":160,"version":"24.05","entities":8,"handlers received":"0x0332","handlers fetched":"0x0314","handlers ignored":"0x0255 0xBF 0x033C 0x16 0x33"},{"type":"controller","name":"BC10/RFM20","device id":"0x09","product id":68,"version":"02.03","entities":0}]}

```

### EXPECTED BEHAVIOUR

I would like to be able to see and change all the entities for each warm water circuit.

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

Thanks for your help.

MichaelDvP, 2024-02-24:
I think the second water circuit is managed by the mixer at 0x29 and shows as `wwc2`.
You can try the test build that maps it as water circuit instead of mixer.

richii93, 2024-02-24:
Thanks for your reply. When I look at the mixer, there are less / other entities:

[image]

I did upload the "test build" EMS-ESP-3_6_5-test_15-ESP32. But when I try to configure the GPIOs for the custom board profile and apply the settings, they get lost again. Therefore the test build does not work for me right now.

MichaelDvP, 2024-02-24:
Ok, if the RC300 have a second dhw circuit we have to find the telegrams, lets try with next to dhw1:
Open terminal and give me the output for:
`read 10 2F6`
`read 10 31C`
`read 10 200`
`read 10 20A`
and for compare to dhw 1:
`read 10 2F5`
`read 10 31B`
`read 10 1FF`
`read 10 209`
> the settings, they get lost again.

What do you mean by that? Is one of the gpios marked as illegal?  
Doesn't the test build uses the same profile as the dev? Why do you have to change?

richii93, 2024-02-24:
```
ems-esp:$ read 10 2F6
000+01:30:28.728 N 1: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x02F6), data: 02 FF 00 02 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00 01
ems-esp:$ read 10 31C
000+01:30:53.752 N 2: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x031C), data: <empty>
ems-esp:$ read 10 200
000+01:31:02.527 N 3: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x0200), data: <empty>
ems-esp:$ read 10 20A
000+01:31:10.179 N 4: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x020A), data: <empty>
ems-esp:$ read 10 2F5
000+01:31:18.204 N 5: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300WWmode(0x02F5), data: 01 FF 00 02 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00 01
ems-esp:$ read 10 31B
000+01:31:30.207 N 6: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300WWtemp(0x031B), data: 3C 2D
ems-esp:$ read 10 1FF
000+01:31:37.754 N 7: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x01FF), data: <empty>
ems-esp:$ read 10 209
000+01:31:46.028 N 8: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x0209), data: <empty>
```

> What do you mean by that? Is one of the gpios marked as illegal?
> Doesn't the test build uses the same profile as the dev? Why do you have to change?

After I uploaded the test build there was no ems-bus connection. Therefore I checked the settings and realized, that the GPIOs were all set to "0". I then set them back to the correct numbers and saved the settings. However, when I switch to the dashboard site and back to the settings all GPIOs are set to "0" again.

[image]

MichaelDvP, 2024-02-24:
Ok, found the bug for custom settings. Will upload soon.
The 0x2F6 have all dhw values except setpoint and setpoint low for the second circuit. Can you set different setpoints for the dhw2 circuit on the RC300? Then set logger to log all and change both setpoints, post log here.

MichaelDvP, 2024-02-25:
Please try this build: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test
The custom profile should work now and i've added the wwc2 circuit to thermostat. I'm not sure if all maps correctly, please check the web (some screenshots would be nice), and the mqtt output for `thermostat_data`.
I leave fist circuit with tag `DHW` for compatbility and use `WWC2` for second circuit (identical to water/mixer circuits).

richii93, 2024-02-25:
> Can you set different setpoints for the dhw2 circuit on the RC300? Then set logger to log all and change both setpoints, post log here.

I will do that later, right now I have just remote access.

> Please try this build: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test

Thank you for your help. 

[image]

I just realized that when I change an entity of `wwc2`, it will update the corresponding entity of `dhw` and leave the value of the `wwc2`-entity unchanged.

richii93, 2024-02-25:
> Can you set different setpoints for the dhw2 circuit on the RC300? Then set logger to log all and change both setpoints, post log here.

I hope, this is what you were looking for:

```
000+01:02:37.823 T 84: [emsesp] mixer(0x20) -B-> All(0x00), MMPLUSStatusMessage_HC(0x02D7), data: 01 (offset 1)
000+01:02:40.026 T 85: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0A 00 88 00 88 00 00 00 03 00 04 D6 15 00 B9 EF 00 80 00
000+01:02:40.234 T 86: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 5A 01 86 64 00 00 01 04 60 80 00 00 88 01 85 00 00 09 2D 2D 01 90 00 00 00
000+01:02:42.148 T 87: [emsesp] mixer(0x22) -W-> boiler(0x08), UBASetPoints(0x1A), data: 1A 64 00
000+01:02:42.173 T 88: [emsesp] mixer(0x22) -B-> All(0x00), ?(0x0255), data: 00
000+01:02:42.375 T 89: [emsesp] mixer(0x22) -B-> All(0x00), MMPLUSStatusMessage_HC(0x02D9), data: 01 00 03 00 E8 18 00 00 08
000+01:02:43.058 T 90: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 5A 01 86 64 00 00 01 04 60 80 00 00 88 01 85 00 00 09 2D 2D 01 90 00 00 00
000+01:02:43.548 T 91: [emsesp] mixer(0x20) -B-> All(0x00), MMPLUSStatusMessage_HC(0x02D7), data: 00 (offset 1)
000+01:02:44.124 T 92: [emsesp] thermostat(0x10) -W-> water(0x29), MMPLUSConfigMessage_WWC(0x0314), data: 28 (offset 5)
000+01:02:44.825 T 93: [emsesp] water(0x29) -B-> All(0x00), MMPLUSConfigMessage_WWC(0x0314), data: 28 (offset 5)
000+01:02:45.746 T 94: [emsesp] mixer(0x20) -W-> boiler(0x08), ?(0x1E), data: 01 81
000+01:02:47.979 T 95: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 03 1A (offset 15)
000+01:02:50.100 T 96: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0A 00 86 00 88 00 00 00 03 00 04 D6 15 00 B9 EF 00 80 00
000+01:02:50.358 T 97: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 5A 01 86 64 00 00 01 04 60 80 00 00 88 01 85 00 00 09 2D 2D 01 90 00 00 00
000+01:02:52.602 T 98: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A6), data: 03 1A (offset 15)
000+01:02:52.789 T 99: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A7), data: 03 1A (offset 15)
000+01:02:54.058 T 100: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 2C 01 86 64 00 01 01 04 60 80 00 00 88 01 85 00 00 09 30 48 00 CB 00 00 00
000+01:02:55.546 T 101: [emsesp] mixer(0x20) -W-> boiler(0x08), ?(0x1E), data: 01 80
000+01:02:58.722 T 102: [emsesp] thermostat(0x10) -W-> boiler(0x08), UBAFlags(0x35), data: 01 01
000+01:03:00.051 T 103: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0A 00 88 00 88 00 00 00 03 00 04 D6 15 00 B9 EF 00 80 00
000+01:03:00.357 T 104: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 2C 01 87 64 00 01 01 04 60 80 00 00 88 01 85 00 00 09 30 59 00 CC 00 00 00
000+01:03:00.625 T 105: [emsesp] mixer(0x21) -B-> All(0x00), MMPLUSStatusMessage_HC(0x02D8), data: 01 (offset 1)
000+01:03:00.801 T 106: [emsesp] mixer(0x21) -R-> boiler(0x08), UBAParameters(0x16), length: 0x01
000+01:03:00.833 T 107: [emsesp] boiler(0x08) -W-> mixer(0x21), UBAParameters(0x16), data: FF
000+01:03:00.856 T 108: [emsesp] mixer(0x21) -W-> boiler(0x08), UBASetPoints(0x1A), data: 2C 64 00
000+01:03:00.894 T 109: [emsesp] mixer(0x21) -R-> boiler(0x08), UBAParameterWW(0x33), length: 0x0B
000+01:03:00.918 T 110: [emsesp] boiler(0x08) -W-> mixer(0x21), UBAParameterWW(0x33), data: 08 FF 3C FB 00 1E FF 04 46 00 00
000+01:03:00.946 T 111: [emsesp] mixer(0x21) -B-> All(0x00), ?(0x0255), data: 00
000+01:03:01.418 T 112: [emsesp] Me(0x0B) -R-> boiler(0x08), UBATotalUptime(0x14), length: 0x1B
000+01:03:01.435 T 113: [emsesp] boiler(0x08) -W-> Me(0x0B), UBATotalUptime(0x14), data: 36 6D 50
000+01:03:01.623 T 114: [emsesp] mixer(0x21) -R-> thermostat(0x10), HPMode(0x0292), length: 0x01 (offset 5)
000+01:03:01.646 T 115: [emsesp] thermostat(0x10) -W-> mixer(0x21), HPMode(0x0292), data: 00 (offset 5)
000+01:03:01.689 T 116: [emsesp] mixer(0x21) -B-> All(0x00), MMPLUSStatusMessage_HC(0x02D8), data: 01 01 0D 01 45 27 00 00 08
000+01:03:02.143 T 117: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameters(0x16), length: 0x1B
000+01:03:02.169 T 118: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameters(0x16), data: FF 5A 64 00 06 FA 0A 01 05 39 1E 00
000+01:03:02.617 T 119: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameterWW(0x33), length: 0x1B
000+01:03:02.643 T 120: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 3C FB 00 1E FF 04 46 00 00
000+01:03:03.118 T 121: [emsesp] Me(0x0B) -R-> boiler(0x08), UBASettingsWW(0x26), length: 0x1B
000+01:03:03.135 T 122: [emsesp] boiler(0x08) -W-> Me(0x0B), UBASettingsWW(0x26), data: <empty>
000+01:03:03.567 T 123: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParametersPlus(0xE6), length: 0x1B
000+01:03:03.585 T 124: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParametersPlus(0xE6), data: <empty>
000+01:03:04.044 T 125: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameterWWPlus(0xEA), length: 0x1B
000+01:03:04.057 T 126: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameterWWPlus(0xEA), data: <empty>
000+01:03:04.223 T 127: [emsesp] mixer(0x20) -B-> All(0x00), MMPLUSStatusMessage_HC(0x02D7), data: 01 (offset 1)
000+01:03:04.672 T 128: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 0x19
000+01:03:04.752 T 129: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 A3 21 1E 27 00 1E 1E 05 A0 01 01 01 05 A0 03 1A 00 00 11 01 04 06 5E 00
[... cut ...]
000+01:03:42.196 T 279: [emsesp] mixer(0x22) -B-> All(0x00), ?(0x0255), data: 00
000+01:03:42.378 T 280: [emsesp] mixer(0x22) -R-> thermostat(0x10), HPMode(0x0293), length: 0x01 (offset 5)
000+01:03:42.401 T 281: [emsesp] thermostat(0x10) -W-> mixer(0x22), HPMode(0x0293), data: 00 (offset 5)
000+01:03:42.444 T 282: [emsesp] mixer(0x22) -B-> All(0x00), MMPLUSStatusMessage_HC(0x02D9), data: 01 00 03 00 E4 18 00 00 08
000+01:03:45.747 T 283: [emsesp] mixer(0x20) -W-> boiler(0x08), ?(0x1E), data: 01 7F
```

MichaelDvP, 2024-02-25:
Ok, i see:
000+01:02:44.124 T 92: [emsesp] thermostat(0x10) -W-> water(0x29), MMPLUSConfigMessage_WWC(0x0314), data: 28 (offset 5)
000+01:02:44.825 T 93: [emsesp] water(0x29) -B-> All(0x00), MMPLUSConfigMessage_WWC(0x0314), data: 28 (offset 5)

setting the reduced temperture to 40 degrees. The setpoints are not saved in thermostat, only in mixer. You also have to set it on the mixer.

Dashboard looks good, i'll check why the tag is not checked correctly.

MichaelDvP, 2024-02-25:
Updated the test build on my github, please test if it changes the right value now.

richii93, 2024-02-26:
It seems to work now. I will have to check all the entities in detail later but the first few ones I tried are working now correctly. 

Thank you very much for your quick help.

MichaelDvP, 2024-02-27:
@proddy The second dhw circuit is managed by mixer (water device in dev2, tagged wwc2) and have also some entities in thermostat. I have tagged them also with wwc2. Now we have:
boiler: tag: DHW, always published as `boiler_data_ww`
thermostat, first circuit: tag DHW, published in ' thermostat_data'  (not nested)
thermostat, second circuit: tag WWC2, published in ' thermostat_data/wwc2'  or for mqtt-single in `thermostat_data_wwc2`
water: tag: WWC1, 2, .., published in ' water_data/wwc1' or (single) in `water_data_wwc1`, etc.

This looks a bit odd. Since we have a breaking change with water device, i think we should bring it in line. But which way?
Should we tag the first thermostat circuit with WWC1? But this does not fit to boiler DHW tag.  

Or: Leave boiler/thermostat at DHW and rename the water circuits to DHW2.... 
For thermostat also nest the first DHW circuit. Then thermostat has dhw tag for the boiler dhw tagged control and dhw2 for the water(mixer/solar) dhw2 control.
What about mqtt topics? Rename `boilder_data_ww` to `boiler_data_dhw` and in non-nested use `thermostat_data_dhw`, `thermostat_data_dhw2`. This is a breaking change for all, as boiler_data_ww topic is changed to boiler_data_dhw, but i think this is best and clean solution.  Or leave `boiler_data_ww` and use `thermostat_data_ww`, `thermostat_data_ww2` and also for water devices `data_ww2`, ...

proddy, 2024-03-24:
...thinking about this one.

proddy, 2024-03-29:
@MichaelDvP I think we should go for DHW everywhere, replacing the old 'ww'. It'll be a breaking change, but it feels like the right thing to do.

MichaelDvP, 2024-03-30:
Ok, i'll make a build. I think this removes a lot of special cases:
- remove TAG_BOILER_DATA_WW and TAG_DEVICE_DATA_WW, only: tag_dhw1, ... where dhw1 shows as "dhw", others with number.
- always tag the dhw entities for all devices
- nested mqt: nest dhw in `boiler_data`
- remove 'ww' from mqtt entities: e.g. wwactivated becomes dhw:{activated}
- commands, api: `dhw/..` and `dhw1/.`. are equal.

i think there will be more, let's start with this, check  and continue.

MichaelDvP, 2024-03-31:
@proddy I've pushed the changes to my dev branch. Please check, I'll test a bit more before PR.

proddy, 2024-04-01:
> @proddy I've pushed the changes to my dev branch. Please check, I'll test a bit more before PR.

checking....

proddy, 2024-04-13:
It looks fine, I'm testing the HA integration now

proddy, 2024-04-15:
all works well @MichaelDvP . I like the nested json object in boiler_data. I think we can merge this into dev at the next opportunity

proddy, 2024-04-26:
closing this.

closed 2024-04-26 by proddy