emsesp/EMS-ESP32 #1640: Reading temperaturesensor via API - 3.6.5-dev.15

proddy, 2024-02-27:
### Discussed in https://github.com/emsesp/EMS-ESP32/discussions/1637

<div type='discussions-op-text'>

<sup>Originally posted by **SoftVienka** February 26, 2024</sup>
Good day.
Can you please tell me how the loading of temperaturesensor via API changed between ver.3.6.5-dev.5 and 3.6.5-dev.15 ?
I used "/api/temperaturesensor/28-FF64-1F45-F4F0/value" to load it, which returned the temperature value of the DS18B20 sensor.
Now it returns httpResponseCode 400.
When I use "/api/temperaturesensor/28-FF64-1F45-F4F0" it returns correct
{
     "id": "28-FF64-1F45-F4F0",
     "name": "28-FF64-1F45-F4F0",
     "value": 25.9,
     "type": "number",
     "uom": "°C",
     "writeable": false
}
Thank you
(translated by google)</div>

proddy, 2024-02-27:
@MichaelDvP fixed it, thanks.

closed 2024-02-27 by proddy

SoftVienka, 2024-02-28:
Very good. Thank you.