emsesp/EMS-ESP32 #1642: Worcester-Bosch Greenstar weather compensation support

V4ler1an, 2024-02-27:
I have a Greenstar 4000 system boiler model: GR4700iW S - 15 NG that has been fitted with the official Worcester outdoor weather sensor kit (part #: 7716192764): https://www.amazon.co.uk/Worcester-7716192764-Greenstar-Outdoor-Weather/dp/B07H63CKPZ

I have an E32 v2 (Wi-Fi + Ethernet v2) EMS-Bus Gateway hooked up through the service jack with 69 entities published to Home Assistant through MQTT. However.... the boiler service menu offers the following controls to manage the weather compensation features but these are not currently supported by the gateway's ESP firmware:

Enable / Disable Heating Curve
Increase / Decrease Curve Base Temp
Increase / Decrease Curve End Temp

Could I have these added to the firmware asap? Syslog trace attached recording the following changes I made to these settings on the boiler system menu soon after 13:53

- Heating curve turned off
- Heating curve turn on
- Htg. curve base increased 2 deg C
- Htg. curve end increased 2 deg C

Many thanks!
[log.txt](https://github.com/emsesp/EMS-ESP32/files/14420580/log.txt)

MichaelDvP, 2024-02-27:
For first test try with custom entities and test values and writing:
[image]

[image]

[image]

V4ler1an, 2024-02-27:
Thanks Michael:
[log - telnet.txt](https://github.com/emsesp/EMS-ESP32/files/14421357/log.-.telnet.txt)

V4ler1an, 2024-02-27:
Just to clarify, the telnet log I provided was taken while changing the settings using the new HA custom entities. Were you preferring that I make the test changes via the boiler LCD control panel or does it not matter?

MichaelDvP, 2024-02-27:
Yes, i can see it in the log, looks godd, we can add the entities, but i want you to type in telnet the command `read 8 28`
This should give the whole telegram back, not only offsets 0-2.

MichaelDvP, 2024-02-27:
Thanks,
`[emsesp] boiler(0x08) -W-> Me(0x0B), ?(0x28), data: 00 3C 32 10 00 05`
3 other parameters with value 16, 0 and 5, any idea what is it?
I've added the on/base/end to my dev, please check, if ok i'll make a PR.
https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

V4ler1an, 2024-02-27:
I've had a look over the changes although don't know the value of my checking the release, i'm certainly no dev :) But the custom entities I have seem to work and i'll continue testing / checking them in my environment and be sure to make you aware of any problems I encounter.

In regards to those values. It's a bit hard to know, can you give me any more context? Looking at the entity values being reported in HA nothing has a value of 16 (currently), loads of things are on 0 and only 'Burner min period' is 5 and 'Dhw set temperature' is 5 deg C.

MichaelDvP, 2024-02-27:
> But the custom entities I have seem to work and i'll continue testing

The custom entities are for first testing, they have some overhead and it's better to integrate the entities if all is ok. So try my build and if it works you can remove the customs.

> In regards to those values. It's a bit hard to know, can you give me any more context?

The telegram from boiler contains 6 values, we know the first three, on/off, base and end, the next three have values 16, 0 ,and 5. Is there anything in the boiler control panel regarding curve that have these values, than we can add them too. A heating curve normally have for start and end values (which are  flowtemps) a corresponding outdoor temperature, Maybe startemp is the required flowtemp at outdoor 16 degrees (offset 3) , endtemp is reached at 0 degrees (offset 4). Dont know, just thinking...

V4ler1an, 2024-02-27:
Ahah! There are three other settings within the 'Heating Curve' service sub-menu. Presumably all are availble once the outdoor weather sensor is fitted to the boiler. Presumably it would be good to add these also? Their values match those you provided:

Summer Mode: 16
Prost Prot.: 0
Front temp. limit: 5

[image]

V4ler1an, 2024-02-27:
Interestingly the outdoor temperature sensor isn't reported in this sub-menu. That's reported under Info / Boiler. I'll hold on until we have all 6 new settings in your build if that's ok? Let me know when you're ready.

MichaelDvP, 2024-02-28:
Ok, please test https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
I think the nofrost-mode (frost prot) is a switch with  on=0xFF, off=0. 
(named "nofrost mode" because most thermostats use this name, but Bosch rename entities often for different brands/models/software versions, we can't include all names, you can change in customizations)

V4ler1an, 2024-02-28:
Yes it's a boolean, i'll test later today. Here's the section for the heating curve menu in the GR4700iW S installer's manual: 
[image]

V4ler1an, 2024-02-28:
After getting the new controls added to a dashboard card and tidied up, I noticed a 7th 'Reset' entity included along with the curve controls. Curious as to if and when it might be used and where it belongs. 

I've done a quick sanity check and the curve enables / disables, base / end temps change ok. To do a more thorough test I really need to build these entities into a new weather comp automation. Am I advsied to wait for the final release before doing this do you think? Also, how do I subscribe for notifications on the new releases? Thanks.

[image]

MichaelDvP, 2024-02-28:
The reset is from #1605, PR #1626 in the official dev.
>  Also, how do I subscribe for notifications on the new releases?

Don't know, proddy announces them on discord.

proddy, 2024-02-28:
you can also watch GitHub:

[image]

V4ler1an, 2024-02-28:
Perfect, thanks.

closed 2024-03-09 by proddy