emsesp/EMS-ESP32 #1644: dns ipv6/ipv4 issue with tasmota builds

MichaelDvP, 2024-02-28:
When connecting the emsesp in IPv6 environment the newer tasmota builds (2024.01.00 and 24.01.01) switch after ~1min (refresh time for dns) to the (router provided) IPv6 address for dns. This also happen when IPv6 is disabled. In consequence the dns sever can not be reached. If IPv6 is enabled the dns could be reched and dns is working ok.

As second issue the network status page formats the dns ip always as IPv4, showing strange adresses after the IPv6 switch. 
e.g.  dns 8.8.8.8 have IPv6 2001:4860:4860::8888 and shows as  32.1.72.96
I'll make a PR to show the dnsIP is correct format. But did not find a solution to fix the dns to IPv4, 

@proddy Should we enable IPv6 by default, and remove the option?

proddy, 2024-02-28:
yes, I saw the Discord thread. I've been out these last days so still need to catch up on EMS-ESP stuff.

Agree to always enable IPv6 if that is possible

proddy, 2024-03-27:
@MichaelDvP espressif officially released their ESP32 SDK 5.2.1 on March 6th https://github.com/espressif/esp-idf/releases/tag/v5.2.1 and I read the platformio-espressif will also be updated in the next weeks, so we could move to that soon. 5.2.1 apparently fixes a lot of the WiFi channel switching issues on mesh networks

proddy, 2024-04-27:
AsyncWebServer and AsyncTCP have recent changes to support IPv6 so I'll update our libraries too

MichaelDvP, 2024-04-27:
IPv6 is supported in our builds since a long time, the issue is IPv4 only config for ems-esp if dhcp also serves a IPv6 address for the DNS.

proddy, 2024-04-27:
I know that! 

In the very soon to be released Arduino 3 based on ESP IDF 5, the `IPv6Address` class has been removed, which is why I'm adding it back in.

MichaelDvP, 2024-04-27:
The newer arduino 2.xx from Jason also have removed the IPv6Address, part of IPAddress now. I have made some changes to AsyncTCP to use the newer IPAddress. But Arduino 3 actual (since 3 weeks) give me:
```
ESP-ROM:esp32s3-20210327
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x158
load:0x3c970000,len:0x440
load:0x179110ff,len:0x20c0ff
Invalid image block, can't boot.
ets_main.c 329
```
Don't know what it is.

proddy, 2024-04-27:
I did a PR with the IPAddress changes for backwards compatibility. Would be good to compare against what you did. Also with Arduino 3 the mbedtls code has changed so I may need to look into that too. I'll compile, test and see if it works.

MichaelDvP, 2024-04-27:
For the mbedtls use in `WebAuthentification.cpp`
```
#if ESP_ARDUINO_VERSION_MAJOR < 3
  mbedtls_md5_init(&_ctx);
  mbedtls_md5_starts_ret(&_ctx);
  mbedtls_md5_update_ret(&_ctx, data, len);
  mbedtls_md5_finish_ret(&_ctx, _buf);
#else
  mbedtls_md5_init(&_ctx);
  mbedtls_md5_starts(&_ctx);
  mbedtls_md5_update(&_ctx, data, len);
  mbedtls_md5_finish(&_ctx, _buf);
#endif
```

MichaelDvP, 2024-04-28:
Got it working wth your new AsyncTCP. NetworkStatus/Settings `.localIPv6()` is now `.linkLocalIPv6()`

But the dns IP6 issue is not solved. DNS gets over DHCP a ipv6 address, if the DHCP server has it. If we disable IPv6 support, nameservice is not reachable. With ipv6 enabled all is working. Have not tested yet with fixed IP.

proddy, 2024-05-06:
@MichaelDvP do you want to remove the IPv6 option and make it permanent then? I think you know that piece of code better

MichaelDvP, 2024-05-06:
Yes, i think it's better, we already have it permanent enabled for tasmota and idf5 frameworks, I'll delete the settings.