emsesp/EMS-ESP32 #1645: EMS-ESP randomly reverts from firmware 3.6.4 back to firmware 3.5.1

djvanhelmond, 2024-02-29:
### PROBLEM DESCRIPTION

I upgraded EMS-ESP on my Gateway S32 to 3.6.4 on November 28, 2023.
Since then, once in a while the Gateway seems to revert automatically back to 3.5.1. When I upgrade it again, the same happens after a while.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT) - I don't have discord
- [X] Provide the output of http://ems-esp.local/api/system :

```lua
{
  "System Info": {
    "version": "3.5.1",
    "platform": "ESP32",
    "uptime": "000+00:12:01.420",
    "uptime (seconds)": 721,
    "free mem": 103,
    "max alloc": 42,
    "free app": 1,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -71,
    "IPv4 address": "10.7.10.160/255.255.255.0",
    "IPv4 gateway": "10.7.10.1",
    "IPv4 nameserver": "10.7.3.249",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": false,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 768,
    "MQTT queued": 0,
    "MQTT publish fails": 1,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "0",
    "keep alive": 60,
    "clean session": true,
    "entity format": 0,
    "base": "emsesp",
    "discovery prefix": "homeassistant",
    "nested format": 2,
    "ha enabled": false,
    "mqtt qos": 0,
[... 227 lines cut ...]
,,0,2024-02-11T00:00:00Z,2024-02-12T00:00:00Z,2024-02-11T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-12T00:00:00Z,2024-02-13T00:00:00Z,2024-02-12T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-13T00:00:00Z,2024-02-14T00:00:00Z,2024-02-13T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-14T00:00:00Z,2024-02-15T00:00:00Z,2024-02-14T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-15T00:00:00Z,2024-02-16T00:00:00Z,2024-02-15T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-16T00:00:00Z,2024-02-17T00:00:00Z,2024-02-16T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-17T00:00:00Z,2024-02-18T00:00:00Z,2024-02-17T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-18T00:00:00Z,2024-02-19T00:00:00Z,2024-02-18T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-19T00:00:00Z,2024-02-20T00:00:00Z,2024-02-19T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-20T00:00:00Z,2024-02-21T00:00:00Z,2024-02-20T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-21T00:00:00Z,2024-02-22T00:00:00Z,2024-02-21T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-22T00:00:00Z,2024-02-23T00:00:00Z,2024-02-22T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-23T00:00:00Z,2024-02-24T00:00:00Z,2024-02-23T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-24T00:00:00Z,2024-02-25T00:00:00Z,2024-02-24T23:59:00Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-25T00:00:00Z,2024-02-26T00:00:00Z,2024-02-25T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-26T00:00:00Z,2024-02-27T00:00:00Z,2024-02-26T23:59:30Z,3.6.4,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-27T00:00:00Z,2024-02-28T00:00:00Z,2024-02-27T23:59:00Z,3.5.1,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-28T00:00:00Z,2024-02-29T00:00:00Z,2024-02-28T07:44:00Z,3.5.1,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
,,0,2024-02-29T00:00:00Z,2024-02-29T08:06:48.450627596Z,2024-02-29T08:06:30Z,3.5.1,System Info_version,http,System,EMS-ESP Gateway,2_Techniek,0,BBQ Kees
```
### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

MichaelDvP, 2024-02-29:
esp have 2 program partitions, on update the non-active is written and made active on successfull upload. IIn a intermediate version the ota_data was corrupted, resulting in boot from first partition on every reset. 
You can flash with usb, which rewrites the ota_data. 
Or upload twice to write both partitions new. Just upload 3.6.5dev15 from running 3.6.5dev15.

djvanhelmond, 2024-02-29:
I've just loaded 3.6.5 dev 15 to see if the problem persists.

closed 2024-03-09 by proddy

zamojski, 2024-03-25:
I've uploaded the update twice but the issue still persists. Do I really need to flash it with USB? (I thought I was on 3.6.4 already but it seems the issues stays for quite long).
```
2024-03-25 18:22:18.000 I 1: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
2024-03-25 18:22:18.000 I 2: [emsesp] Loading board profile S32
2024-03-25 18:22:18.000 I 3: [system] Current version from settings is 3.6.5-
2024-03-25 18:22:18.000 N 4: [system] Downgrading to version 3.5.1-
2024-03-25 18:22:18.000 I 5: [emsesp] Starting EMS-ESP version 3.5.1 (hostname: ems-esp)
2024-03-25 18:22:18.000 I 6: [dallassensor] Starting Dallas sensor service
2024-03-25 18:22:18.000 I 7: [analogsensor] Starting Analog sensor service
2024-03-25 18:22:18.000 I 8: [emsesp] Loaded EMS device library (113 records)
2024-03-25 18:22:18.000 I 9: [emsesp] Starting Access Point with captive portal on 192.168.4.1
```

proddy, 2024-03-25:
yes, your flash is corrupted and the only way to fix it is a) export settings and any configurations b) flash with USB and c) export back the settings. You're one of the unfortunate people to have a bugged firmware. Sorry about that. But you should be up and running in minutes, provided you have a USB cable

djvanhelmond, 2024-04-26:
I flashed with USB, but didn't resolve the issue.
I assumed it was a hardware issue, bought a new S3 and all is working fine now.