emsesp/EMS-ESP32 #1650: Incorrect switching of "controlmode" entity of CW100 thermostat

SoftVienka, 2024-03-09:
### PROBLEM DESCRIPTION

Switching entity "name": "controlmode", "fullname": "hc1 control mode" of CW100 thermostat via webUI does not work correctly. The best way to show it is the video (link below).

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
`{
    "System Info": {
        "version": "3.6.5-dev.15",
        "uptime": "001+21:44:33.216",
        "uptime (seconds)": 164673,
        "platform": "ESP32",
        "arduino": "Tasmota Arduino v2.0.14",
        "sdk": "4.4.6.240105",
        "free mem": 161,
        "max alloc": 95,
        "used app": 1588,
        "free app": 396,
        "partition": "app0",
        "reset reason": "Software reset CPU / Software reset CPU"
    },
    "Network Info": {
        "network": "WiFi",
        "hostname": "ems-esp",
        "RSSI": -49,
        "TxPower setting": 78,
        "static ip config": true,
        "enable IPv6": false,
        "low bandwidth": false,
        "disable sleep": true,
        "enable MDNS": true,
        "enable CORS": false,
        "AP provision mode": "disconnected",
        "AP security": "wpa2",
        "AP ssid": "ems-esp"
    },
    "NTP Info": {
        "NTP status": "connected",
        "enabled": true,
        "server": "0.sk.pool.ntp.org",
        "tz label": "Europe/Bratislava"
    },
    "OTA Info": {
        "enabled": true,
        "port": 8266
    },
    "MQTT Info": {
        "MQTT status": "disconnected",
        "enabled": false,
        "client id": "ems-esp",
        "keep alive": 60,
        "clean session": false,
        "entity format": 1,
        "base": "ems-esp",
        "discovery prefix": "homeassistant",
        "discovery type": 0,
        "nested format": 1,
        "ha enabled": false,
        "mqtt qos": 0,
        "mqtt retain": false,
        "publish time heartbeat": 60,
        "publish time boiler": 10,
        "publish time thermostat": 10,
        "publish time solar": 10,
        "publish time mixer": 10,
        "publish time other": 10,
        "publish time sensor": 10,
        "publish single": false,
        "publish2command": false,
        "send response": false
    },
    "Syslog Info": {
        "enabled": false
    },
    "Sensor Info": {
        "temperature sensors": 3,
        "temperature sensor reads": 98738,
        "temperature sensor fails": 18,
        "analog sensors": 0,
        "analog sensor reads": 0,
        "analog sensor fails": 0
    },
    "API Info": {
        "API calls": 41074,
        "API fails": 7
    },
    "Bus Info": {
        "bus status": "connected",
        "bus protocol": "HT3",
        "bus telegrams received (rx)": 273659,
        "bus reads (tx)": 65903,
        "bus writes (tx)": 68,
        "bus incomplete telegrams": 84,
        "bus reads failed": 32,
        "bus writes failed": 2,
        "bus rx line quality": 100,
        "bus tx line quality": 100
    },
    "Settings": {
        "board profile": "S32",
        "locale": "en",
        "tx mode": 2,
        "ems bus id": 11,
        "shower timer": false,
        "shower alert": false,
        "hide led": false,
        "notoken api": false,
        "readonly mode": false,
        "fahrenheit": false,
        "dallas parasite": false,
        "bool format": 1,
        "bool dashboard": 1,
        "enum format": 1,
        "analog enabled": true,
        "telnet enabled": true,
        "max web log buffer": 100,
        "web log buffer": 0
    },
    "Devices": [
        {
            "type": "boiler",
            "name": "Logamax Plus GB122/Condense 2300",
            "device id": "0x08",
            "product id": 234,
            "version": "05.06",
            "entities": 70,
            "handlers received": "0xBF 0xC2 0x14 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x04",
            "handlers fetched": "0xE6 0xEA",
            "handlers pending": "0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A",
            "handlers ignored": "0xE7 0x02E0 0x02EA 0xC4 0x2E 0x0DAC 0x17 0xF7"
        },
        {
            "type": "thermostat",
            "name": "RC200/CW100",
            "device id": "0x18",
            "product id": 157,
            "version": "41.09",
            "entities": 41,
            "handlers received": "0x06 0xA2 0x031D",
            "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x02F5 0x023A",
            "handlers pending": "0xA3 0x12 0x13 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x04",
            "handlers ignored": "0xEA 0xBF 0xF7 0xF6 0xEF 0x04 0xE6 0xE9 0xE4 0xF9 0x15 0x14 0x17 0x0241"
        }
    ]
}`

```

### TO REPRODUCE

WebUI - thermostat - entity "hc1 control mode"

### EXPECTED BEHAVIOUR

CW100/CR100 thermostat has "controlmode" with only 4 items to choose from.
1 - External simple (outside basepoint)
2 - External opt. (weather compensated)
3 - Room supply (room)
4 - Room output (power)

### SCREENSHOTS
Here's a video
https://drive.google.com/file/d/1hyoUdbf-JjdBI9MSIYssS9uC3QPv-OFR/view?usp=sharing

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2024-03-09:
very clear, love the video! I'll take a look

MichaelDvP, 2024-03-09:
For this thermostat there something mixed up in code. It shows the option of RC310 (enum_controlmode1) and used other otions (enum_controlmode) for setting. But both are not identcal to your list. We need to know the numbers of the options you can choose on the thermostat. Please log all and change on the thermostat the mode to all options one after another and post the log and tell the order you have changed.

https://github.com/emsesp/EMS-ESP32/blob/e00eb8e64f1926a6eda8075c34afd8cca38a37d6/src/locale_common.h#L323-L324

SoftVienka, 2024-03-09:
I don't know if I got it right.
> We need to know the numbers of the options you can choose on the thermostat.

It can be set on the thermostat as I described above.
In this order:
1. External simple
2. External opt.
3. Room supply
4. Room output

> Please log all and change on the thermostat the mode to all options one after another and post the log and tell the order you have changed.

[log all.txt](https://github.com/emsesp/EMS-ESP32/files/14546332/log.all.txt)

MichaelDvP, 2024-03-09:
Thanks, i see the changes at:
```
T 67: [telegram] Rx: 98 00 FF 00 01 9B 02 03
T 79: [telegram] Rx: 98 00 FF 00 01 9B 01 00
T 109: [telegram] Rx: 98 00 FF 00 01 9B 04 05
T 123: [telegram] Rx: 98 00 FF 00 01 9B 05 04
```
with order 2, 1, 4, 5. The 3 is not available, but we have an enum, so showing `na`

Please check ths works: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

proddy, 2024-03-09:
that should work - thanks Michael for the fix. Verified with standalone too:

`pio run -e standalone -t exec` and in console...
```
test add 0x18 157
call thermostat controlmode “weather compensated” 
```

SoftVienka, 2024-03-09:
Very well.
Now all items are toggled correctly.
The `optimizet` and `simple` items are reversed compared to the thermostat.
(not a problem though)

MichaelDvP, 2024-03-09:
> The optimizet and simple items are reversed compared to the thermostat.

Do you mean the display order or is `optimized` on the thermostat shown as `simple` on ems-esp?
Display order is given by the code numbers. 1-optimized, 2-simple, (3-na), 4-room, 5-power 
Reordering or skipping the 3 is a lot of extra code.

SoftVienka, 2024-03-09:
On the thermostat, the items are in order (as I wrote above):
External simple
External opt.
Room supply
Room output

In the webUI, the items are displayed in the following order:
  (enum_controlmode)
`    "enum": [
        "optimized",
        "simple",
        "n/a",
        "room",
        "power"
    ]`

However, switching "Control type" in the thermostat is fine.
enum "optimized" change to thermostat "External opt."
enum "simple" change to thermostat "External simple"

(I hope google translated it correctly.)

commit fc1e009f (2024-03-10): CW100/RC100 controlmode, fix #1650
closed 2024-03-23 by proddy