emsesp/EMS-ESP32 #1652: Fix missing log entries in web

proddy, 2024-03-09:
There is a chance of log messages being skipped when displayed on the Web specially on slow networks.

MichaelDvP suggested the following:

"Maybe set a longer timeout (300-500ms) in EMS-ESP and post back the message-id on an event received. If the post is received in EMS-ESP, the next message-event can be sent immediately; otherwise, you can just repeat the message after a timeout. It could also help to pack all queued messages into one event as json-array, which reduces the number of events."

which is a perfect solution.

_Originally posted by @MichaelDvP in https://github.com/emsesp/EMS-ESP32/issues/1611#issuecomment-1986835773_

proddy, 2024-03-31:
There was a bug in the React code, which I fixed in the 3.7.0-dev.1, where the EventStream could be initiated more than once when the web page is mounted, which could account for some messages getting lost. I'll see if this helps with this issue by slowing down the WiFi on a real ESP32 next,.

proddy, 2024-03-31:
`WebLogService.h` had REFRESH_SYNC as 50, I changed to 30 and it may have caused #1611 missing logs, then it was changed to 80. I'll change it back to 30 and see if the fix in the web solves the sync issue.

If you compile with `-DEMSESP_PINGTEST` it will send a log message every 1/2 second, which i used for testing.

MichaelDvP, 2024-03-31:
Wow, no missing logs with 30ms on my old notebook with compiler running in background. I think you can close the issue. Also no doubletts when removing the id checks.

proddy, 2024-04-02:
> Wow, no missing logs with 30ms on my old notebook with compiler running in background. I think you can close the issue. Also no doubletts when removing the id checks.

Great, I close this one then. Was a silly thing in how React renders the DOM on page rendering. Before it was easier with onMount().

closed 2024-04-02 by proddy

MichaelDvP, 2024-04-02:
I've also tested with wait-delay completly removed and buffer 100. **Works good.** Next i send all buffer messages at once (in one loop), works also, but maybe this blocks the main loop too long and it takes much memory for TCP buffering. Better leave it by one message per loop.

proddy, 2024-09-11:
@MichaelDvP in 3.7.0-dev.36 I've removed the buffer delay after updating AsyncEventSource with this: https://github.com/mathieucarbou/ESPAsyncWebServer/commit/0163e043747d3ad8cbe70da8e78b88cc652ec12b

with Ping test enabled, and setting the browser to slow 3G network and high CPU throttle I'm not missing any incoming messages.

PR #1683 (2024-03-31): updated screenshots, faster log sync
fixes #1652 hopefully