emsesp/EMS-ESP32 #1656: Can't load the web interface with 3.6.5_dev_17

sosnek1, 2024-03-11:
### PROBLEM DESCRIPTION

Can't load the web interface after upgrading to 3.6.5_dev_17 - page http://ems-esp/ is returning empty result - so have no options to downgrade the firmware. 

Integration with HA via MQTT is working properly

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2024-03-11:
Thanks for reporting; I'll take a look. probably the redirect to index.html in that last update I did

K3vb3rt, 2024-03-12:
Same problem on my side. Not able to load the interface, but MQTT works as expected.

Any manual possibility to get to the interface to do a downgrade?

MichaelDvP, 2024-03-12:
I can confirm that the web navigation is broken. API/mqtt still working. @proddy You should remove the bin, otherwise more people run into this issue.

@K3vb3rt :For uploading firmware see https://emsesp.github.io/docs/Getting-Started/#uploading-the-firmware
If OTA is enabled, (you can check with `http://ems-esp.local/api/system`) you can use espota. otherwise you'll need usb flashing.

@proddy For such a case it would be nice to go back to old version by booting to other partition (if not empty). We can add a option to telnet restart command `restart` and `restart old` and/or use button long-pressed. But i think telnet is ok, we can use serial console if wifi is also broken.

proddy, 2024-03-12:
I already removed the release and will fix the broken build this week when I'm home. A fallback to the previous build on the other partition is good - how could we easily add this? Perhaps a script?

MichaelDvP, 2024-03-12:
I thought of a simple extension to the restart command, like:
```
void System::system_restart(const bool other_partition) {
    if (other_partition) {
        LOG_INFO("Restarting EMS-ESP to other partition...");
    } else {
        LOG_INFO("Restarting EMS-ESP...");
    }
    Shell::loop_all();
    delay(1000); // wait a second
#ifndef EMSESP_STANDALONE
    if (other_partition) {
        const esp_partition_t * partition = esp_ota_get_next_update_partition(NULL);
        if (partition) {
            uint64_t buffer;
            esp_partition_read(partition, 0, &buffer, 8);
            if (buffer != 0xFFFFFFFFFFFFFFFF) { // partition not empty
                esp_ota_set_boot_partition(partition);
            }
        }
    }
    ESP.restart();
#endif
}
```
Works already in restart_service for the boot to loader (with find of factory partiton)
I'll make a PR.

K3vb3rt, 2024-03-12:
> I can confirm that the web navigation is broken. API/mqtt still working. @proddy You should remove the bin, otherwise more people run into this issue.
>
> @K3vb3rt :For uploading firmware see [...]

> I can confirm that the web navigation is broken. API/mqtt still working. @proddy You should remove the bin, otherwise more people run into this issue.
>
> @K3vb3rt :For uploading firmware see [...]

Thank you so much! Downgraded via USB with ESP-Tool to latest official.

proddy, 2024-03-12:
should be fixed now, sorry about that. I should have tested better. Good job you had an older version of the firmware. I'll create a new issue for the fall-back partition.

closed 2024-03-12 by proddy

proddy, 2024-03-13:
### IMPORTANT 

If you were unlucky enough to grab the 3.6.5-dev17 build in the last 24hrs, it had a horrible bug preventing the Web UI from loading in the browser. Luckily it's now been fixed., but you're left with a version of EMS-ESP you can't easily upgrade. 

These are the options to get it back operational:
 - use OTA if you still have enabled. You can check by going into the Telnet Console and typing 'call system info'. It's disabled by default. The commands for using ota are in https://emsesp.github.io/docs/Getting-Started/#first-time-install
 - use the EMS-ESP Flasher, but take the latest v1.0.2 version which you can find here: https://github.com/emsesp/EMS-ESP-Flasher/actions/runs/8270785787 or otherwise in https://github.com/emsesp/EMS-ESP-Flasher/releases

following any of these two options above will not erase your settings.

My sincere apologies for any pain caused. It was a rush job on my side and I should be taken more time to thoroughly test.

Paul

commit be6bb1de (2024-03-12): fixes #1656