emsesp/EMS-ESP32 #1660: [Junkers Cerapur Compact] Missing 0x22 telegram interpreter

bszczepanik, 2024-03-13:
### PROBLEM DESCRIPTION

There is a lot of missing parameters for boiler Junker Cerapur Compact. Device is recognized as "Condes 2500/Logamax/Logomatic/Cerapur Top/Greenstart/Generic HT3, so that seems to be correct. I found some telegrams in the log with missing parameters value.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{
  "System Info": {
    "version": "3.6.4",
    "platform": "ESP32",
    "uptime": "000+00:01:09.563",
    "uptime (seconds)": 69,
    "free mem": 171,
    "max alloc": 107,
    "free app": 419,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -74,
    "IPv4 address": "192.168.1.56/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "1.1.1.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 155,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": true,
    "publish time heartbeat": 10,
    "publish time boiler": 0,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 1,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 66,
    "bus reads (tx)": 19,
    "bus writes (tx)": 1,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "MH-ET",
    "locale": "pl",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": true,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 2,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 25,
    "web log buffer": 25
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "43.10",
      "entities": 69,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x2A"
    },
    {
      "type": "controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "43.10",
      "entities": 0
    }
  ]
}

```

### TO REPRODUCE

Change some parameter on the boiler. See the telegram coming.

### EXPECTED BEHAVIOUR

Log results:
```
000+00:18:08.351 T 667: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x22), data: 01 33 46 01 00 03
000+00:18:32.138 T 675: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x22), data: 01 33 46 01 00 01
```
```
000+00:18:55.361 T 686: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x22), data: 01 37 45 01 00 01
000+00:19:20.367 T 700: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x22), data: 01 37 45 01 00 00
```

### SCREENSHOTS

Not necessary.

### ADDITIONAL CONTEXT

As I see, the last byte of the telegram is the payload, the value of the parameter being modified.

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2024-03-13:
Thanks. Could you try creating custom entities for these to validate they work and then send us the screenshots so we can add the code?

https://emsesp.github.io/docs/Troubleshooting/#im-missing-certain-data-from-an-ems-device

bszczepanik, 2024-03-13:
Sure! I will do my best. I need some time for reverse engineering. I will let you know here if I have some results.

bszczepanik, 2024-03-14:
Any hint how to create custom entities with that kind of telegram?
[image]

proddy, 2024-03-14:
The approach to adding entities to EMS-ESP is
 - first determine what information you need. If you don't know, you probably don't need it
 - Use the EMS device (boiler, thermostat, etc) to make the change manually whilst recording the telegrams
 - Identify the telegram type and offset for the data point
 - create a custom entity to validate
 - report it to the dev team via GitHub

This is also documented in the wiki.

So, this 0x22 telegram - what is it and why do you need it?

bszczepanik, 2024-03-14:
I understand, but here I see different problem. In 0x22 telegram I got two (or many) different variables.

In the telegrams below, I've got information that I want, comfort/eco mode for DHW. Variable type is BOOL. I see it in the last byte. (01 -> 00).
```
000+00:18:55.361 T 686: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x22), data: 01 37 45 01 00 01
000+00:19:20.367 T 700: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x22), data: 01 37 45 01 00 00
```

For reverese engineering tests, I changed other variable on the boiler (some delay time for DHW, integer variable) and I've got this telegrams with data on the last position (03 -> 01):
```
000+00:18:08.351 T 667: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x22), data: 01 33 46 01 00 03
000+00:18:32.138 T 675: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x22), data: 01 33 46 01 00 01
```

As I see in documentation, Message ID is inside a telegram, I think it is byte 2 (position 3). 0x45 in the first example, 0x46 in the second.

The point is, How can I define custiom entity, if I have the same telegram number for more than one data point.

proddy, 2024-03-14:
you can create as many custom entities as you want for the same telegram. If you make them read only its harmless so you can just play around until you get the right results back.

For DHW comfort (hot/eco/intelligent etc) should already be reported (called wwComfort) in the 0x33 or 0xEA telegrams. Are you not seeing it?

MichaelDvP, 2024-03-15:
> I changed other variable on the boiler 

If you specify what exactly you have changed with what values. maybe we can help.
> Message ID is inside a telegram, I think it is byte 2 

Device-id is 0x08 for boiler, Type-id is 0x22, offset starts with 0 for first byte, read the last byte is offset 5.

bszczepanik, 2024-03-17:
Ok I found that comfort/eco mode of DHW for my boiler is already configured. But there is no reaction from the boiler to changes in parameters.
```
000+00:18:34.574 I 593: [command] Calling command 'boiler/wwactivated' (activated) with value on
000+00:18:34.660 N 594: [emsesp] Me(0x0B) -W-> boiler(0x08), UBAParameterWW(0x33), data: FF (offset 1)
000+00:18:34.727 N 595: [emsesp] boiler(0x08) -B-> All(0x00), UBAParameterWW(0x33), data: 08 00 30 00 00 00 00 02 4B 00 FF FF
```

If I change this manually on the boiler I see change:
```
000+00:21:13.104 N 612: [emsesp] boiler(0x08) -B-> All(0x00), UBAParameterWW(0x33), data: 08 FF 30 00 00 00 00 02 4B 00 FF FF
```

How can I turn on possibility to change this parameter?

bszczepanik, 2024-03-17:
Maybe that problem is because the boiler wait for command from exact Device ID. I know that my boiler works with CR100 controller. I found a list of device IDs where this controller is:
[image]

Can I change here BUS ID for any custom? I would try 0x10 and 0x18
[image]

proddy, 2024-03-24:
you could export the settings (`emsesp_settings.json`) modify "Settings" "ems_bus_id" with the new value to 16 (0x10), save, and import it back. It's a hack but see if it works.

proddy, 2024-06-25:
@bszczepanik did it work in the end?

bszczepanik, 2024-06-25:
I apologize, I forgot aboute this issue. I gave up with this. I achieved my goal other way. I warm up hot water when lights in the bathroom turn on

closed 2024-06-25 by proddy