emsesp/EMS-ESP32 #1664: Regular poll requests for invalid telegrams are within error log

tp1de, 2024-03-15:
I installed a second gateway at my vacation house together with a new boiler and thermostat.
Boiler is Netfit Trendline II (product id 123) and thermostat Netfit Moduline 2050 (product id 157).

In the log I get regular errors on poll reads. That is not what I am used to have at my home location.
I assume that this telegram types are not supported by the hardware (boiler & thermostat).

2024-03-15 13:46:45.915 E 237: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 01 94
2024-03-15 13:51:23.040 E 238: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 01 40
2024-03-15 13:53:29.599 E 240: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 01 AF
2024-03-15 13:53:48.056 E 241: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 01 94
2024-03-15 13:58:29.541 E 242: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 01 91
2024-03-15 14:03:59.340 E 244: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 03 71
2024-03-15 14:04:49.240 E 245: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 14 00 1B
2024-03-15 14:15:24.692 E 246: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 01 94
2024-03-15 14:26:18.498 E 247: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 26 00 1B
2024-03-15 14:41:18.287 E 248: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 14 00 1B
2024-03-15 14:43:40.835 E 249: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 01 91
2024-03-15 14:46:39.772 E 250: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 01 CC
2024-03-15 14:53:53.029 E 251: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 BB 00 1B
2024-03-15 15:02:43.403 E 252: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 01 B9
2024-03-15 15:07:36.884 E 254: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 33 00 1B
2024-03-15 15:21:24.781 E 255: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 01 B9
2024-03-15 15:31:16.134 E 256: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 01 B9
2024-03-15 15:31:46.040 E 257: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 98 FF 00 19 01 91

@MichaelDvP you asked on Discord for more information. What do you need?

MichaelDvP, 2024-03-15:
Are this all logged errors, or only some of them? 
There are some boiler requests for telegrams 0x14, 0x26, 0x14, 0x33. The two 0x14 happen 14:04 and 14:41, so it is not every fetch (every minute). Have you tried `read 8 14` if you get an answer?
Same for the thermostat requests, often 0x294 and 0x2B9 and some others.
Please post the system info. Have you tried with a different tx-mode (what mode are you using)?

tp1de, 2024-03-16:
I was using EMS+ until now. I changed now to EMS mode. 
Seems to work better.
I need to observe a bit longer. I had a reboot during night.

The Moduline 2050 thermostat seems to be quite different from RC200.

tp1de, 2024-03-16:
@MichaelDvP anything changed in 3.6.5-dev.18?

I do not get any errors anymore.

proddy, 2024-03-18:
In dev18 Michael did this PR (https://github.com/emsesp/EMS-ESP32/pull/1661) to ignore fetches if they're empty after 5 minutes, to stop the log spamming. this is the only change.

MichaelDvP, 2024-03-18:
But this change have no effect to the above listed errors. The errors happen random and there are a lot of valid receives between. The fetch is only canceled if there is not a single valid telegram of this type in 5 minutes. I've asked for system info (i always ask for it and get no answer). In this we can see if the telegrams are lsted as received.

tp1de, 2024-03-18:
[systeminfo.txt](https://github.com/emsesp/EMS-ESP32/files/14638639/systeminfo.txt)

tp1de, 2024-03-18:
I am using a S3 gateway from @bbqkees. I am actually getting much less errors now since I I changed to EMS, but nearly every day a WiFi or MQTT or NTP disconnect. Reconnecting is working. I am not in my vacation house anymore and I can observe only from remote. (Home assistant with MQTT  is installed on a Raspberry Pi 4 locally)
Everything is running stable - I just observed the differences to my home setup.

proddy, 2024-03-24:
ok, I'll close this for now then.

closed 2024-03-24 by proddy