emsesp/EMS-ESP32 #1665: Refresh UI - moving settings to one location

proddy, 2024-03-17:
Moving some of the web screens around, so focus is on dashboards and settings are all behind one section.

proddy, 2024-03-17:
[image]

proddy, 2024-03-17:
[image]

bbqkees, 2024-03-18:
Looks great. Now that I think of it it makes more sense to do it this way.
Did you 'dock' the Devices tabs back?

proddy, 2024-03-18:
That's a good idea, @MichaelDvP—having all the Statuses in one place. I'll think of an elegant way to do that.
yes @bbqkees the Devices and Sensors are no longer tabs, so just one click away

proddy, 2024-03-22:
Latest is in a PR - https://github.com/emsesp/EMS-ESP32/pull/1667

closed 2024-03-24 by proddy