emsesp/EMS-ESP32 #1666: EMS bus disconnected

mjvanderveen, 2024-03-22:
### PROBLEM DESCRIPTION

Frequent EMS bus disconnects that do not  get resolved without a full restart. Huge daily data gaps as a result.

EMS bus disconnected. If this warning still persists after a few seconds please check settings and board profile

The esp-32 is connected to the nefit bosch compress 3000i heat pump using the dc connector (not wired to the bus).

### REQUESTED INFORMATION

```lua
  System information output here:

{
  "System Info": {
    "version": "3.6.4",
    "platform": "ESP32",
    "uptime": "000+13:59:45.204",
    "uptime (seconds)": 50385,
    "free mem": 186,
    "max alloc": 107,
    "free app": 419,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -80,
    "IPv4 address": "192.168.178.56/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "253.0.0.0",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": true,
    "server": "pool.ntp.org",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 886,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 2,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": true,
    "syslog started": true,
    "syslog level": "info",
    "syslog ip": "192.168.178.72",
    "syslog queue": 0
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "disconnected"
  },
  "Settings": {
    "board profile": "E32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 100,
    "web log buffer": 8
  }
}

```

### TO REPRODUCE

No active steps. Error is random within usually 24 hours.

### EXPECTED BEHAVIOUR

The bus to stay connected to my heatpump NEFIT BOSCH compress 3000i. 

### SCREENSHOTS

[image]

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

2024-03-21,15:56:29,Gegevens,ems-esp.fritz.box,kern,1,2024-03-21T15:56:29.799547+1:00 ems-esp emsesp - - - 000+00:00:20.038 I 14: Stopping Access Point2024-03-21,15:56:27,Gegevens,ems-esp.fritz.box,kern,1,2024-03-21T15:56:27.026902+1:00 ems-esp system - - - 000+00:00:17.265 I 13: NTP connected2024-03-21,15:56:27,Gegevens,ems-esp.fritz.box,daemon,1,2024-03-21T15:57:12.136469+1:00 ems-esp mqtt - - - 000+00:00:16.467 I 12: MQTT connected2024-03-21,15:56:27,Gegevens,ems-esp.fritz.box,kern,1,2024-03-21T15:57:11.919571+1:00 ems-esp emsesp - - - 000+00:00:16.221 I 11: mDNS responder service started2024-03-21,15:56:27,Gegevens,ems-esp.fritz.box,kern,1,"2024-03-21T15:57:11.891078+1:00 ems-esp emsesp - - - 000+00:00:16.221 I 10: WiFi connected with IP=192.168.178.56, hostname=ems-esp"2024-03-21,15:56:27,Gegevens,ems-esp.fritz.box,kern,1,2024-03-21T15:57:11.889571+1:00 ems-esp emsesp - - - 000+00:00:16.221 I 9: Starting NTP service2024-03-21,15:56:27,Waarschuwing,ems-esp.fritz.box,kern,1,- ems-esp emsesp - - - 000+00:00:16.000 W 8: WiFi disconnected. Reason: assoc leave (8)2024-03-21,15:56:27,Waarschuwing,ems-esp.fritz.box,kern,1,- ems-esp emsesp - - - 000+00:00:12.992 W 7: WiFi disconnected. Reason: assoc leave (8)2024-03-21,15:56:27,Gegevens,ems-esp.fritz.box,kern,1,- ems-esp emsesp - - - 000+00:00:10.026 I 6: Starting Access Point with captive portal on 192.168.4.12024-03-21,15:56:26,Waarschuwing,ems-esp.fritz.box,kern,1,- ems-esp emsesp - - - 000+00:00:09.930 W 5: WiFi disconnected. Reason: assoc leave (8)2024-03-21,15:56:26,Waarschuwing,ems-esp.fritz.box,kern,1,- ems-esp emsesp - - - 000+00:00:09.291 W 4: WiFi disconnected. Reason: unknown (39)2024-03-21,15:56:26,Waarschuwing,ems-esp.fritz.box,kern,1,- ems-esp emsesp - - - 000+00:00:03.929 W 3: WiFi disconnected. Reason: assoc leave (8)2024-03-21,15:56:26,Gegevens,ems-esp.fritz.box,kern,1,- ems-esp emsesp - - - 000+00:00:00.000 I 2: Loaded EMS device library (117 records)2024-03-21,15:56:26,Gegevens,ems-esp.fritz.box,daemon,1,- ems-esp analogsensor - - - 000+00:00:00.000 I 1: Starting Analog sensor service2024-03-21,15:56:26,Gegevens,ems-esp.fritz.box,daemon,1,- ems-esp temperaturesensor - - - 000+00:00:00.000 I 0: Starting Temperature sensor service

proddy, 2024-03-22:
It could be hardware. How frequent are the blackouts? Were you seeing good Rx and Tx line quality when it was working and connected?

I'll change the 3.6.5 version to show the last EMS stats in the 'Bus Info' of the System Info export, even if the bus is disconnected. Which would give us more data. For example:

```
  "Bus Info": {
    "bus status": "disconnected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 758230,
    "bus reads (tx)": 195335,
    "bus writes (tx)": 2,
    "bus incomplete telegrams": 181,
    "bus reads failed": 223,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
```

BTW, I also noticed your WiFi RSSI strength is super low, not that it should affect the EMS bus, but you may find you'll get performance issues or more frequent network restarts.

mjvanderveen, 2024-03-22:
"Bus Info":{"bus status":"connected","bus protocol":"HT3","bus telegrams received (rx)":326,"bus reads (tx)":114,"bus writes (tx)":0,"bus incomplete telegrams":1,"bus reads failed":0,"bus writes failed":0,"bus rx line quality":100,"bus tx line quality":100}

The failure is at least once daily.

MichaelDvP, 2024-03-23:
> The esp-32 is connected to the nefit bosch compress 3000i heat pump using the dc connector (not wired to the bus).

I think you mean the service-key connector, not the dc-connector. Make sure the plugs are fully plugged in. Some cables have a too large housing and make bad connections. 

Also seems you have restarts, a bus disconnect does not clear the device list.

mjvanderveen, 2024-03-24:
I automatically restart it at 6.30am every day just to ensure after a failure it comes back at least once a day.

And i manually restart it when there are issues. So yes, plenty of restarts.

I will double check the plug.

bbqkees, 2024-03-26:
It seems you are using the 'old' E32. 
The problem with this Gateway is that while the ESP32 itself has more than enough processing power for EMS-ESP, for the latest firmware versions the 512KB RAM memory is not really sufficient anymore for using with large EMS systems like heat pumps with 120+ entities, MQTT Discovery etc etc.
The latest Gateways have 8MB of PSRAM. 
Here are some tips to free up some memory;
https://emsesp.github.io/docs/Troubleshooting/#it-may-be-memory-related

If you would like to upgrade to a new Gateway I can offer a discount. Please respond to the order confirmation email of the original order if you still have it or use the contact form in the webshop.

Cedi00123, 2024-05-27:
Hello,

I have a Issue with the BUS-Connection. I connect the Bus to the correct port but it have very much issues on the rx.

Can someone help me?

I'm using a esp32 "MH-ET Live 32" on a "https://github.com/rocksolidsr/esp-ems-board/tree/main" to a "Logamatic MC110"

[image]

[image]

proddy, 2024-05-27:
@Cedi00123 please create a new discussion topic (and not an issue) for this. And attach the system info. It looks like lots of things wrong. Both Rx and Tx are not working. The circuit you built may not be working (it was designed for an esp8266), or the pins are switched or you're using blocked GPIOs.

closed 2024-05-27 by proddy