emsesp/EMS-ESP32 #1673: 3.7.0-test.0 (ESP32) - Error Menu -> System

SoftVienka, 2024-03-26:
### PROBLEM DESCRIPTION

After clicking Menu -> System, an error message is displayed (only SK language):
```
Unexpected Application Error!

t.reduce is not a function

TypeError: t.reduce is not a function
    at http://192.168.1.240/assets/vendor-OahXKFOG.js:163:84379
    at http://192.168.1.240/assets/vendor-OahXKFOG.js:163:84640
    at Array.map (<anonymous>)
    at http://192.168.1.240/assets/vendor-OahXKFOG.js:163:84001
    at http://192.168.1.240/assets/vendor-OahXKFOG.js:163:84676
    at http://192.168.1.240/assets/vendor-OahXKFOG.js:163:84705
    at b.to [as constructor] (http://192.168.1.240/assets/index-CG2_D2J_.js:7:108407)
    at b.I [as render] (http://192.168.1.240/assets/vendor-OahXKFOG.js:1:9023)
    at z (http://192.168.1.240/assets/vendor-OahXKFOG.js:1:6532)
    at w (http://192.168.1.240/assets/vendor-OahXKFOG.js:1:1476)
```
**3.7.0-dev.0 (ESP32) - no problem.**

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

{
  "System Info": {
    "version": "3.7.0-test.0",
    "uptime": "000+00:28:10.246",
    "uptime (seconds)": 1690,
    "platform": "ESP32",
    "arduino": "Tasmota Arduino v2.0.14",
    "sdk": "4.4.6.240105",
    "free mem": 153,
    "max alloc": 87,
    "free caps": 102,
    "used app": 1606,
    "free app": 378,
    "partition": "app0",
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -53,
    "TxPower setting": 78,
    "static ip config": true,
    "enable IPv6": true,
    "low bandwidth": false,
    "disable sleep": true,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "0.sk.pool.ntp.org",
    "tz label": "Europe/Bratislava"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time water": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 3,
    "temperature sensor reads": 1011,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 432,
    "API fails": 6
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 2488,
    "bus reads (tx)": 559,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "en",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 100,
    "web log buffer": 0
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Logamax Plus GB122/Condense 2300",
      "device id": "0x08",
      "product id": 234,
      "version": "05.06",
      "entities": 76,
      "handlers received": "0xBF 0xC2 0x14 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x04",
      "handlers fetched": "0xE6 0xEA 0x28",
      "handlers pending": "0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0xC4 0xE7 0x02E0 0x02EA 0x2E 0x0DAC"
    },
    {
      "type": "thermostat",
      "name": "RC200/CW100",
      "device id": "0x18",
      "product id": 157,
      "version": "41.09",
      "entities": 43,
      "handlers received": "0x06 0xA2 0x031D",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x02F5 0x023A",
      "handlers pending": "0xA3 0x12 0x13 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02",
      "handlers ignored": "0xEA 0xBF"
    }
  ]
}
```

### TO REPRODUCE

Set SK language for WebUI

### EXPECTED BEHAVIOUR

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

MichaelDvP, 2024-03-26:
Seems the new pluralization here:
https://github.com/emsesp/EMS-ESP32/blob/8ef8eeb9ec45c1080b9164ba5ef87e85252d4c19/interface/src/i18n/sk/index.ts#L94-L100
is wrong. Can you help to fix it?
In dev-build it is:
https://github.com/emsesp/EMS-ESP32/blob/6cab0202415a6f145297812dbd2be46aefe664f1/interface/src/i18n/sk/index.ts#L94-L100

SoftVienka, 2024-03-26:
These plurals

>  NUM_DEVICES: '{num} Zariaden{{í|ie|ia|ia|í}}', 
 NUM_TEMP_SENSORS: '{num} Teplotn{{ých|ý|é|é|ých}} snímač{{ov||e|e|ov}}', 
 NUM_ANALOG_SENSORS: '{num} Analogov{ých|ý|é|é|ých}} snímač{{ov||e|e|ov}}', 
 NUM_DAYS: '{num} d{{ní|eň|ní|ní|ní}}', 
 NUM_SECONDS: '{num} sek{{únd|unda|undy|undy|únd}}', 
 NUM_HOURS: '{num} hod{{ín|ina|iny|iny|ín}}', 
 NUM_MINUTES: '{num} minú{{t|ta|ty|ty|t}}',

should be correct. I don't know why it doesn't work.

In the Polish translation it is functional.

>   NUM_DEVICES: '{num} urządze{{ń|nie|nia|nia|ń}} EMS',
  NUM_TEMP_SENSORS: '{num} czujni{{ków|k|ki|ki|ków}} temperatury',
  NUM_ANALOG_SENSORS: '{num} inn{{ych|e|e|e|ych}} urządze{{ń|nie|nia(two)|nia|ń}} podłączon{{ych|e|e|e|ych}} do EMS-ESP',
  NUM_DAYS: '{num} d{{ni|zień|ni|ni|ni}}',
  NUM_SECONDS: '{num} sekun{{d|da|dy|dy|d}}',
  NUM_HOURS: '{num} godzi{{n|na|ny|ny|n}}',
  NUM_MINUTES: '{num} minu{{t|ta|ty|ty|t}}',

MichaelDvP, 2024-03-26:
It does not like the empty field, it's working with:
```
  NUM_TEMP_SENSORS: '{num} Teplotn{{ých|ý|é|é|ých}} sníma{{čov|č|če|če|čov}}',
  NUM_ANALOG_SENSORS: '{num} Analogov{ých|ý|é|é|ých}} sníma{{čov|č|če|če|čov}}',
```

SoftVienka, 2024-03-26:
So there was a dog buried here.
Moreover was missing there one compound parenthesis
...Analogov{ých...
...Analogov{{ých...
Now it doesn't throw the error anymore.

But ...
- 0 sekúnd - OK
- 1 sekunda - OK
- 2 sekundy - OK
- 3 sekundy - OK
- 4 sekundy - OK
- 5 sek   -   NO OK
- 6 sek   -   NO OK
etc, etc

It will probably be the same with other plurals.  

I tried to transcribe this part of the translation from the Polish translation.
Same error. 0-1-2-3-4 OK, 5 or more NO OK.

proddy, 2024-03-27:
Could this be a bug in the i18n library that we should report? or is the "dog" somewhere else?

MichaelDvP, 2024-03-27:
The plurals here have 5 fields, but full spec is 6 fields `{{zero, one, two, few, many, other}}`, see https://github.com/ivanhofer/typesafe-i18n/tree/main/packages/runtime#plural
But not all are used, language specific it is matched to fewer cases: https://www.unicode.org/cldr/charts/43/supplemental/language_plural_rules.html
For sk we have `{{one, few, may, others}}`, with `few` matches 2-4, `many` matches flowts and `others` matches 0 and 5-up
Please check with 4 cases or use all 6.

SoftVienka, 2024-03-27:
Good work.
Embedded. [index.ts.SK2.txt](https://github.com/emsesp/EMS-ESP32/files/14772326/index.ts.SK2.txt)

MichaelDvP, 2024-03-28:
@SoftVienka Your latest changes are now in the official [dev](https://github.com/emsesp/EMS-ESP32/releases/tag/latest), please check and close this issue if ok.

closed 2024-03-28 by SoftVienka