emsesp/EMS-ESP32 #1677: MS200 Bosch Heat Assist valve

jeffmando, 2024-03-28:
i have an ms200 module with a cs200 control unit from bosch. 2 collector arrays are connected to it and the system is used as heating support. the valve status is also displayed, but it does not change even if the valve is open

I use this configuration (GHK):
[image]

in Ems-ESP i have the heat assist valve m1 should this be vs1 or vs 3 ?

and the temprature ts8 is not displayed in ems-esp.

Here ist my log:
```
002+06:29:26.455 TRACE 132245: [emsesp] solar(0x30) -B-> All(0x00), SM100Status2(0x036A), data: 03 03 03 03 03 03 03 03 03 03 04 03
002+06:29:48.516 TRACE 132246: [emsesp] solar(0x30) -B-> All(0x00), SM100Energy(0x038E), data: 00 00 2F 44 00 00 13 C4 00 00 04 19
002+06:29:49.365 TRACE 132247: [emsesp] thermostat(0x10) -B-> All(0x00), RCTime(0x06), data: 18 03 0F 1C 09 24 03 00 10 FF 00
002+06:29:54.897 TRACE 132248: [emsesp] solar(0x30) -B-> All(0x00), SM100Config(0x0366), data: 05 69 01 03 00
002+06:30:02.867 TRACE 132249: [emsesp] solar(0x30) -B-> All(0x00), UBADevices(0x07), data: 08 01 00 00 00 01 00 00 00 00 00 00 00 00 00
002+06:30:05.072 TRACE 132250: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Settings(0x0240), length: 0x19
002+06:30:05.137 TRACE 132251: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Settings(0x0240), data: 32 00 01 00 00 00 00 00 00 00 00 00 00 00 00
002+06:30:05.372 TRACE 132252: [emsesp] Me(0x0B) -R-> solar(0x30), SM100SystemConfig(0x0358), length: 0x19
002+06:30:05.425 TRACE 132253: [emsesp] solar(0x30) -W-> Me(0x0B), SM100SystemConfig(0x0358), data: FF 00 FF 00 FF 00 00 00 00 00 00 00 FF 00 00 00 00 00 00 FF 00 FF 01 00 00
002+06:30:05.562 TRACE 132254: [emsesp] Me(0x0B) -R-> solar(0x30), SM100SystemConfig(0x0358), length: 0x19 (offset 25)
002+06:30:05.614 TRACE 132255: [emsesp] solar(0x30) -W-> Me(0x0B), SM100SystemConfig(0x0358), data: 00 00 00 00 00 00 (offset 25)
002+06:30:05.772 TRACE 132256: [emsesp] Me(0x0B) -R-> solar(0x30), SM100CircuitConfig(0x035A), length: 0x19
002+06:30:05.810 TRACE 132257: [emsesp] solar(0x30) -W-> Me(0x0B), SM100CircuitConfig(0x035A), data: 78 05 00 5A 28 00 01 32 64 00 00 00 5A 0C
002+06:30:06.028 TRACE 132258: [emsesp] Me(0x0B) -R-> solar(0x30), SM100Circuit2Config(0x035D), length: 0x19
002+06:30:06.057 TRACE 132259: [emsesp] solar(0x30) -W-> Me(0x0B), SM100Circuit2Config(0x035D), data: 00 00 01 32 64 0C 5A
002+06:30:06.201 TRACE 132260: [emsesp] Me(0x0B) -R-> solar(0x30), SM100CollectorConfig(0x0380), length: 0x19
002+06:30:06.250 TRACE 132261: [emsesp] solar(0x30) -W-> Me(0x0B), SM100CollectorConfig(0x0380), data: 5A 64 00 00 28 01 00 00 01
002+06:30:06.443 TRACE 132262: [emsesp] Me(0x0B) -R-> solar(0x30), SM100Energy(0x038E), length: 0x19
002+06:30:06.478 TRACE 132263: [emsesp] solar(0x30) -W-> Me(0x0B), SM100Energy(0x038E), data: 00 00 2F 44 00 00 13 C4 00 00 04 19
002+06:30:06.668 TRACE 132264: [emsesp] Me(0x0B) -R-> solar(0x30), SM100Time(0x0391), length: 0x19
002+06:30:06.725 TRACE 132265: [emsesp] solar(0x30) -W-> Me(0x0B), SM100Time(0x0391), data: 00 00 07 E9 00 00 00 00 00 00 0F B9 00 00 00 00 00 00 10 BF 00 00 00 00 00
002+06:30:06.936 TRACE 132266: [emsesp] Me(0x0B) -R-> solar(0x30), SM100Time(0x0391), length: 0x19 (offset 25)
002+06:30:06.989 TRACE 132267: [emsesp] solar(0x30) -W-> Me(0x0B), SM100Time(0x0391), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 24)
002+06:30:06.989 TRACE 132268: [emsesp] Me(0x0B) -R-> solar(0x30), SM100Config1(0x035F), length: 0x19
002+06:30:07.219 TRACE 132270: [emsesp] Me(0x0B) -R-> solar(0x30), SM100HeatAssist(0x035C), length: 0x19
002+06:30:07.433 TRACE 132273: [emsesp] solar(0x30) -W-> Me(0x0B), SM100Differential(0x0361), data: 3C 5A 14 1E 3C
002+06:30:25.399 TRACE 132274: [emsesp] solar(0x30) -B-> All(0x00), SM100Monitor(0x0362), data: 02 36 00 CE 80 00 00 C1 01 2C 01 80 80 00 80 00 80 00 80 00 80 00 01 A8
002+06:30:25.576 TRACE 132275: [emsesp] solar(0x30) -B-> All(0x00), SM100Monitor(0x0362), data: 80 00 (offset 24)
002+06:30:25.961 TRACE 132276: [emsesp] solar(0x30) -B-> All(0x00), SM100Monitor2(0x0363), data: 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00
002+06:30:26.171 TRACE 132277: [emsesp] solar(0x30) -B-> All(0x00), SM100Status(0x0364), data: 64 00 00 04 00 00 FF 00 00 64 00 00 64 00 00 00 00 00
002+06:30:26.625 TRACE 132278: [emsesp] solar(0x30) -B-> All(0x00), SM100Config(0x0366), data: 05 69 01 03 00
002+06:30:26.804 TRACE 132279: [emsesp] solar(0x30) -B-> All(0x00), ?(0x0368), data: 0C 00
002+06:30:49.368 TRACE 132282: [emsesp] thermostat(0x10) -B-> All(0x00), RCTime(0x06), data: 18 03 0F 1C 0A 24 03 00 10 FF 00
002+06:30:55.467 TRACE 132283: [emsesp] solar(0x30) -B-> All(0x00), SM100Config(0x0366), data: 05 69 01 03 00
002+06:31:03.589 TRACE 132284: [emsesp] solar(0x30) -B-> All(0x00), UBADevices(0x07), data: 08 01 00 00 00 01 00 00 00 00 00 00 00 00 00
002+06:31:05.072 TRACE 132285: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Settings(0x0240), length: 0x19
002+06:31:05.141 TRACE 132286: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Settings(0x0240), data: 32 00 01 00 00 00 00 00 00 00 00 00 00 00 00
002+06:31:05.313 TRACE 132287: [emsesp] Me(0x0B) -R-> solar(0x30), SM100SystemConfig(0x0358), length: 0x19
002+06:31:05.359 TRACE 132288: [emsesp] solar(0x30) -W-> Me(0x0B), SM100SystemConfig(0x0358), data: FF 00 FF 00 FF 00 00 00 00 00 00 00 FF 00 00 00 00 00 00 FF 00 FF 01 00 00
002+06:31:05.581 TRACE 132289: [emsesp] Me(0x0B) -R-> solar(0x30), SM100SystemConfig(0x0358), length: 0x19 (offset 25)
002+06:31:05.610 TRACE 132290: [emsesp] solar(0x30) -W-> Me(0x0B), SM100SystemConfig(0x0358), data: 00 00 00 00 00 00 (offset 25)
002+06:31:05.772 TRACE 132291: [emsesp] Me(0x0B) -R-> solar(0x30), SM100CircuitConfig(0x035A), length: 0x19
002+06:31:05.812 TRACE 132292: [emsesp] solar(0x30) -W-> Me(0x0B), SM100CircuitConfig(0x035A), data: 78 05 00 5A 28 00 01 32 64 00 00 00 5A 0C
002+06:31:06.000 TRACE 132293: [emsesp] Me(0x0B) -R-> solar(0x30), SM100Circuit2Config(0x035D), length: 0x19
002+06:31:06.029 TRACE 132294: [emsesp] solar(0x30) -W-> Me(0x0B), SM100Circuit2Config(0x035D), data: 00 00 01 32 64 0C 5A
002+06:31:06.221 TRACE 132295: [emsesp] Me(0x0B) -R-> solar(0x30), SM100CollectorConfig(0x0380), length: 0x19
002+06:31:06.256 TRACE 132296: [emsesp] solar(0x30) -W-> Me(0x0B), SM100CollectorConfig(0x0380), data: 5A 64 00 00 28 01 00 00 01
002+06:31:06.473 TRACE 132297: [emsesp] Me(0x0B) -R-> solar(0x30), SM100Energy(0x038E), length: 0x19
002+06:31:06.507 TRACE 132298: [emsesp] solar(0x30) -W-> Me(0x0B), SM100Energy(0x038E), data: 00 00 2F 44 00 00 13 C4 00 00 04 19
002+06:31:06.670 TRACE 132299: [emsesp] Me(0x0B) -R-> solar(0x30), SM100Time(0x0391), length: 0x19
002+06:31:06.715 TRACE 132300: [emsesp] solar(0x30) -W-> Me(0x0B), SM100Time(0x0391), data: 00 00 07 E9 00 00 00 00 00 00 0F B9 00 00 00 00 00 00 10 BF 00 00 00 00 00
002+06:31:06.877 TRACE 132301: [emsesp] Me(0x0B) -R-> solar(0x30), SM100Time(0x0391), length: 0x19 (offset 25)
002+06:31:06.927 TRACE 132302: [emsesp] solar(0x30) -W-> Me(0x0B), SM100Time(0x0391), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 24)
002+06:31:06.927 TRACE 132303: [emsesp] Me(0x0B) -R-> solar(0x30), SM100Config1(0x035F), length: 0x19
002+06:31:07.001 TRACE 132304: [emsesp] solar(0x30) -W-> Me(0x0B), SM100Config1(0x035F), data: 00 00 3C 00 1E 0A 0C 19 00 3C 19
002+06:31:07.221 TRACE 132305: [emsesp] Me(0x0B) -R-> solar(0x30), SM100HeatAssist(0x035C), length: 0x19
002+06:31:07.244 TRACE 132306: [emsesp] solar(0x30) -W-> Me(0x0B), SM100HeatAssist(0x035C), data: 3C 1E 00
002+06:31:07.408 TRACE 132307: [emsesp] Me(0x0B) -R-> solar(0x30), SM100Differential(0x0361), length: 0x19
002+06:31:07.433 TRACE 132308: [emsesp] solar(0x30) -W-> Me(0x0B), SM100Differential(0x0361), data: 3C 5A 14 1E 3C
002+06:31:25.732 TRACE 132309: [emsesp] solar(0x30) -B-> All(0x00), SM100Monitor(0x0362), data: 02 37 00 CE 80 00 00 C1 01 2C 01 80 80 00 80 00 80 00 80 00 80 00 01 A8
002+06:31:25.909 TRACE 132310: [emsesp] solar(0x30) -B-> All(0x00), SM100Monitor(0x0362), data: 80 00 (offset 24)
002+06:31:26.324 TRACE 132311: [emsesp] solar(0x30) -B-> All(0x00), SM100Monitor2(0x0363), data: 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00
002+06:31:26.534 TRACE 132312: [emsesp] solar(0x30) -B-> All(0x00), SM100Status(0x0364), data: 64 00 00 04 00 00 FF 00 00 64 00 00 64 00 00 00 00 00
002+06:31:26.988 TRACE 132313: [emsesp] solar(0x30) -B-> All(0x00), SM100Config(0x0366), data: 05 69 01 03 00
002+06:31:27.168 TRACE 132314: [emsesp] solar(0x30) -B-> All(0x00), ?(0x0368), data: 0C 00
002+06:31:27.532 TRACE 132315: [emsesp] solar(0x30) -B-> All(0x00), ?(0x0369), data: 00

```

MichaelDvP, 2024-03-28:
The valves we expect here:
`solar(0x30) -B-> All(0x00), SM100Status2(0x036A), data: 03 03 03 03 03 03 03 03 03 03 04 03`
Offset 10 with value 04 means pump switched on, all valves closed. Can you switch the valve manally or trigger a switching? Then log a minute before and  a minute after switching the valve to see the change. 
We have VS1 in offset 0, VS2 in offset 4, M1 in offset 7, Pump1 in offset 10 and pump2 in offset1, all other values are unknown.

Do you have a name for TS8? Looks lke offset 22 in 0x362
`solar(0x30) -B-> All(0x00), SM100Monitor(0x0362), data: 02 36 00 CE 80 00 00 C1 01 2C 01 80 80 00 80 00 80 00 80 00 80 00 01 A8`
with value 42.4°C (0x01A8).

proddy, 2024-11-03:
closing, no activity.

closed 2024-11-03 by proddy