emsesp/EMS-ESP32 #1679: MM200 mixer module - new entities only shown for hc2, not for hc1

demerstraat, 2024-03-29:
### PROBLEM DESCRIPTION

ESP32-S3 - 3.7.0-dev.1 and 3.7.0-test.0

3 new entites for MM200 (product id = 161, version 29.05) 
appear only for hc2 (dev id = 0x21)  and not for hc1 (dev id = 0x20)
- hc2 activated
- hc2 time to set valve
- hc2 flow temperature offset for mixer

### REQUESTED INFORMATION

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

```lua
{
  "System Info": {
    "version": "3.7.0-dev.1",
    "uptime": "000+00:04:55.527",
    "uptime (seconds)": 295,
    "platform": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.14",
    "sdk": "v4.4.6-dirty",
    "free mem": 185,
    "max alloc": 171,
    "free caps": 8322,
    "used app": 2156,
    "free app": 5972,
    "partition": "app1",
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -88,
    "BSSID": "set",
    "TxPower setting": 78,
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 335,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time water": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 985,
    "bus reads (tx)": 109,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 1,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32S3",
    "locale": "en",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 100,
    "web log buffer": 100
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si",
      "device id": "0x08",
      "product id": 123,
      "version": "07.00",
      "entities": 83,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
      "handlers fetched": "0x14 0x16 0x33 0x26 0x28",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE9",
      "handlers ignored": "0x1E"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "18.06",
      "entities": 94,
      "handlers received": "0x06 0xA2 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x0291 0x02CE 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x13 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02D0 0x0469 0x02D2 0x046A 0x031E"
    },
    {
      "type": "mixer",
      "name": "MM200",
      "device id": "0x21",
      "product id": 161,
      "version": "29.05",
      "entities": 7,
      "handlers received": "0x02D8",
      "handlers fetched": "0x02CE",
      "handlers ignored": "0x02E2 0x02EC 0x0255 0x16 0x33"
    },
    {
      "type": "mixer",
      "name": "MM200",
      "device id": "0x20",
      "product id": 161,
      "version": "29.05",
      "entities": 4,
      "handlers received": "0x02D7",
      "handlers ignored": "0x02E1 0x02EB 0x0255 0x16 0x33"
    },
    {
      "type": "controller",
      "name": "BC25",
      "device id": "0x09",
      "product id": 125,
      "version": "03.05",
      "entities": 0
    }
  ]
}

```

### TO REPRODUCE

N/A

### EXPECTED BEHAVIOUR

identical entities for hc1 

### SCREENSHOTS

N/A

**(Please, remember to close the issue when the problem has been addressed)**

MichaelDvP, 2024-03-30:
As mentioned already in #1554, telegrams not in a row are uncommon. Could you check interminal `read 20 2CD` if ths isthe info we need?
(Or add a custom entity for device 0x20, type 0x2CD, raw, length 3 and check the value in dashboard).

demerstraat, 2024-04-09:
```lua
{
ems-esp:# read 20 2CD
010+18:56:45.531 N 8: [emsesp] mixer(0x20) -W-> Me(0x0B), ?(0x02CD), data: FF 0E 05 FF 1E 00
}

demerstraat, 2024-04-09:
great, thanks.
can confirm it works as expected on 3.7.0-dev.2 (ESP32-S3)
write command also works

[image]
[image]

closed 2024-04-09 by demerstraat