emsesp/EMS-ESP32 #1688: Whatever I am choosing as formant the entity keeps the new, longer entity name

proddy, 2024-04-06:
Whatever I am choosing as formant the entity keeps the new, longer entity name… what am I doing wrong?

Edit: Seems the old config was "single topic" + "MQTT name", not "Long Name". Now the topics are back to what they have been. Thanks for the hint and sorry to hijack the thread.

> Am 06.04.2024 um 10:04 schrieb Proddy ***@***.***>:
> 
> 
> Maybe not related, but since I updated to 3.6.5 most of the ems-entities in HA have changed their name. They have an added "ems-esp.*" now in front of the entity-name, as well as different naming-scheme, which** makes them essentially new entities in HA. Which means as well, that all data colllected is now attached to old entities, so long terms statistics are gone for these entities.
> 
> Just one example of the current flow temp (upper on is the old, lower one the new name)  <https://private-user-images.githubusercontent.com/25442184/320189502-622cfb6d-b93d-43e4-8fd9-4ed075b36d44.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIzOTA1MjgsIm5iZiI6MTcxMjM5MDIyOCwicGF0aCI6Ii8yNTQ0MjE4NC8zMjAxODk1MDItNjIyY2ZiNmQtYjkzZC00M2U0LThmZDktNGVkMDc1YjM2ZDQ0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA2VDA3NTcwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBmOGU4MDM0NzJkOTZhYmRiNDVhMjM0ZjkzMDU0YjNjMzdmMTllNjQwNjQ2YjRjZTY4ODg5ZTQ3NTAxN2FhOTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.iHTuxupPaZ4CacUHI1xU4v6hM1-wUAsm1lR51tuuBd4>
> I don't think they are related, but its a good call-out.
> 
> Only the title has changed, not the HA entity name, e.g.
> 
> state_class: measurement
> unit_of_measurement: °C
> device_class: temperature
> friendly_name: ems-esp Boiler Set flow temperature
> This was done because some users have multiple EMS-ESPs, and it was causing conflicts in HA, so this way, each entity is unique. If you want the old 3.6.1 naming, there's a setting in EMS-ESP called "Entity ID format" as described on https://emsesp.github.io/docs/Configuring/#temperature-sensors
> 
> —
> Reply to this email directly, view it on GitHub <https://github.com/emsesp/EMS-ESP32/issues/1684#issuecomment-2041010697>, or unsubscribe <https://github.com/notifications/unsubscribe-auth/AGCDPCGPFT62GKJOHGAWIOTY36UBLAVCNFSM6AAAAABFSYP4GCVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDANBRGAYTANRZG4>.
> You are receiving this because you commented.
>

_Originally posted by @zaphood1967 in https://github.com/emsesp/EMS-ESP32/issues/1684#issuecomment-2041048399_

MichaelDvP, 2024-04-07:
Can not reproduce, example for one entity:
Single/long(v3.4):
`{"uniq_id":"boiler_selected_flow_temperature","obj_id":"boiler_selected_flow_temperature","cmd_t":"esp32s3/boiler/selflowtemp","mode":"box","step":1,"min":0,"max":90,"ic":"mdi:coolant-temperature","name":"Selected flow temperature","stat_t":"esp32s3/boiler_data","val_tpl":"{{value_json.selflowtemp if value_json.selflowtemp is defined else 0}}","avty":[{"t":"esp32s3/boiler_data","val_tpl":"{{'online' if value_json.selflowtemp is defined else 'offline'}}"}],"avty_mode":"all","dev":{"ids":["esp32s3-boiler"],"name":"esp32s3 Boiler"}}`
Single/short:
`{"uniq_id":"boiler_selflowtemp","obj_id":"boiler_selflowtemp","cmd_t":"esp32s3/boiler/selflowtemp","mode":"box","step":1,"min":0,"max":90,"ic":"mdi:coolant-temperature","name":"Selected flow temperature","stat_t":"esp32s3/boiler_data","val_tpl":"{{value_json.selflowtemp if value_json.selflowtemp is defined else 0}}","avty":[{"t":"esp32s3/boiler_data","val_tpl":"{{'online' if value_json.selflowtemp is defined else 'offline'}}"}],"avty_mode":"all","dev":{"ids":["esp32s3-boiler"],"name":"esp32s3 Boiler"}}`
Mult/short:
`{"uniq_id":"esp32s3_boiler_selflowtemp","obj_id":"esp32s3_boiler_selflowtemp","cmd_t":"esp32s3/boiler/selflowtemp","mode":"box","step":1,"min":0,"max":90,"ic":"mdi:coolant-temperature","name":"Selected flow temperature","stat_t":"esp32s3/boiler_data","val_tpl":"{{value_json.selflowtemp if value_json.selflowtemp is defined else 0}}","avty":[{"t":"esp32s3/boiler_data","val_tpl":"{{'online' if value_json.selflowtemp is defined else 'offline'}}"}],"avty_mode":"all","dev":{"ids":["esp32s3-boiler"],"name":"esp32s3 Boiler"}}`

proddy, 2024-04-07:
Yes, I tried too and it worked. The reporter said it wasn't an issue after all so I'll close it

https://github.com/emsesp/EMS-ESP32/issues/1684#issuecomment-2041048399

closed 2024-04-07 by proddy

zaphood1967, 2024-04-07:
Fine with me. After several attempts and reboots on both sides (ems and HA) it finally worked. Tnx for your support

#1684 (comment)